apache/nuttx #1568: STM32F4 core over-drive enable/disable sequence problem.

fjpanag, 2020-08-12:
During my recent work in over-drive settings of STM32F4, I came across the function `void stm32_pwr_enableoverdrive(bool state)`.

ST recommends a specific sequence for enabling and disabling core over-drive, but this function violates the sequence.  
The correct sequence can be found in [RM0090](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&ved=2ahUKEwjnqJaDupXrAhVE1xoKHasjDScQFjAAegQIARAB&url=https%3A%2F%2Fwww.st.com%2Fresource%2Fen%2Freference_manual%2Fdm00031020-stm32f405-415-stm32f407-417-stm32f427-437-and-stm32f429-439-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf&usg=AOvVaw2x8tbTRz8d9PfqXBk3qZ74), page 123.

Unfortunately fixing this function is not easy. This is because:
1. It requires for all peripheral clocks to be disabled, and disabling/re-enabling of the system PLL.
2. There is the note: 
> It is recommended to enter Over-drive mode when the application is not running critical
> tasks and when the system clock source is either HSI or HSE.

I don't know how to interpret "critical tasks" here...

I am not sure how this can be implemented. It seems that this function must keep a state of the previous configuration, disable everything, do its work, and re-initialize everything according to the last configuration. Complicated, and still I don't know what consequences there may be for other drivers, or the application code. 

Luckily this function is not used anywhere now. Core over-drive is handled directly by the clock initialization procedure, without calling it, following a correct procedure.

fjpanag, 2020-09-17:
@davids5 I agree that it is best to delete it.
It seems that it cannot be fixed properly, and as no other NuttX code uses it, it won't cause any issues removing it.

Users' code may use it, however. By deleting it, their code may break.
But I think this is good. It will clearly indicate the bug, on next build, and will encourage people to fix their code.

Shall I proceed with a PR?

davids5, 2020-09-17:
@fjpanag - 

> Users' code may use it, however. By deleting it, their code may break.
But I think this is good. It will clearly indicate the bug, on next build, and will encourage people to fix their code.

Agreed!

> Shall I proceed with a PR?

Yes Please

slorquet, 2020-09-18:
Is this function really broken or hard to use?

it has some benefit, it allows stm32f4 to run at 180 Mhz instead of being restricted to lower speed. If I understand correctly, it just has to be called early in the clock setup process?

Why not do this instead of removing a useful performance feature?

closed 2022-11-04 by fjpanag