apache/nuttx #1570: Hardcoded Pin mux, pad control and Drive Strength (AKA Slew-rate and Frequency) Settings

davids5, 2020-08-12:
Discussion from [STM32 IO compensation cell initialization fix. Discussion](https://github.com/apache/incubator-nuttx/pull/1544#issuecomment-671897714)

I discovered some issues with the massive over shoot on SPI. (way out of spec, MAX VIH to the receiving IC's) I am sure there are others. This is total a HW use case dependent setting and effects EMI, crosstalk etc.

I think we need to do a pass on ALL the pin mux files to removes the slew-rate setting from the pin maps. Unfortunately has massive ripple effects and will need to be done in such a way to allow an override, and not break all the boards.

@fjpanag:
 @davids5 I think that the default slew rate is for 50Mhz. This is not excessive for SPI.
This seems like a hardware design problem (lines termination etc), and I don't thing we should make a change in NuttX for this. It make break other user's code. Maybe we should add a new configuration in Kconfig, so the user can select the slew rate for every SPI, while keeping the default at 50Hz.

(As a side note, ringing usually does not have enough energy to cause latch-ups in other chips, although max. VIH is violated. But of course you know better, as you tested the actual hardware...)

@davids5:
>@davids5 I think that the default slew rate is for 50Mhz. This is not excessive for SPI.

I here you, I original thought the same thing, hence the defaults*. 

> This seems like a hardware design problem (lines termination etc), and I don't thing we should make a change in NuttX for this. 

This is the point. The line termination and characteristic of an interface is hardware dependent, really board dependent.  
  1) Signal integrity is effected. (cross talk) but over driving the lines.
  2) FCC certifications is effected by slew-rate (drive strength).  

Our EE group (amazing Wizards) have suggested we want to set the minimum drive strength that achieves quality signal integrity and no more.  

Contrast a board 3 x 3 cm PCB (controlled impedance) with an SoC and SPI to devices on it and one with connectors and cables to it devices. The impedance will vary greatly.  This really requires looking at the signals (with proper equipment) and adjusting drive strength correctly. 

>It make break other user's code. Maybe we should add a new configuration in Kconfig, so the user can select the slew rate for every SPI, while keeping the default at 50Hz.

Yes I 100% agree we should not make a change that is breaking, and I would choose to default it as is.

> (As a side note, ringing usually does not have enough energy to cause latch-ups in other chips, although max. VIH is violated. But of course you know better, as you tested the actual hardware...)

The failure was on an H7 using SPI DMA on a mems sensor 7mm from the SoC. When the clock was continuous because of the DMA the device failed at the high drive strength. Lower it and it works. Probe it with a 10X 14pf Probe and it works. Probe it with .9 pf active Probe and you see .7V over VDD on every rising edge. 

My suspicion is that the current injection to the substrate via the body diodes, caused the ADC to or loose its bias and clamp as the data was a FIFO full of 00s. Lower the drive strength and once the over shoot was gone it works fine. 

The code need to fix the overcome the problem is [ugly and does not scale well](https://github.com/PX4/Firmware/blob/master/boards/px4/fmu-v6x/nuttx-config/include/board.h#L411) but moreover we [state](https://github.com/apache/incubator-nuttx/blob/master/arch/arm/src/stm32h7/hardware/stm32h7x3xx_pinmap.h#L67-L70) the setting should be done in board.h. We (I) broke that contract based on *above. 

This opens another heated can of worms. There are defaults. where the pin mux has only one appearance. These will need to be have numbers added (I suggest _0 to make it obvious). To allow proper usage. Again massive ripple.

davids5, 2023-04-07:
While some solutions were discussed in
- https://github.com/apache/nuttx/pull/5012

I would like to propose a solution for this issue as a request for comment:

1. That will not effect any existing boards
2. Will allow us to fix the issues without forcing massive changes.
3. Eventually after N more releases of NuttX deprecate the solution.

Steps to get there:

1. Kconfig for all effected arches will have STM32xxx_USE_LEGACY_PINMAP set to yes as a default.
2. Rework top level pinmap files E.G. hardware/stm32_pinmap.h. 
3. The current pinmap file will be renamed with `_legacy`  E.G. hardware/stm32h7x3xx_pinmap_legacy.h
3. Rework chip specific files removing speeds and adding _0 to the previous no-selectable pins with speeds
4. Rework chip specific files adding _0 to the previous no-selectable pins

The `hardware/stm32_pinmap.h` will have the following structure

```
#if defined(STM32H7_USE_LEGACY_PINMAP )
#  if defined(CONFIG_STM32H7_STM32H7X3XX)
#    include "hardware/stm32h7x3xx_pinmap_legacy.h"
#  elif defined(CONFIG_STM32H7_STM32H7X7XX)
#    include "hardware/stm32h7x3xx_pinmap_legacy.h"
#  else
#     error "Unsupported STM32 H7 Legacy Pin map"
#  endif
#else
#  if defined(CONFIG_STM32H7_STM32H7X3XX)
#    include "hardware/stm32h7x3xx_pinmap.h"
#  elif defined(CONFIG_STM32H7_STM32H7X7XX)
#    include "hardware/stm32h7x3xx_pinmap.h"
#  else
#     error "Unsupported STM32 H7 Pin map"
#  endif
# endif

```

Moving forward boards will turn off STM32xx_USE_LEGACY_PINMAP and update the board.h files to fully define
the pins with selected speeds.

was:
```
#define GPIO_SDMMC2_CK   GPIO_SDMMC2_CK_1  /* PD6  FC_PD6_SDMMC2_CK  */
#define GPIO_SDMMC2_CMD  GPIO_SDMMC2_CMD_1 /* PD7  FC_PD7_SDMMC2_CMD */
//      GPIO_SDMMC2_D0   No Remap          /* PB14 FC_PB14_SDMMC2_D0 */
//      GPIO_SDMMC2_D1   No Remap          /* PB15 FC_PB15_SDMMC2_D1 */
#define GPIO_SDMMC2_D2   GPIO_SDMMC2_D2_1  /* PG11 FC_PG11_SDMMC2_D2 */
//      GPIO_SDMMC2_D3    No Remap         /* PB4  FC_PB4_SDMMC2_D3  */
```
is:
```
#define GPIO_SDMMC2_CK      (GPIO_SDMMC2_CK_1    |  GPIO_SPEED_25MHz)  /* PD6  FC_PD6_SDMMC2_CK  */
#define GPIO_SDMMC2_CMD  (GPIO_SDMMC2_CMD_1 |  GPIO_SPEED_2MHz)  /* PD7  FC_PD7_SDMMC2_CMD */
#define GPIO_SDMMC2_D0   (GPIO_SDMMC2_D0_0  | GPIO_SPEED_2MHz) /* PB14 FC_PB14_SDMMC2_D0 */
#define GPIO_SDMMC2_D1   (GPIO_SDMMC2_D1_0  |  GPIO_SPEED_2MHz)/* PB15 FC_PB15_SDMMC2_D1 */
#define GPIO_SDMMC2_D2    GPIO_SDMMC2_D2_1  | GPIO_SPEED_2MHz)  /* PG11 FC_PG11_SDMMC2_D2 */
#define GPIO_SDMMC2_D3   (GPIO_SDMMC2_D3_0  | GPIO_SPEED_2MHz) /* PB4  FC_PB4_SDMMC2_D3  */
```

To maintain the speed that was set in the legacy files just diff the `_legacy` file with the non legacy file and use that
speed when creating the #define.

slorquet, 2023-04-07:
I am a stm32h7 user, thanks for the mailing list notification or I would have missed this.

I agree with this proposed change.

It is important to notify all stm32h7 users in a very visible way, so everyone can switch to the new definitions without default speeds. I suggest not to restrict this info to final release notes, but have it somewhere visible that indicates the important trunk change, including the commit that introduces the change.

It is also important to make sure that the legacy mechanism actually works for everyone.
As soon as this proposed change is implemented, I will be able to test that the behaviour of my board is not broken.

slorquet, 2023-04-07:
wait, if I build your new commit with a full stored defconfig, and I dont regenerate this config, the default setting will not be added!

davids5, 2023-04-10:
@slorquet have a look at https://github.com/apache/nuttx/pull/8992. Let me know if it addresses all the concerns you have.