apache/nuttx #1598: stm32f7 serial miss SERIAL_HAVE_CONSOLE_DMA definition

lukegluke, 2020-08-17:
stm32f7/stm32_uart.h miss defining of SERIAL_HAVE_CONSOLE_DMA as it is done in general stm32:
https://github.com/apache/incubator-nuttx/blob/85428036abfe5d2e8fc9caec8b573dc2dafed7ac/arch/arm/src/stm32/stm32_uart.h#L338-L357

This can leads to hardfault when trying to write with not yet initialized serial dma in case when serial is used for console and something try to write to console on boot (for example board_hardfault_init in px4 can write to syslog (and to console if CONFIG_SYSLOG_CONSOLE defined), just for your note @davids5 if it is important). 

I could done a PR, but I wonder with several things:
1) Why CONFIG_USARTx_TXDMA are not listed in [configuration variables](https://cwiki.apache.org/confluence/display/NUTTX/Configuration+Variables)?
2) Can TXDMA be used without RXDMA? Because now TXDMA checks are performed inside SERIAL_HAVE_RXDMA
https://github.com/apache/incubator-nuttx/blob/85428036abfe5d2e8fc9caec8b573dc2dafed7ac/arch/arm/src/stm32f7/stm32_serial.c#L89
3) Why stm32/stm32_uart.h checks only **RX** DMA to define SERIAL_HAVE_CONSOLE_DMA?

davids5, 2020-08-17:
> stm32f7/stm32_uart.h miss defining of SERIAL_HAVE_CONSOLE_DMA as it is done in general stm32: [...]

These come in from the serial driver. I believe  that @patacongo did a pass were he moved all the different UxART into common code.   But I we lost the enables. I did not follow the change at the time. We should get a clarification from Greg, on how it is suppose to work. 

[image]

> 2. Can TXDMA be used without RXDMA? Because now TXDMA checks are performed inside SERIAL_HAVE_RXDMA
>    https://github.com/apache/incubator-nuttx/blob/85428036abfe5d2e8fc9caec8b573dc2dafed7ac/arch/arm/src/stm32f7/stm32_serial.c#L89

Yes as it ends at Line 183
https://github.com/apache/incubator-nuttx/blob/85428036abfe5d2e8fc9caec8b573dc2dafed7ac/arch/arm/src/stm32f7/stm32_serial.c#L183

> 3. Why stm32/stm32_uart.h checks only **RX** DMA to define SERIAL_HAVE_CONSOLE_DMA?

There is no TX DMA on the F4.

lukegluke, 2020-08-17:
> Yes as it ends at Line 183

Right! My bad.

> There is no TX DMA on the F4.

Ah, ok. Missed that.

Thanks for clarifications on this questions.

lukegluke, 2020-11-02:
Still an issues. But found out that issue is in stm32f7/stm32_serial.c
https://github.com/apache/incubator-nuttx/blob/04ec9cf98960d6d6f3de50d67d4c53ce78bde3f5/arch/arm/src/stm32f7/stm32_serial.c#L3664

Should be 
#if defined(SERIAL_HAVE_CONSOLE_RXDMA) || defined(SERIAL_HAVE_CONSOLE_TXDMA)

I missed that stm32f7/stm32_uart.h had already defines of two separate SERIAL_HAVE_CONSOLE_RXDMA, SERIAL_HAVE_CONSOLE_TXDMA added within your @davids5 commit baeaf54c6c3aa9371696a2012769403a9b169b50 but not used anywhere.
https://github.com/apache/incubator-nuttx/blob/04ec9cf98960d6d6f3de50d67d4c53ce78bde3f5/arch/arm/src/stm32f7/stm32_uart.h#L269-L273
https://github.com/apache/incubator-nuttx/blob/04ec9cf98960d6d6f3de50d67d4c53ce78bde3f5/arch/arm/src/stm32f7/stm32_uart.h#L290-L294

closed 2020-11-03 by davids5