apache/nuttx #1638: Documentation roadmap

v01d, 2020-08-24:
These are the pending tasks to complete the new documentation:

- [ ] devise the CI workflows necessary to have this published on nuttx website. documentation should be deployed for master and for every tagged release. The documentation itself allows to choose which version to read (see mockup dropdown just above search field in documentation)
- [ ] fill supported boards section using content from current README files
- [ ] autogenerate the list of supported platforms based on all boards existing on the repo
- [ ] allow the docs to pickup information from the application repo so that applications are also documented (or at least initially, listed)
- [ ] look into defining a workflow which would build documentation for each PR somewhere (github pages for the nuttx repo?) so that previewing a PR is simple.

Various parts of the documentation require writing new content (possibly already existing in other external sources):
- [ ] quickstart: this is a very important section for newcomers. this guide should give instructions on how to try nuttx on a commonly available board with good support for nuttx (some stm32 nucleo/discovery?).
- [ ] introduction -> development environments: check if this is still up to date and revise whether any given particular toolchain should be recommended. this also requires a bit of rewrite to be more like a documentation and not simply a report of toolchain support status.
- [ ] introduction -> trademarks: revise this. it mentions NuttX to be trademarked which I'm not sure it is still true.
- [ ] releases: if the release notes file was in markdown we could directly include it here. anyway, it is quite large so it may be better to only hold release notes of latest version (I believe this was discussed before)
- [ ] start looking into the wiki, looking for information that should be converted to documentation of a subsystem not currently documented and for guides for the corresponding section. providing documentation for a subsystem should be prioritized over just providing a guide covering a particular use case.

Feel free to suggest new tasks or discuss the ones already listed.

adamfeuer, 2020-08-24:
A couple of smaller items:

- [ ] Add [`INVIOLABLES.txt`](https://github.com/apache/incubator-nuttx/blob/master/INVIOLABLES.txt) to the docs in a prominent place
- [ ] Add a `README.md` to the Documentation folder to explain the basics of how to build the docs, or point the user at the right page in the built docs.
- [ ] Extend Quickstart docs to cover a number of different boards, maybe using tabbed instructions or something else to let people have updated docs for several popular boards.

v01d, 2020-08-24:
Good points. I would convert inviolables to Markdown in that case.

adamfeuer, 2020-08-24:
@v01d Yeah, `INVIOLABLES.txt` is already pretty close to Markdown already. I updated my item in the comment above to reflect this.

adamfeuer, 2020-08-24:
@v01d @btashton I'm going to hold off on any doc PRs until we get the docs deploying via CI. Let me know if I can help get that working.

btashton, 2020-08-25:
See https://github.com/apache/incubator-nuttx-website/pull/28 for the first round of integration.  It is manual trigger for now, but the hook is there for automation just need apache infra to add a token to this repo to trigger it.

btashton, 2020-08-25:
Releases should probably link here https://nuttx.apache.org/download/ where we manage this already

v01d, 2020-08-25:
Right, then maybe a separate section does not make much sense, I will change the task.

btashton, 2020-08-30:
The Doxygen build for NXWidgets does not exist any more.  This can be integrated into Sphinx without too much issue which would simplify the whole process.

btashton, 2020-08-30:
Could we maybe integrate that work that was done with the Kconfig webview tool as well?

v01d, 2020-08-30:
> The Doxygen build for NXWidgets does not exist any more. This can be integrated into Sphinx without too much issue which would simplify the whole process.

How was it built before?
I understand you mean to include this as part of the build process of Sphinx docs, right? If so, there is some integration that is required via breathe extension I believe.

v01d, 2020-08-30:
> Could we maybe integrate that work that was done with the Kconfig webview tool as well?

You mean @w8jcik 's work from http://nuttx-config.nxtlabs.pl ?
I see it supports linking as: nuttx-config.nxtlabs.pl/#/?current=MM_SMALL (without CONFIG_ prefix), so we could setup something in sphinx so that we get links with this format (probably should require some new tag such as :c:kconfig:).
In terms of embedding this web viewer itself, it seems it is designed to be standalone since it integrates its own sidebar so it would probably not look right if embedded inside the content section of sphinx docs. Instead it could be under a different website subdirectory (ie: nuttx.apache.org/kconfig/<release>).

patacongo, 2020-08-30:
> How was it built before?

NxWidgets used to be in a separate repository.  The Doxygen build was in a directory there, but was not integrated when NxWidgets was moved to apps/.  You can get the last NxWidgets standalone release from SourceForge:  https://sourceforge.net/projects/nuttx/files/NxWidgets/ if you want the Doxygen build directory.

v01d, 2020-08-30:
> NxWidgets used to be in a separate repository. The Doxygen build was in a directory there, but was not integrated when NxWidgets was moved to apps/. You can get the last NxWidgets standalone release [...]

Ah, I didn't realise it was not part of the apache repos.

patacongo, 2020-08-30:
> Ah, I didn't realise it was not part of the apache repos.

'It' being NxWidgets?  It was merged into apps/ in 2018.

Or 'it' be the Doxygen build?  Hmm... looks like it is there:  apps/graphics/nxwidgets/Deoxygen.  I was mistaken.  Sorry, I answered too quickly (recurring problem).  But I don't think it has never been used since NxWidgets was relocated.

v01d, 2020-08-31:
@patacongo I would like to fix the issue about two sections being named the same (this happened in the original HTML docs), but I don't know what was your original intention at the time. You probably have the correct name for one of them in mind. They are the following ones:

* https://nuttx.apache.org/docs/latest/reference/user/01_task_control.html
* https://nuttx.apache.org/docs/latest/reference/user/03_task_control.html

Let me know and I can use the proper section name.

btashton, 2020-09-02:
> @patacongo I would like to fix the issue about two sections being named the same (this happened in the original HTML docs), but I don't know what was your original intention at the time. You [...]

I think we should merge the three task sections.  The third one spans both the first and the second in terms of scope.

cogburnd02, 2020-09-17:
"introduction -> trademarks: revise this. it mentions NuttX to be trademarked which I'm not sure it is still true."

'NuttX' is described as LIVE trademark by USPTO.

http://tess2.uspto.gov/

patacongo, 2020-09-17:
> 
> 
> "introduction -> trademarks: revise this. it mentions NuttX to be trademarked which I'm not sure it is still true."
> 
> 'NuttX' is described as LIVE trademark by USPTO.
> 
> http://tess2.uspto.gov/

Yes, I hold the NuttX trademark.  I will be donated to the ASF if/when the project graduates.

v01d, 2020-09-17:
Ok, then no change needed for that section.

btashton, 2020-09-19:
@v01d for converting content that exists in the wiki that should be integrated into the documentation. This is what I did that actually worked quite well:

1. Export the the whole wiki or specific pages [here](https://cwiki.apache.org/confluence/spaces/exportspacewelcome.action?key=NUTTX).  Note this requires admin access to the space so let me know if you need that.
2. Take the generated HTML and use [turndown](https://domchristie.github.io/turndown/) to convert to markdown.

Here is a sample for the (Building NuttX with Applications Outside of the Source Tree)[https://cwiki.apache.org/confluence/display/NUTTX/Building+NuttX+with+Applications+Outside+of+the+Source+Tree] page where I show the generated HTML and the generated Markdown.

https://gist.github.com/btashton/8f5593e3280da073d801d658a84a65c9