apache/nuttx #1643: stm32f746g-disco:lvgl, fb, nxdemo, nxterm  "arm_hardfault: PANIC!!! Hard fault:"

TA1DB, 2020-08-25:
Among the stm32f746g-disco configurations, only the "stm32f746-disco:nsh" works as expected, and nsh> prompt comes through the usart6 terminal. For all others; stm32f746g-disco:lvgl, fb, nxdemo, nxterm the usart6 terminal does not respond, and from acm0 terminal the "arm_hardfault: PANIC !!! Hard fault:" response is received. Am I making a mistake or is this a bug?

acassis, 2020-08-25:
Hi TA1DB, it is not a BUG. The other configs are using UART1 instead UART6. Thank you for reporting it, I will submit a fix!

raiden00pl, 2020-08-25:
Hi Alan, UART6 seems like a better choice for the default serial port as it is connected to the ST-link virtual COM. Connecting an external UART interface can be annoying if the board already has one built-in.

acassis, 2020-08-25:
Hi Raiden, yes I was looking the schematic and datasheet when I saw your message here. I will move nsh config from USART6 to USART1 and update the README.

TA1DB, 2020-08-25:
Hi @acassis , thanks for your prompt and helpful answer. 
My choice would be UART6 because we used UART6 in my other linux trials, so I could continue my nuttx trials without changing pins, but I respect your explanations and preferences. 
Could I somehow easily choose UART6 instead of UART1 in my configuration with "make menuconfig" or in any way? I am new to Nuttx; can you give me information about this possibility.

acassis, 2020-08-25:
Hi TA1DB, yes all you need to do it:
$ make menuconfig

System Type  --->
    STM32 Peripheral Support  --->
        [*] USART6

Device Drivers  --->
    [*] Serial Driver Support  --->
        Serial console (USART1)  --->
            (X) USART6

It is important to keep it on USART1 because it will make the live of other users more easy. I just submitted the PR #1644 to fix it.

TA1DB, 2020-08-25:
Hi @acassis 
I have to re-open the issue because my main problem was related to receiving 
`"arm_hardfault: PANIC!!! Hard fault:""`
instead of a **nsh>** prompt.
After this revision now the properly working :nsh config moved from USART6 to USART1, it works well as it was working previously but the other configs (lvgl, fb, nxdemo..) respond with "arm_hardfault: PANIC!!! Hard fault:" after reset on USART1 (ttyACM0 in my case).

acassis, 2020-08-25:
Hi @TA1DB,
I don't have this board here to help you with debugging. Probably it could be something easy to fix, like when the stack size defined is too small. BTW, you can try to verify what is causing the hardfault, please take a look in this documentation:
https://cwiki.apache.org/confluence/display/NUTTX/Analyzing+Cortex-M+Hardfaults

If you have other LVGL supported board, like the STM32F429I-Disco, please try it. I have this board here, case you face similar issue.

TA1DB, 2020-08-26:
I have done some experiments to resolve the issue with my limited knowledge and limited experience about Nuttx. Stm32f746g-disco: nsh building and running, no problem! I canfigured this. Then I activated lvgl with "make menuconfig". It compiled and worked again without any problems. When I activate the "lvgl demo" in examples then the source code is not compiled, "make" returns with the following error messages:

 `In file included from ./tp.c:57:0:
./tp.h:63:4: error: #error "Input device support is not enabled (CONFIG_INPUT)"
 #  error "Input device support is not enabled (CONFIG_INPUT)"
    ^
ERROR: arm-none-eabi-gcc failed: 1
       command: arm-none-eabi-gcc -MT ./tp.home.user.nuttxspace.apps.examples.lvgldemo.o  -M '-fno-builtin' '-Wall' '-Wstrict-prototypes' '-Wshadow' '-Wundef' '-g' '-mcpu=cortex-m7' '-mthumb' '-mfloat-abi=soft' '-isystem' '/home/user/nuttxspace/nuttx/include' '-D__KERNEL__' '-pipe' '-I' '/home/user/nuttxspace/apps/graphics/lvgl' '-I' '/home/user/nuttxspace/apps/graphics/lvgl/lvgl' '-I' '/home/user/nuttxspace/apps/include' '-DLV_LVGL_H_INCLUDE_SIMPLE' ./tp.c
/home/user/nuttxspace/apps/Application.mk:228: recipe for target '.depend' failed
make[2]: *** [.depend] Error 1
make[2]: Leaving directory '/home/user/nuttxspace/apps/examples/lvgldemo'
Makefile:55: recipe for target '/home/user/nuttxspace/apps/examples/lvgldemo_depend' failed
make[1]: *** [/home/murat/nuttxspace/apps/examples/lvgldemo_depend] Error 2
make[1]: Leaving directory '/home/murat/nuttxspace/apps'
tools/Makefile.unix:446: recipe for target 'pass2dep' failed
make: *** [pass2dep] Error 2
`

acassis, 2020-08-26:
Hi Murat,
in this case the build error is reporting the issue to you:

#error "Input device support is not enabled (CONFIG_INPUT)" error "Input device support is not enabled (CONFIG_INPUT)"

The LVGL demo needs an Input device (i.e. a TouchScreen driver or a Mouse) to work. You could see in the lvgl/defconfig what is enabled and try to enable it too.

TA1DB, 2020-08-27:
Hello Alan. 
By this way I could build a working .bin file together with some "examples" including lvgl as well. nsh> promt comes and if I enter ? or help I see the list of builtin apps. However if I start lvgl or tc then I get the same "arm_hardfault: PANIC!!! Hard fault:" response. On the other hand the "Hello" works as expected.

acassis, 2020-08-27:
Hi Murat, the functions up_fbinitialize() and up_fbgetvplane() are here:

boards/arm/stm32f7/stm32f746g-disco/src/stm32_lcd.c:int up_fbinitialize(int display)

But this file depends on CONFIG_STM32F7_LTDC enabled in the menuconfig to be compiled:

ifeq ($(CONFIG_STM32F7_LTDC),y)
CSRCS += stm32_lcd.c
endif

So, try to enable it and see what you get.

TA1DB, 2020-08-28:
Hi, Alan, now the compiler complains about the ltdc bubber size . How can I enter a good buffer size? Can you recommend a good value to satisfy the compiler :) 

`make[2]: Leaving directory '/home/murat/nuttxspace/nuttx/boards/arm/stm32f7/stm32f746g-disco/src'
chip/stm32_ltdc.c:293:4: error: #error "STM32_LTDC_BUFFER_SIZE not large enough for frame buffers"
 #  error "STM32_LTDC_BUFFER_SIZE not large enough for frame buffers"
    
ERROR: arm-none-eabi-gcc failed: 1
       command: arm-none-eabi-gcc -M '-fno-builtin' '-Wall' '-Wstrict-prototypes' '-Wshadow' '-Wundef' '-g' '-mcpu=cortex-m7' '-mthumb' '-mfloat-abi=soft' '-isystem' '/home/murat/nuttxspace/nuttx/include' '-D__KERNEL__' '-pipe' '-I' '/home/murat/nuttxspace/nuttx/arch/arm/src/chip' '-I' '/home/murat/nuttxspace/nuttx/arch/arm/src/common' '-I' '/home/murat/nuttxspace/nuttx/arch/arm/src/armv7-m' '-I' '/home/murat/nuttxspace/nuttx/sched' '-D__KERNEL__' chip/stm32_ltdc.c
Makefile:199: recipe for target '.depend' failed
make[1]: *** [.depend] Error 1
make[1]: Leaving directory '/home/murat/nuttxspace/nuttx/arch/arm/src'
tools/Makefile.unix:446: recipe for target 'pass2dep' failed
make: *** [pass2dep] Error 2
`

acassis, 2020-08-28:
For stm32f429i-disco it is about 300KiB, try to use the same value:
CONFIG_STM32_LTDC_FB_SIZE=307200

patacongo, 2020-08-28:
> For stm32f429i-disco it is about 300KiB, try to use the same value:
> CONFIG_STM32_LTDC_FB_SIZE=307200
>
This is the internal SRAM:

CONFIG_RAM_SIZE=114688
CONFIG_RAM_START=0x20000000

The internal SRAM is insufficient to support frambebuffers. There are 
two framebuffers in the fb configuration that lie in external RAM:

CONFIG_STM32_DMA2D_FB_BASE=0xD07B5000
CONFIG_STM32_DMA2D_FB_SIZE=307200

CONFIG_STM32_LTDC_FB_BASE=0xD076A000
CONFIG_STM32_LTDC_FB_SIZE=307200

The address is determined by which FMC chip select you use.  The 
framebuffers are excluded from the heap via:

CONFIG_HEAP2_BASE=0xD0000000
CONFIG_HEAP2_SIZE=7774208

Just setting the framebuffer size without initializing and managing the 
external memory would not work.

TA1DB, 2020-08-28:
> Just setting the framebuffer size without initializing and managing the external memory would not work.

Yes. From my previous experince I remember that stm32f746g-disco board's display resolution 448x272 pixel 3x8 bit wide, here in this application there is the necessary informaton: 
[Creating a graphical application with LTDC](https://www.st.com/resource/en/application_note/dm00287603-lcdtft-display-controller-ltdc-on-stm32-mcus-stmicroelectronics.pdf) page 33

embeddedt, 2020-08-28:
I think the 448x272 in that PDF is just for the example image they're displaying; the display's actual dimensions are **480**x272.

TA1DB, 2020-08-28:
Framebuffer size is 480x272x3 = 391680 bytes
Framebuffer start address (can be) 0xC0000000
`In file included from chip/stm32_ltdc.c:58:0:
chip/stm32_ltdc.c: In function 'stm32_ltdc_lclutenable':
chip/stm32_ltdc.c:2018:43: error: 'struct stm32_ltdc_s' has no member named 'oinfo'
   regval = getreg32(stm32_cr_layer_t[layer->oinfo.overlay]);
                                           ^
/home/murat/nuttxspace/nuttx/arch/arm/src/common/arm_arch.h:43:55: note: in definition of macro 'getreg32'
 # define getreg32(a)          (*(volatile uint32_t *)(a))
                                                       ^
chip/stm32_ltdc.c:2033:42: error: 'struct stm32_ltdc_s' has no member named 'oinfo'
   putreg32(regval, stm32_cr_layer_t[layer->oinfo.overlay]);
                                          ^
/home/murat/nuttxspace/nuttx/arch/arm/src/common/arm_arch.h:44:55: note: in definition of macro 'putreg32'
 # define putreg32(v,a)        (*(volatile uint32_t *)(a) = (v))
                                                       ^
chip/stm32_ltdc.c: In function 'stm32_ltdc_lputclut':
chip/stm32_ltdc.c:2080:50: error: 'struct stm32_ltdc_s' has no member named 'oinfo'
       putreg32(regval, stm32_clutwr_layer_t[layer->oinfo.overlay]);
                                                  ^
/home/murat/nuttxspace/nuttx/arch/arm/src/common/arm_arch.h:44:55: note: in definition of macro 'putreg32'
 # define putreg32(v,a)        (*(volatile uint32_t *)(a) = (v))
                                                       ^
Makefile:154: recipe for target 'stm32_ltdc.o' failed
make[1]: *** [stm32_ltdc.o] Error 1
make[1]: Leaving directory '/home/murat/nuttxspace/nuttx/arch/arm/src'
tools/LibTargets.mk:161: recipe for target 'arch/arm/src/libarch.a' failed
make: *** [arch/arm/src/libarch.a] Error 2`

acassis, 2020-08-30:
Hi Murat,
please take a look in the source could to find the root of the issue:

#ifdef CONFIG_FB_OVERLAY
  struct   fb_overlayinfo_s oinfo;            /* Overlay info */
#endif

"oinfo" only will exist if CONFIG_FB_OVERLAY is selected in the menuconfig.

TA1DB, 2020-09-01:
Hi Alan,
As I've said in linkedin messaging I still get the `"arm_hardfault: PANIC!!! Hard fault:"` error at the end even if I solve the particular errors in compiling the applications/examples. My knowledge and experience in Nuttx is not yet sufficient to find a permanent solution to this issue (I want to improve it). I hope and look forward you get your stm32f746g-disco board soon and be able to resolve the issue. Thanks BR.

TA1DB, 2020-09-16:
just a note: the same problem happens in version 9.0.0 as well

TA1DB, 2020-10-28:
> Hi Murat, don't worry, my board should arrive this week and by the weekend I will try to help you. BTW, if you get a F429I-Disco you can play with LVGL without these issues, but keep in mind that this board it not so complete like the F746G0-Disco

Hi @acassis , did you receive your board ?
Please note there is a very similar issue [here](https://github.com/apache/incubator-nuttx/issues/1409#issuecomment-664523382) Hard fault in STM32H7 protected mode, myabe if you can resolve this issue our solution may help tehm tehere.

In the meantime I already played with lvgl on F429I-Disco but it's display is very small and the board has much less features, therefore not only me, I beleive many others will thank to you for your help. Regards, Murat.

TA1DB, 2020-11-11:
Hi Everybody,
Upon suggestion of Jonathan Gonzaga in [this discussion](https://forum.lvgl.io/t/lvgl-nuttx-touch-screen-performance/1498?u=ta1db) I checked my arm-none-eabi installation, surprisingly found that there was 2 installs, one from 2015 Ubuntu's default toollchain and the other one was my new 2020 installation, I just simply did apt-get uninstall gcc-arm-none-eabi then my new Nuttx builds started working normally. Lvgldemo also works fine. I am closing this issue. Thanks.

closed 2020-11-11 by TA1DB