apache/nuttx #1649: Clean up warnings from documentation build

btashton, 2020-08-25:
Right now there are a bunch of warning generated that need to be cleaned up.  Most are duplicate c declaration warnings and duplicate label warnings.  I think we can use C domains to handle the duplicated declarations.

v01d, 2020-08-25:
Yes, it is one of the tasks described in #1638 
The duplicated declarations are because they were duplicated in original docs. They should only be documented in their corresponding section and only referenced from everywhere else.
The duplicate labels are possibly due to using the "autosectionlabel" extension, which generates labels for all sections. It has the problem that if you repeat a section name within a document it will generate a duplicate label. Not sure what's better (to use manual labels everywhere or to avoid repeating section names within documents).

v01d, 2020-08-31:
I've opened #1676 to address duplicate declarations.
Other ones will be fixed when autosectionlabel is no longer used.
There is one which I'm not sure how to fix:
<pre>
/home/v01d/coding/nuttx_doc/nuttx/Documentation/components/nxflat.rst:345: WARNING: Could not lex literal_block as "c-objdump". Highlighting skipped.
</pre>

It is due to a block of assembly that looks like this:
<pre>
.. code-block:: c-objdump

          ldr     r1, .L0         /* Fetch the offset to the GOT entry */
          ldr     r1, [r10, r1]   /* Fetch the (relocated) address of 'x' from the GOT */
          ldr     r0, [r1, #0]    /* Fetch the value of 'x' */
          /* ... */
  .L1     .word   x(GOT)          /* Offset to entry in the GOT */
</pre>

I used `objdump-c` for it but it seems it does not recognize something. I'm not sure what is the proper highlight definition for this, maybe someone else knows.

btashton, 2020-08-31:
I would have to look later but there are a bunch of lexers for assembly supported. 

https://pygments.org/docs/lexers/#lexers-for-assembly-languages

v01d, 2020-09-02:
Thanks for checking. Indeed it works on that site but I still get this error:

/home/v01d/coding/nuttx_doc/nuttx/Documentation/components/nxflat.rst:345: WARNING: Could not lex literal_block as "gas". Highlighting skipped.

For `asm` is the same. Maybe there's something in the syntax that confuses sphinx?

btashton, 2020-09-02:
just built with this with no error:
```
diff --git a/Documentation/components/nxflat.rst b/Documentation/components/nxflat.rst
index 115a14fb79..22cce9f3bc 100644
--- a/Documentation/components/nxflat.rst
+++ b/Documentation/components/nxflat.rst
@@ -332,7 +332,7 @@ and, in ELF environments, another is the PLT (Procedure Lookup Table.
 For example, if your C code generated (ARM) assembly language like this
 without PIC:
 
-.. code-block:: c-objdump
+.. code-block:: asm
 
           ldr     r1, .L0         /* Fetch the offset to 'x' */
           ldr     r0, [r10, r1]   /* Load the value of 'x' with PIC offset */
@@ -342,7 +342,7 @@ without PIC:
 Then when PIC is enabled (say with the -fpic compiler option), it will
 generate code like this:
 
-.. code-block:: c-objdump
+.. code-block:: asm
 
           ldr     r1, .L0         /* Fetch the offset to the GOT entry */
           ldr     r1, [r10, r1]   /* Fetch the (relocated) address of 'x' from the GOT */
```

v01d, 2020-09-02:
That's really strange. Maybe my pygments version differs? I have 2.3.1 from system installed Ubuntu.

btashton, 2020-09-02:
We are quite a ways apart `Pygments==2.6.1`

v01d, 2020-09-02:
Should that be in the requirements.txt then?

v01d, 2020-09-02:
I did pip3 install pygments so that I installed latest version in my home and it is still complaining. If I use pyenv it works OK. I'm wondering about the usability of the requirements.txt then. Or maybe I'm doing something wrong.

btashton, 2020-09-02:
I think you have multiple versions stepping on each other.  I just use `requirements.txt`, I'm still not quite sure why we are fully pinning in requirements.txt and then `*` in Pipfile, these are both wrong to me.  I would expect to set minimal versions in the requirements.txt file.

```
--- a/Documentation/requirements.txt
+++ b/Documentation/requirements.txt
@@ -1,3 +1,4 @@
 sphinx-rtd-theme==0.5.0
 m2r2==0.2.5
 sphinx==3.2.1
+Pygments>=2.6.1

```

```
❯ pip freeze
alabaster==0.7.12
Babel==2.8.0
certifi==2020.6.20
chardet==3.0.4
docutils==0.16
idna==2.10
imagesize==1.2.0
Jinja2==2.11.2
m2r2==0.2.5
MarkupSafe==1.1.1
mistune==0.8.4
packaging==20.4
Pygments==2.6.1
pyparsing==2.4.7
pytz==2020.1
requests==2.24.0
six==1.15.0
snowballstemmer==2.0.0
Sphinx==3.2.1
sphinx-rtd-theme==0.5.0
sphinxcontrib-applehelp==1.0.2
sphinxcontrib-devhelp==1.0.2
sphinxcontrib-htmlhelp==1.0.3
sphinxcontrib-jsmath==1.0.1
sphinxcontrib-qthelp==1.0.3
sphinxcontrib-serializinghtml==1.1.4
urllib3==1.25.10
```

v01d, 2020-09-02:
Ok, it seems I had other packages out of date. I can confirm that after adding minimum version to requirements.txt and doing pip3 install -r requirements.txt it now builds.
As per using minimum versions instead of exact ones, I agree it is better in principle, unless regressions occur on those packages. The exact version ensures that docs will always build no matter what. Using "*" in the Pipfile indeed sounds that it could eventually break (besides the fact that both files express different requirements). @adamfeuer seemed to think differently.

btashton, 2020-09-02:
I am going to just submit a PR to make the pipenv the correct flow and have it generate the requirements.txt file.  I cannot handle both workflows.

btashton, 2020-09-02:
The pipenv PR is up now and the pipenv.lock file already is pinning to 2.6.1 for pygments so need to bother with the requirements.txt file.

v01d, 2020-09-02:
Great, I'll make a PR fixing this highlight issue then.

xiaoxiang781216, 2024-10-07:
should be fixed now.

closed 2024-10-07 by xiaoxiang781216