apache/nuttx #1652: arch code shouldn't include sched/sched.h which violate the modular rule

xiaoxiang781216, 2020-08-26:


patacongo, 2020-08-26:
Yes, this is an exception to the rule which has been there since day 1.  The intention is that that these are interfaces that are shared only between arch/ and sched/ and nowhere else.  This is the way that that scoping is enforced.

The alternative would be to move the prototypes into include/nuttx/sched.h, but that then reveals internal interfaces that should not be used outside of arch. So that is a not a very good idea.

I have thought about this before and decided that although the current solution is not clean, it is better than any other solution that I have know of.  We need to have a way to share internal OS definitions with arch/ and no where else.  I think it just needs to be better documented.  I don't think any code change is required.

xiaoxiang781216, 2020-08-27:
If we change the macro to function, the internal detail don't need to expose outside of sched folder.