apache/nuttx #1670: Build error for nucleo-l476rg

falloutkid, 2020-08-30:
I would build for nucleo-l476rg. But it occur following error message. I cannot resolve this error. Would you tell me to resolve this error/

SHA-1:797bf447d1171e46e42e6bee1036c5511f23f1d

```bash
# tools/configure.sh nucleo-l476rg:nsh
  Copy files
  Select CONFIG_HOST_LINUX=y
  Refreshing...
make[1]: Entering directory '/mnt/rust/src/nuttx/incubator-nuttx'
make[2]: Entering directory '/mnt/rust/src/nuttx/incubator-nuttx/boards'
make[2]: Leaving directory '/mnt/rust/src/nuttx/incubator-nuttx/boards'
make[2]: Entering directory '/mnt/rust/src/nuttx/incubator-nuttx/graphics'
make[3]: Entering directory '/mnt/rust/src/nuttx/incubator-nuttx/graphics/nxglib'
make[3]: Leaving directory '/mnt/rust/src/nuttx/incubator-nuttx/graphics/nxglib'
make[3]: Entering directory '/mnt/rust/src/nuttx/incubator-nuttx/graphics/nxglib'
make[3]: Leaving directory '/mnt/rust/src/nuttx/incubator-nuttx/graphics/nxglib'
make[3]: Entering directory '/mnt/rust/src/nuttx/incubator-nuttx/graphics/nxglib'
make[3]: Leaving directory '/mnt/rust/src/nuttx/incubator-nuttx/graphics/nxglib'
make[2]: Leaving directory '/mnt/rust/src/nuttx/incubator-nuttx/graphics'
make[1]: Leaving directory '/mnt/rust/src/nuttx/incubator-nuttx'
LN: include/arch to arch/arm/include
LN: include/arch/board to /mnt/rust/src/nuttx/incubator-nuttx/boards/arm/stm32l4/nucleo-l476rg/include
LN: include/arch/chip to arch/arm/include/stm32l4
LN: arch/arm/src/board to /mnt/rust/src/nuttx/incubator-nuttx/boards/arm/stm32l4/nucleo-l476rg/src
LN: arch/arm/src/chip to arch/arm/src/stm32l4
LN: /mnt/rust/src/nuttx/incubator-nuttx/drivers/platform to /mnt/rust/src/nuttx/incubator-nuttx/drivers/dummy
make[1]: Entering directory '/mnt/rust/src/nuttx/incubator-nuttx/libs/libxx'
make[1]: Nothing to be done for 'dirlinks'.
make[1]: Leaving directory '/mnt/rust/src/nuttx/incubator-nuttx/libs/libxx'
make[1]: Entering directory '/mnt/rust/src/nuttx/incubator-nuttx/boards'
make[1]: Leaving directory '/mnt/rust/src/nuttx/incubator-nuttx/boards'
make[1]: Entering directory '/mnt/rust/src/nuttx/incubator-nuttx/openamp'
make[1]: Nothing to be done for 'dirlinks'.
make[1]: Leaving directory '/mnt/rust/src/nuttx/incubator-nuttx/openamp'
make: the '-C' option requires a non-empty string argument
Usage: make [options] [target] ...
Options:
  -b, -m                      Ignored for compatibility.
  -B, --always-make           Unconditionally make all targets.
  -C DIRECTORY, --directory=DIRECTORY
                              Change to DIRECTORY before doing anything.
  -d                          Print lots of debugging information.
  --debug[=FLAGS]             Print various types of debugging information.
  -e, --environment-overrides
                              Environment variables override makefiles.
  --eval=STRING               Evaluate STRING as a makefile statement.
  -f FILE, --file=FILE, --makefile=FILE
                              Read FILE as a makefile.
  -h, --help                  Print this message and exit.
  -i, --ignore-errors         Ignore errors from recipes.
  -I DIRECTORY, --include-dir=DIRECTORY
                              Search DIRECTORY for included makefiles.
  -j [N], --jobs[=N]          Allow N jobs at once; infinite jobs with no arg.
  -k, --keep-going            Keep going when some targets can't be made.
  -l [N], --load-average[=N], --max-load[=N]
                              Don't start multiple jobs unless load is below N.
  -L, --check-symlink-times   Use the latest mtime between symlinks and target.
  -n, --just-print, --dry-run, --recon
                              Don't actually run any recipe; just print them.
  -o FILE, --old-file=FILE, --assume-old=FILE
                              Consider FILE to be very old and don't remake it.
  -O[TYPE], --output-sync[=TYPE]
                              Synchronize output of parallel jobs by TYPE.
  -p, --print-data-base       Print make's internal database.
  -q, --question              Run no recipe; exit status says if up to date.
  -r, --no-builtin-rules      Disable the built-in implicit rules.
  -R, --no-builtin-variables  Disable the built-in variable settings.
  -s, --silent, --quiet       Don't echo recipes.
  -S, --no-keep-going, --stop
                              Turns off -k.
  -t, --touch                 Touch targets instead of remaking them.
  --trace                     Print tracing information.
  -v, --version               Print the version number of make and exit.
  -w, --print-directory       Print the current directory.
  --no-print-directory        Turn off -w, even if it was turned on implicitly.
  -W FILE, --what-if=FILE, --new-file=FILE, --assume-new=FILE
                              Consider FILE to be infinitely new.
  --warn-undefined-variables  Warn when an undefined variable is referenced.

This program built for x86_64-pc-linux-gnu
Report bugs to <bug-make@gnu.org>
make: *** [tools/Makefile.unix:319: dirlinks] Error 2
ERROR: failed to refresh
```

patacongo, 2020-08-30:
I do not see this problem on the current repository:

    $ tools/configure.sh -c nucleo-l476rg:nsh
     ?? Copy files
     ?? Select CONFIG_HOST_WINDOWS=y
     ?? Select CONFIG_WINDOWS_CYGWIN=y
     ?? Refreshing...
    make[1]: Entering directory
    '/cygdrive/d/Spuda/Documents/projects/nuttx/master/nuttx_fork'
    make[2]: Entering directory
    '/cygdrive/d/Spuda/Documents/projects/nuttx/master/nuttx_fork/boards'
    make[2]: Leaving directory
    '/cygdrive/d/Spuda/Documents/projects/nuttx/master/nuttx_fork/boards'
    ...
    make[1]: Leaving directory
    '/cygdrive/d/Spuda/Documents/projects/nuttx/master/nuttx_fork'
    LN: include/arch to arch/arm/include
    LN: include/arch/board to
    /cygdrive/d/Spuda/Documents/projects/nuttx/master/nuttx_fork/boards/arm/stm32l4/nucleo-l476rg/include
    LN: include/arch/chip to arch/arm/include/stm32l4
    LN: arch/arm/src/board to
    /cygdrive/d/Spuda/Documents/projects/nuttx/master/nuttx_fork/boards/arm/stm32l4/nucleo-l476rg/src
    LN: arch/arm/src/chip to arch/arm/src/stm32l4
    LN:
    /cygdrive/d/Spuda/Documents/projects/nuttx/master/nuttx_fork/drivers/platform
    to
    /cygdrive/d/Spuda/Documents/projects/nuttx/master/nuttx_fork/drivers/dummy
    make[1]: Entering directory
    '/cygdrive/d/Spuda/Documents/projects/nuttx/master/nuttx_fork/libs/libxx'
    make[1]: Nothing to be done for 'dirlinks'.
    make[1]: Leaving directory
    '/cygdrive/d/Spuda/Documents/projects/nuttx/master/nuttx_fork/libs/libxx'
    make[1]: Entering directory
    '/cygdrive/d/Spuda/Documents/projects/nuttx/master/nuttx_fork/boards'
    make[1]: Leaving directory
    '/cygdrive/d/Spuda/Documents/projects/nuttx/master/nuttx_fork/boards'
    make[1]: Entering directory
    '/cygdrive/d/Spuda/Documents/projects/nuttx/master/nuttx_fork/openamp'
    make[1]: Nothing to be done for 'dirlinks'.
    make[1]: Leaving directory
    '/cygdrive/d/Spuda/Documents/projects/nuttx/master/nuttx_fork/openamp'
    make[1]: Nothing to be done for 'dirlinks'.
    make[1]: Leaving directory
    '/cygdrive/d/Spuda/Documents/projects/nuttx/master/nuttx_fork/openamp'
    make[1]: Entering directory
    '/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps'
    make[2]: Entering directory
    '/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/platform'
    LN: platform/board to
    /cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/platform/dummy
    ...
    make[1]: Leaving directory
    '/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps'
    #
    # configuration written to .config
    #

So it appears to have failed on:

    make[1]: Entering directory '/mnt/rust/src/apps'

Some $(MAKE) -C $(CONFIG_APPS_DIR) ... failed probably because the 
CONFIG_APPS_DIR is not valid.?? Check your .config to see.?? You might 
need to use the -a option with configure.sh to let it know where the 
apps/ directory is located.

v01d, 2020-08-30:
Seems the exact bug described in #844

falloutkid, 2020-08-30:
Thank you for replay.
I can resolve it. The CONFIG_APPS_DIR of my `.config` file was empty. I followed your advice and ran the command adding `-a` option. Then, the build was successful!

falloutkid, 2020-08-30:
@v01d @patacongo 
Thank you for your support! This issue is resolved. Then, I close this issue.

closed 2020-08-30 by falloutkid