apache/nuttx #1718: Configure.sh slow on MacOS

btashton, 2020-09-04:
I have been trying to improve our build times, especially on the CI side.  One thing that I have noticed is that on MacOS the configure phase is really slow.  I don't have easy access to a Mac outside of CI, so debugging this is really slow.

Anyone have thoughts on this?  Here is a sample where configure took 61sec.
```
Fri, 04 Sep 2020 19:15:49 GMT ====================================================================================
Fri, 04 Sep 2020 19:15:49 GMT Configuration/Tool: amber/hello
Fri, 04 Sep 2020 19:15:49 GMT ------------------------------------------------------------------------------------
Fri, 04 Sep 2020 19:15:55 GMT   Cleaning...
Fri, 04 Sep 2020 19:15:59 GMT   Configuring...
Fri, 04 Sep 2020 19:17:00 GMT   Building NuttX...
Fri, 04 Sep 2020 19:17:05 GMT   Normalize amber/hello
```

btashton, 2020-09-04:
Looks like the logs are lying here... Raw logs show that is build, I'm not sure why they have the timestamps mixed in the UI.

closed 2020-09-04 by btashton

patacongo, 2020-09-04:
Most of the time in tools.testbuild.sh after the "Configuring..." message is in executing tools/configure.sh.

Most of the time executing/configure.sh is in the execution of tools/sethost.sh

Most of the time in tools/sethost.sh is in the 'make olddefconfig'

Most of the time in 'make olddefconfig' will be in the Makefile's preconfig target.

Since you are using the same version of NuttX, differing only the select defconfig file, the preconfig is the same on all targets but must be performed again because of the 'make distclean' (which is also pretty slow).  To bad there isn't a way to cache the apps/Kconfig files to avoid the prefconfig.

Ouss4, 2020-09-04:
>  To bad there isn't a way to cache the apps/Kconfig files to avoid the prefconfig.

@patacongo I think we can do that with a define we pass to distclean, maybe `make distclean CACHE_KCONFIG=1` and:

``` diff
-preconfig: $(foreach SDIR, $(CONFIGSUBDIRS), $(SDIR)_preconfig)
+.kconfig: $(foreach SDIR, $(CONFIGSUBDIRS), $(SDIR)_preconfig)
 ifneq ($(MENUDESC),)
        $(Q) $(MKKCONFIG) -m $(MENUDESC)
        $(Q) touch .kconfig
 endif

+preconfig: .kconfig
+
 clean: $(foreach SDIR, $(CLEANSUBDIRS), $(SDIR)_clean)

 distclean: $(foreach SDIR, $(CLEANSUBDIRS), $(SDIR)_distclean)
 ifneq ($(MENUDESC),)
+ifeq ($(CACHE_KCONFIG),)
        $(call DELFILE, Kconfig)
        $(call DELFILE, .kconfig)
 endif
+endif
```
 But how do we clean when we finish?  The last distclean shouldn't set the cache_kconfig.