apache/nuttx #1720: Misc documentation issues

btashton, 2020-09-05:
Here are some minor documentation issues

- [ ] Bad references in the Root Directory and Environment Variables sections here https://nuttx.apache.org/docs/latest/quickstart/build_and_make.html
[image]

- [ ] Bad escape in "Install a Cross-Compiler Toolchain"  https://nuttx.apache.org/docs/latest/quickstart/quickstart.html
```
``/opt/gcc`
```

- [ ] Bad heading "#. Download Apache NuttX " https://nuttx.apache.org/docs/latest/quickstart/quickstart.html

- [ ] Number wrong (1) for all items https://nuttx.apache.org/docs/latest/quickstart/install.html#install-required-tools-all-platforms
- [ ] We list gperf as required for kconfig-frontends, but then install it afterwards.  Are we sure it is required? https://nuttx.apache.org/docs/latest/quickstart/install.html#install-required-tools-all-platforms

- [ ]  Gnu should be GNU
```
guides/drivers.rst:83:    * `GDB — the Gnu Debugger <https://www.gnu.org/software/gdb/>`_
quickstart/quickstart.rst:18:   `ARM Embedded Gnu Toolchain <https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm>`_
quickstart/debugging.rst:9:debugging using the Gnu Debugger (gdb).
quickstart/install.rst:171:`ARM Embedded Gnu Toolchain <https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm>`_
quickstart/running.rst:25:#. Launch the Gnu Debugger
quickstart/running.rst:28:   ARM Embedded Gnu Toolchain we downloaded in the Install step.
```

- [ ] Missing screenshot image https://nuttx.apache.org/docs/latest/quickstart
/configuring.html#configuring
[image]

- [ ] We suggest editing `.config` which we never should do https://nuttx.apache.org/docs/latest/quickstart/debugging.html#debug-logging

adamfeuer, 2020-09-05:
Thanks @btashton, I'm working on a PR to fix these issues.

adamfeuer, 2020-09-05:
@btashton I don't really know much about whether gperf is really needed, we should investigate that; for now I moved the instruction before the kconfig-frontends install.

adamfeuer, 2020-09-05:
@btashton https://github.com/apache/incubator-nuttx/pull/1722 should fix all these issues. I also fixed the section numbering in quickstart so all the instructions should be correct now.

closed 2020-09-06 by Ouss4