apache/nuttx #1733: tickless mode: allow scheduling timer to stop when not needed

v01d, 2020-09-08:
When NuttX is configure in tickless mode, the scheduling timer will continue to run even when there is a single task and no timeouts are pending. This means that unless the processor goes to sleep (thus disabling the timer), in some cases this interrupt can unnecessarily burden de processor which may be important in very low power applications.

This scheduling timer is currently triggered due to a "hack" which is documented as:

<pre>
/* In the original design, it was planned that nxsched_reassess_timer() be
 * called whenever there was a change at the head of the ready-to-run
 * list.  That call was intended to establish a new time-slice or to
 * stop an old time-slice timer.  However, it turns out that that
 * solution is too fragile:  The system is too vulnerable at the time
 * that the ready-to-run list is modified in order to muck with timers.
 *
 * The kludge/work-around is simple to keep the timer running all of the
 * time with an interval of no more than the timeslice interval.  If we
 * do this, then there is really no need to do anything when on context
 * switches.
 */

#define KEEP_ALIVE_HACK 1
</pre>

Some extra explanation from Greg:

>     Use of almost all internal OS interfaces require that the OS data structures always be in the correct state. During a context switch that internal state for one task is torn down and the new state for the news task is set up. In between those times, the OS is in an indeterminate state and the use of any OS internal interface (such as the timer interface) may not be reliable.
> 
>     I don't recall the specific issue that I faced when I wrote that comment, but I do remember that there was no simple work-around. It is, however, certainly worth revisiting if someone is motivated to really dig into the guts of the OS.

patacongo, 2020-09-08:
This only applies to round robin scheduling, correct?  Changing this could be quite complex in the SMP case.

v01d, 2020-09-08:
I have really no idea, my understanding of this part of NuttX is very limited.

patacongo, 2020-09-08:
> This only applies to round robin scheduling, correct? Changing this could be quite complex in the SMP case.
> 
> Is it a work around to (1) make sure that all tasks have a unique priority, and (2) disable round robin scheduling?

Yes, KEEP_ALIVE_HACK only applies if round robin or sporadic scheduling is enabled.  See sched/sched/sched_timerexpiration.c

This issue is, as I recall, not that the timer is needed all of the time.  The issue is that the timer cannot be started when a a new round robin task is started.  The work around to avoid having to start the time is just to never stop it.

So the only real fix would be to improve the logic that starts the timer when a new round robin (or sporadic) task is started.

patacongo, 2020-09-08:
> 
> 
> I have really no idea, my understanding of this part of NuttX is very limited.

And my memory is very limited.  People think that because I wrote most of this code (at least originally), I should understand it all.  That KEEP_ALIVE_HACK was added way back in 2014 with this commit:

    commit 6546fa39c7a292e919d1ea396d67bea8242f2c4b
    Author: Gregory Nutt <gnutt@nuttx.org>
    Date:   Tue Aug 12 11:12:00 2014 -0600
    
        Tickless Stuff:  Back out the risky timer operations when the ready-to-run list is modified. 
        That is unsafe.  An ugly workaround is just to keep an interval timer going all of the time
        with a minimum duration equal to the timeslice interval.

That was after some attempts to use the timer interfaces in that context.  That commit is a good reference because it shows some of the non-viable code that was removed.  Commit 0d71260bf2d which added Sporadic support brought that KEEP_ALIVE_HACK logic to its current state.

v01d, 2020-11-10:
hi @patacongo I'm at a point where I would like to try to deal with the recurring interrupt of the RR scheduler in tickless mode. While the above discussed issue applies to any situation when no ready tasks are available (and thus the scheduler interrupts could be stopped until one becomes ready), I'm interested in the particular case of not having these interrupts fire while system is sleeping (entered with WFI). I'm thinking that since NuttX is not running in this case, I should be able to disable the scheduler altogether, and re-enable it when waking up. I imagine this should involve removing the periodic alarm and being reset when scheduler starts again, right? Is this implemented?
I found `nxsched_suspend_scheduler` but it does not seem to do anything for the RR case. Maybe this is not really what I think it is.
@xiaoxiang781216 do you have any insight into this? I think you did a lot of SMP stuff and imagine you understand the scheduler much better than me.

v01d, 2020-11-10:
Note that I'm not suggesting to deal with the original issue directly (which would be the most correct way to do it) mostly because I don't feel confident about doing it (wouldn't even know where to start)

v01d, 2020-11-10:
Looking at `nxsched_reasses_timer` makes me think I could do something like:
```
nxsched_cancel_timer()
wfi
nxsched_timer_start(MSEC2TICK(CONFIG_RR_INTERVAL));
```
Does this make sense?

closed 2023-12-17 by patacongo