apache/nuttx #1745: Build system fail to catch the error sometime

xiaoxiang781216, 2020-09-10:
Here is an example:
https://github.com/apache/incubator-nuttx/pull/1693/checks?check_run_id=1085270278
You can see that there are some compiler errors in arch/sim/src/sim/up_tapdev.c, but the precheck consider build pass the test.
Now, the typo error is merged into the mainline. The strange thing happen: some build consider it fail:
https://github.com/apache/incubator-nuttx-apps/pull/379/checks?check_run_id=1094370101
some build consider it pass:
https://github.com/apache/incubator-nuttx/pull/1742/checks?check_run_id=1094753731
@btashton and @liuguo09

btashton, 2020-09-10:
Hmm also strange because you can see the build on master that passed for this. I'm not sure how this PR would have passed both prior to merge and and then also on the build on master branch triggered by the merge. All the builds keep an artifact that is the source repos used for the build so I can take a look in the morning.

liuguo09, 2020-09-10:
@btashton  I'm not very sure. But it seems it's caused by ccache enabled in incubator-nuttx, while incubator-nuttx-apps doesn't enable ccache now,  it works rightly.

btashton, 2020-09-10:
Cache key is specific to the run so it's not that. Also with the new cache v2 I have noticed the source bundle is usually not there so it grabs it from the artifact instead. Something else funny is going on.

btashton, 2020-09-10:
Ahhh ccache not cache. You might be right. I wonder why ccache is getting the false hits.

btashton, 2020-09-10:
I pulled the source bundle that it used and it failed to build as expected.  That really leaves ccache as the only option but I could not reproduce it by clearing my cache and using an earlier build and then building the bad build.  I do notice that we have direct mode enabled which we should not but the hash of `up_tapdev.c` must have changed so I don't see how this could have been missed.

closed 2020-09-11 by xiaoxiang781216