apache/nuttx #1769: All arch need colorize the idle thread stack correctly

xiaoxiang781216, 2020-09-12:
Copy from TODO:
               2. Another problem is colorizing that stack to use with
                 stack usage monitoring logic.  There is logic in some
                 start functions to do this in a function called go_nx_start.
                 It is available in these architectures:

                 ./arm/src/efm32/efm32_start.c:static void go_nx_start(void *pv, unsigned int nbytes)
                 ./arm/src/kinetis/kinetis_start.c:static void go_nx_start(void *pv, unsigned int nbytes)
                 ./arm/src/sam34/sam_start.c:static void go_nx_start(void *pv, unsigned int nbytes)
                 ./arm/src/samv7/sam_start.c:static void go_nx_start(void *pv, unsigned int nbytes)
                 ./arm/src/stm32/stm32_start.c:static void go_nx_start(void *pv, unsigned int nbytes)
                 ./arm/src/stm32f7/stm32_start.c:static void go_nx_start(void *pv, unsigned int nbytes)
                 ./arm/src/stm32l4/stm32l4_start.c:static void go_nx_start(void *pv, unsigned int nbytes)
                 ./arm/src/tms570/tms570_boot.c:static void go_nx_start(void *pv, unsigned int nbytes)
                 ./arm/src/xmc4/xmc4_start.c:static void go_nx_start(void *pv, unsigned int nbytes)

                 But no others.

xiaoxiang781216, 2020-09-12:
@davids5 please complete this work item with your approach. Thanks!

davids5, 2020-09-14:
@xiaoxiang781216 I want to be sure:

1)  I understand what you atr asking me to do here.
2) both understand why  CONFIG_STACK_COLORATION was added in the first place. in 2013-11-01 10:16:5 b8085906b9db4cbb05977ed96105a473a0836eb1 

Here is my understanding:

1) The feature to color the stack is a **debugging feature**. It is enabled with CONFIG_STACK_COLORATION
2) If CONFIG_STACK_COLORATION is enabled the <ARCH>_checkstack will be included in the build.
3) hardfaults will dump stack usage
4) proc file system will provide the stack used amount.
5) Debug application can used this information to monitor the stack penetration.

What do you mean by "your approach." ?

Are you asking me to add the assembly based coloring to all the archs and chips defined in nuttx?
 
If this is because of the TLS changes:  https://github.com/apache/incubator-nuttx/issues/986 

Do we now have a debugging ONLY feature that the TLS is dependent on?

xiaoxiang781216, 2020-09-14:
The method you talk at PR 1369:
Here is how I was looking at it. 1) I Draw a picture of the stack frame
[ ]
[ ]
[ ]
[ ]<---At the function
[ marker1]
[ ]
[ what is here or not does not matter ]
[ ]
[ marker2 ]
[ ] <- the allocation

if the compiler messes with you and it looks like this, does not matter.

[ marker2]
[ ]
[ what is here or not does not matter ]
[ ]
[ marker1 ]
[ ] <- the allocation

safe = min of (&marker2, &marker1) word gets you the lowest point on the stack you can write to,

davids5, 2020-09-14:
Here you go https://github.com/xiaoxiang781216/incubator-nuttx/pull/5

xiaoxiang781216, 2020-09-15:
@davids5 please create a PR to fix all arch, thanks.

davids5, 2020-09-15:
> Since you prefer this method, I will drop my colorize related change and wait your implementation.

Now that the separation of the critical bug fix and debug feature is done in #1369 . 

My preferred method is to have the _start code do the coloring. For the simple reason that the benefit of the coloring is needed to debug problems well before the IDLE task is created. Things like device driver init etc.

There are also complex factors to consider the value of it after init. This is because NuttX does not support nested interrupts, and on an ARM machine that uses a separate Interrupt stack is the full IDLE stack size after system init is wasted of memory.

closed 2021-06-10 by davids5