apache/nuttx #1804: Assertion failed at file:mqueue/mq_sndinternal.c line: 420

anchao, 2020-09-16:
We already have the corresponding fix on https://github.com/apache/incubator-nuttx/pull/1585
but which still pending on review, @patacongo  could you please kindly fix this issue if you do not like our solution?

Enable config:

```
CONFIG_DEBUG_ASSERTIONS=y
CONFIG_DEBUG_FEATURES=y
```

Example code as below:

```
#include <stdio.h>
#include <fcntl.h>
#include <unistd.h>
#include <mqueue.h>
#include <pthread.h>
#include <errno.h>

static mqd_t qid;

static void *recv_thread(void *arg)
{
  struct mq_attr attr;
  int deadcounter;
  int ret;

  mq_getattr(qid, &attr);

  while (1) {
    ret = mq_receive(qid, (void *)&deadcounter, attr.mq_msgsize, NULL);
    if (ret > 0) {
      printf("deadcounter: %d\n", deadcounter);
    }
  }

  return NULL;
}

int main(int argc, char *argv[])
{
  int counter = 3;
  pthread_t pid;
  int ret;

  qid = mq_open("/test", O_RDWR | O_CREAT | O_EXCL, 0666, NULL);

  if (qid < 0) {
    if (errno == EEXIST) {
      mq_unlink("/test");
      qid = mq_open("/test", O_RDWR | O_CREAT, 0666, NULL);
    } else
      return -1;
  }

  ret = pthread_create(&pid, NULL, recv_thread, NULL);
  if (ret < 0)
    goto bail;

  while (mq_send(qid, (void *)&counter, sizeof(int), 1) >= 0) {
    sleep(1);
    if (counter-- == 0)
      pthread_cancel(pid);
  }

bail:
  mq_close(qid);

  return 0;
}

```

Assertion:

```
NuttShell (NSH) NuttX-9.1.0
MOTD: username=admin password=Administrator
nsh> hello
deadcounter: 3
deadcounter: 2
deadcounter: 1
deadcounter: 0
Assertion failed at file:mqueue/mq_sndinternal.c line: 420
```

xiaoxiang781216, 2020-09-16:
Regressed by:
```
commit e24f2814015c6dcd9fc67edcd399ccbaf41c3669
Author: Gregory Nutt <gnutt@nuttx.org>
Date:   Sun Nov 20 07:57:18 2016 -0600

    This commit adds a new internal interfaces and fixes a problem with three APIs in the SMP configuration.  The new internal interface is sched_cpu_p
ause(tcb).  This function will pause a CPU if the task associated with 'tcb' is running on that CPU.  This allows a different CPU to modify that OS dat
a stuctures associated with the CPU.  When the other CPU is resumed, those modifications can safely take place.
    
    The three fixes are to handle cases in the SMP configuration where one CPU does need to make modifications to TCB and data structures on a task tha
t could be running running on another CPU.  Those three cases are task_delete(), task_restart(), and execution of signal handles.  In all three cases t
he solutions is basically the same:  (1) Call sched_cpu_pause(tcb) to pause the CPU on which the task is running, (2) perform the necessary operations,
 then (3) call up_cpu_resume() to restart the paused CPU.
```

btashton, 2020-10-27:
@xiaoxiang781216 what do you want to do with this?  We had marked it as a blocker for the 10.0.0 release?

xiaoxiang781216, 2020-10-27:
The fix is right since tcb is destroyed anyway, nobody will touch tcb_s::task_state anymore.

closed 2020-10-28 by xiaoxiang781216