apache/nuttx #1831: Echo stucks in initial script if USB console is used

lukegluke, 2020-09-17:
If usb CDC ACM is used as console and echo is called in init script it will hangup in nsh_output. Other apps that called from init script and print something to console are handled correctly - I can see their output in dmesg after connecting by usb console.

CONFIG_CDCACM_CONSOLE=y
CONFIG_NO_SERIAL_CONSOLE=y
no CONFIG_DEV_CONSOLE

NuttX 8.2
(in fact project based on px4, but I don't think it is relevant)

v01d, 2020-11-04:
Possible duplicate of #2203

lukegluke, 2020-11-04:
> Possible duplicate of #2203

I don't think so. NuttX 8.2 did preallocate streams, so #2203 was not the issue for this version.
But digging with #2203 showed that I possibly had misconfig and CONFIG_DEV_CONSOLE was set by default, so maybe it is the reason. I will try to check latter and also with NuttX 10 when #2203 will be fixed.

lukegluke, 2020-11-13:
I've tested with latest master that includes fix #2263 for #2203. No problems.
I'm still not sure about NuttX 8.2. Don't have much time to test, so I'm closing this issue anyway.

closed 2020-11-13 by lukegluke