apache/nuttx #1838: Do we still need IFF_DOWN?

masayuki2009, 2020-09-18:
I noticed that IFF_DOWN is used both nuttx and apps to control network interface.
In my understanding, IFF_UP is enough and IFF_DOWN is rather redundant.

So my question is that do we still need IFF_DOWN?

xiaoxiang781216, 2020-09-19:
Same question for me, should we remove one? or change the coe from:
```
#define IFF_DOWN           (1 << 0) /* Interface is down */
#define IFF_UP             (1 << 1) /* Interface is up */
```
to:
```
#define IFF_DOWN           (0 << 1) /* Interface is down */
#define IFF_UP             (1 << 1) /* Interface is up */
```

masayuki2009, 2020-09-23:
> Same question for me, should we remove one? or change the code from:
> 
> ```
> #define IFF_DOWN           (1 << 0) /* Interface is down */
> #define IFF_UP             (1 << 1) /* Interface is up */
> ```
> 
> to:
> 
> ```
> #define IFF_DOWN           (0 << 1) /* Interface is down */
> #define IFF_UP             (1 << 1) /* Interface is up */
> ```

@xiaoxiang781216 

I think that the above change does not work.
So I think we should replace all IFF_DOWN related code both in nuttx and in apps with IFF_UP.

closed 2024-10-08 by GUIDINGLI