apache/nuttx #1844: BEST SUPPORTED BOARD + NUTTX VERSION

TA1DB, 2020-09-18:
Hi,

I'm new to Nuttx. I want to make experiments with a board having (or possible to add easily) a display, keypad and network connection; by this way I want to learn nuttx, develop applications on it. 

I may also contribute to Nuttx itself in time, but I need to learn it well first. 

I don't want to deal with Nuttx's issues right now because I don't have enough experience to resolve these issues anyway and I don't have enough time either!

First I started with stm32f746g-discovery board, I encountered some problems, I wrote them in [this issue]( https://github.com/apache/incubator-nuttx/issues/1643) 

Afterwards considering it is more supported I got a stm32f429i-disco (disc1) board, again I encountered a some other problems, but when I used Nuttx version 9.0.0 the problems decreased, still I can't say the situation is satisfactory.

Could someone tell me "I use this card together with this Nuttx version, I recommend these"?

Since I couldn't find any other active forum to ask this question, I had to make my request here, I apologize if I was wrong. 

Thank you.

btashton, 2020-09-18:
@TA1DB We are doing some serious work trying to improve the state of the documentation around getting started.   So I would love to help get you going and maybe we can build that out a bit better.

This is actually why I am trying to build out complete support for this "featherwing"  right now https://www.tindie.com/products/arturo182/keyboard-featherwing-qwerty-keyboard-26-lcd/ which could then provide a lot of support for playing with NuttX on some of the lower cost platforms with touch screen keyboard etc...

The **stm32f746g-discovery** looks like an interesting board with the integrated LCD and network support.  I just checked about ordering on, but it looks like everyone is out of stock until late November so I am not sure how much help I would be able to provide.

The **stm32f429i-disco** is a rather old board and should be fairly well supported although some board support code may have aged.  I do have one of the original ones and would be happy to help get you started with it and could verify that some of the existing configuration are working correctly.  What configuration specifically did you try with it?  I would start with `stm32f429i-disc:nsh` and then since you were interested in the display `stm32f429i-disc:lvgl`.  There are some notes on that here:

**boards/arm/stm32/stm32f429i-disco/README.txt**:
```
  lgvl:
  ----

    STM32F429I-DISCO LittlevGL demo example.

    The ltdc is initialized during boot up.  Interaction with NSH is via
    the serial console at 115200 8N1 baud.  From the nsh command line
    execute the lvgldemo example:

      nsh> lvgldemo

    The test will execute the calibration process and then run the
    LittlevGL demo project.
```

I think going forward we were thinking about doing some focused tutorials on the STM32 Nucleo-144 platform which I personally have the H745 and F767 variants.  Note that these do not have displays but do have Ethernet.  Connecting and SPI based ILI9341 TFT display via SPI is fairly well supported and I just verified it with the featherwing I mentioned.

Happy to help you get going on this and try to improve the documentation at the same time.

TA1DB, 2020-09-19:
@btashton 
Thank you very much for your kind and helpful response.

I am aware that you are working hard to develop the documentation; this is very good. I also follow your work on the "featherwing" (https://www.linkedin.com/feed/update/urn:li:activity:6712105195817447424).

I would appreciate if you resolve the issues about the stm32f746g-discovery board because it is a very advanced board; In addition to LCD capacitive touch screen and network connection there are many other hardware interfaces (sdcard, microphones, audio output, camera interface, etc.) in addition to microprocessor's own advanced capabilities. I think this board is a very good platform for Nuttx developments and therefore should be well supported.

I had no startup problems with stm32f429i-disco. All configurations including lvgldemo compiled, loaded and worked without any problems. However, when I tried to write my own application using Nuttx as a library with "make export" in version 9.1.0, I encountered errors during compiling. When I went back to version 9.0.0, this time I could pass the compilation phase but got the following errors in the link phase _(tried on both  Ubuntu 16.04 and macOS High Sierra installations, same results!)_ :

`apple-MacBook-Pro:nuttx-export-0.0.0 murat$ arm-none-eabi-gcc -c -fno-builtin -Wall -Wstrict-prototypes -Wshadow -Wundef -g -mcpu=cortex-m4 -mthumb -mfloat-abi=soft -I. -isystem ./include -pipe -I "./include" -I "./arch/chip" hello.c -o hello.o
apple-MacBook-Pro:nuttx-export-0.0.0 murat$ arm-none-eabi-ld --entry=__start -nostartfiles -nodefaultlibs -T./scripts/ld.script -L./libs hello.o -o nuttx.elf --start-group -lc -larch -lbinfmt -lboard -lboards -ldrivers -lfs -lmm -lsched -lxx "/usr/lib/gcc/arm-none-eabi/9.2.1/thumb/v7e-m/nofp/libgcc.a" --end-group
arm-none-eabi-ld: cannot find -lc
arm-none-eabi-ld: cannot find -larch
arm-none-eabi-ld: cannot find -lbinfmt
arm-none-eabi-ld: cannot find -lboard
arm-none-eabi-ld: cannot find -lboards
arm-none-eabi-ld: cannot find -ldrivers
arm-none-eabi-ld: cannot find -lfs
arm-none-eabi-ld: cannot find -lmm
arm-none-eabi-ld: cannot find -lsched
arm-none-eabi-ld: cannot find -lxx
arm-none-eabi-ld: cannot find /usr/lib/gcc/arm-none-eabi/9.2.1/thumb/v7e-m/nofp/libgcc.a: No such file or directory` 

I have to overcome this problem because what I really want to do is to be able to write my own applications. 

On the other hand I got my answer to my question and therefore I am closing the issue. Thanks again.

closed 2020-09-19 by TA1DB

btashton, 2020-09-19:
I'm glad that worked for you, I'll keep and eye on that other board for when it comes back in stock. Let's figured out what is up with `make export` I personally don't use it, instead I leverage the external apps functionally.

@v01d do you have thoughts on the export issue?

v01d, 2020-09-19:
> I'm glad that worked for you, I'll keep and eye on that other board for when it comes back in stock. Let's figured out what is up with `make export` I personally don't use it, instead I leverage the external apps functionally.
> 
> @v01d do you have thoughts on the export issue?

Same here, I use the external apps. @TA1DB maybe you can try this approach? It is simpler. You just add your application in apps/external (it can be a symlink to your app somewhere else).

v01d, 2020-09-19:
I think this approach should be documented in our documentation (both for a single app and multiple apps).

btashton, 2020-09-19:
@TA1DB @v01d is speaking about option 3 [here](https://cwiki.apache.org/confluence/display/NUTTX/Building+NuttX+with+Applications+Outside+of+the+Source+Tree)

@v01d @adamfeuer  This doc is probably very important to move from the wiki and into the documentation and we should document the export function.

v01d, 2020-09-19:
@btashton I've been meaning to ask: are the wiki pages accessible in original format? Are they markdown? It would be much easier to port them to sphinx if so

btashton, 2020-09-19:
> @btashton I've been meaning to ask: are the wiki pages accessible in original format? Are they markdown? It would be much easier to port them to sphinx if so

Yes but many of them have been modified and I did a lot of work fixing syntax issues during the conversion. I know I can export the space, let me see what I can do on that front.  Let's pick it up in the roadmap issue.

Ouss4, 2020-09-19:
@v01d @btashton
There is also a paragraph in apps/' README file that explains how to add an external application.
(Sorry I'm on my phone and internet is so bad, I'd have given the url directly otherwise)

Ouss4, 2020-09-19:
Btw if @xiaoxiang781216 or @anchao see this they can provide some insights on the state of export.  They've been submitting some PRs regarding that.

anchao, 2020-09-20:
@Ouss4 , Thanks for reminding!

Hi @TA1DB ,

Welcome!

I tried "make export" compiling locally on "Ubuntu 16.04", it seems there is no problem and everything is working as expected, I suspect there may be something wrong with system compatibility, could you please share the detailed compilation process?

The compilation process on my desktop as below:

Target board: stm32f746g-disco
```

chao@archer:~/code/incubator-nuttx$ ./tools/configure.sh stm32f746g-disco/lvgl
...
chao@archer:~/code/incubator-nuttx$ make
...
chao@archer:~/code/incubator-nuttx$ make export
...
chao@archer:~/code/incubator-nuttx$ ls -l nuttx-export-9.1.0.zip 
-rw-rw-r-- 1 chao chao 5824499 9  20 09:16 nuttx-export-9.1.0.zip
...
chao@archer:~/code/incubator-nuttx$ cd ../apps/
chao@archer:~/code/apps$ ./tools/mkimport.sh -x ../incubator-nuttx/nuttx-export-9.1.0.zip
...
chao@archer:~/code/apps$ make import V=1
...
echo "LD: nuttx"
LD: nuttx
/home/chao/code/m4/prebuilts/gcc/linux/arm/bin/arm-none-eabi-ld --entry=__start -nostartfiles -nodefaultlibs -g -L /home/chao/code/apps/import/libs -L "/home/chao/code/m4/prebuilts/gcc/linux/arm/bin/../lib/gcc/arm-none-eabi/9.2.1/thumb/v7e-m/nofp"  \
  -L/home/chao/code/apps/import/scripts -Tflash.ld \
  -o nuttx   /home/chao/code/apps/builtin/exec_builtin.home.chao.code.apps.builtin.o /home/chao/code/apps/builtin/builtin_list.home.chao.code.apps.builtin.o  --start-group \
  /home/chao/code/apps/libapps.a -lsched -ldrivers -lboards -lc -lmm -larch -lxx -lapps -lfs -lbinfmt -lboard  -lgcc -lgcc -lm --end-group
echo "CP: nuttx.hex"
CP: nuttx.hex
/home/chao/code/m4/prebuilts/gcc/linux/arm/bin/arm-none-eabi-objcopy  -O ihex nuttx nuttx.hex
echo "CP: nuttx.bin"
CP: nuttx.bin
/home/chao/code/m4/prebuilts/gcc/linux/arm/bin/arm-none-eabi-objcopy  -O binary nuttx nuttx.bin
make[1]: Leaving directory '/home/chao/code/apps'
```

TA1DB, 2020-09-20:
Thank you very much to everyone who is trying to help me by answering my question here. 
On this occasion I would also like to thank to @acassis for kindly answering my questions in LinkedIn messaging.

@anchao I just applied exactly same process [here in @acassis 's blog Using NuttX as library once again ]( https://acassis.wordpress.com/2020/06/28/using-nuttx-as-library-once-again/ )

Everything went as expected until I enter this command:
`$ arm-none-eabi-ld --entry=__start -nostartfiles -nodefaultlibs -T./scripts/ld.script -L./libs hello.o -o nuttx.elf --start-group -lc -larch -lbinfmt -lboard -lboards -ldrivers -lfs -lmm -lsched -lxx "/usr/lib/gcc/arm-none-eabi/9.2.1/thumb/v7e-m/nofp/libgcc.a" --end-group`

then I got the following:
`.apple-MacBook-Pro:hello murat$ cd nuttx-export-9.0.0
apple-MacBook-Pro:nuttx-export-9.0.0 murat$ arm-none-eabi-gcc -c -fno-builtin -Wall -Wstrict-prototypes -Wshadow -Wundef -g -mcpu=cortex-m4 -mthumb -mfloat-abi=soft -I. -isystem ./include   -pipe -I "./include" -I "./arch/chip" hello.c -o  hello.o
apple-MacBook-Pro:nuttx-export-9.0.0 murat$ arm-none-eabi-ld --entry=__start -nostartfiles -nodefaultlibs -T./scripts/ld.script -L./libs hello.o -o nuttx.elf --start-group -lc -larch -lbinfmt -lboard -lboards -ldrivers -lfs -lmm -lsched -lxx "/usr/lib/gcc/arm-none-eabi/9.2.1/thumb/v7e-m/nofp/libgcc.a" --end-group
arm-none-eabi-ld: cannot find -lc
arm-none-eabi-ld: cannot find -larch
arm-none-eabi-ld: cannot find -lbinfmt
arm-none-eabi-ld: cannot find -lboard
arm-none-eabi-ld: cannot find -lboards
arm-none-eabi-ld: cannot find -ldrivers
arm-none-eabi-ld: cannot find -lfs
arm-none-eabi-ld: cannot find -lmm
arm-none-eabi-ld: cannot find -lsched
arm-none-eabi-ld: cannot find -lxx
arm-none-eabi-ld: cannot find /usr/lib/gcc/arm-none-eabi/9.2.1/thumb/v7e-m/nofp/libgcc.a: No such file or directory
`
Note: with Nuttx version 9.1.0 the first command line (arm-none-eabi-gcc -c ... ) didn't work either.

TA1DB, 2020-09-20:
> @TA1DB @v01d is speaking about option 3 [here](https://cwiki.apache.org/confluence/display/NUTTX/Building+NuttX+with+Applications+Outside+of+the+Source+Tree)
> 
I didn't know about this 3rd option, thanks for informing, I will look and try.

anchao, 2020-09-20:
Hi @TAIDB ,

The latest version simplifies the link phase and will not require such a complex command line in the future,

Could you please replace :

`$ arm-none-eabi-ld --entry=__start -nostartfiles -nodefaultlibs -T./scripts/ld.script -L./libs hello.o -o nuttx.elf --start-group -lc -larch -lbinfmt -lboard -lboards -ldrivers -lfs -lmm -lsched -lxx "/usr/lib/gcc/arm-none-eabi/9.2.1/thumb/v7e-m/nofp/libgcc.a" --end-group`

instead of :

`$ make import`

anchao, 2020-09-20:
I @TA1DB 

Sorry, I misread the command line, please ignore my last comment,  I will try again

anchao, 2020-09-20:
Hi @TA1DB

I tried the tutorial in the Mr. Acassis 's blog, linker command works as expected,  

```
chao@archer:~/nuttx-export-9.1.0$ arm-none-eabi-ld --entry=__start -nostartfiles -nodefaultlibs -T./scripts/flash.ld -L./libs test.o -o nuttx.elf --start-group -lc -larch -lbinfmt -lboard -lboards -ldrivers -lfs -lmm -lsched -lxx -lgcc --end-group
chao@archer:~/nuttx-export-9.1.0$ ls
arch  include  libs  nuttx.elf  registry  scripts  startup  System.map  test  test.c  test.o  tools
```

The error you provided seems to be that the ld path does not take effect,  It is better to confirm whether the libs/ directory exist or these libraries are actually in the libs/ directory.

TA1DB, 2020-09-21:
Hi @anchao 
Yes there are libs/ directories: ~nuttxspace/nuttx/libs/ and ~hello/nuttx-export-9.0.0/libs , however ~nuttxspace/nuttx/libs/ has only libc.a in libc folder, and ~hello/nuttx-export-9.0.0/libs/ has libapps.a and libnuttx.a

Regarding unfound files; Apart from the libboard.a file, all of these "unfound files" are only in the nuttxspace/nuttx/staging/ folder. There is also a libapps.a file in the nuttxspace/nuttx/staging/ folder. On the other hand, libboard.a is only in the folder nuttxspace/nuttx/boards/arm/stm32/stm32f429i-disco/src/.

TA1DB, 2020-09-23:
Any comment?