apache/nuttx #1870: Cygwin Build Is Broken Again

patacongo, 2020-09-22:
$ make -j distclean
    $ tools/configure.sh -c stm32f4discovery:nsh
    $ make
    ...
    CC:  readline.c
    CC:  readline_common.c
    make[3]: Leaving directory '/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/system/readline'
    make[3]: Entering directory '/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/builtin'
    /bin/sh: -c: line 0: unexpected EOF while looking for matching `"'
    /bin/sh: -c: line 1: syntax error: unexpected end of file
    make[3]: *** [/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/Application.mk:143: archive] Error 1
    make[3]: Leaving directory '/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/builtin'
    make[3]: Entering directory '/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/examples/hello'
    /bin/sh: -c: line 0: unexpected EOF while looking for matching `"'
    /bin/sh: -c: line 1: syntax error: unexpected end of file
    make[3]: *** [/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/Application.mk:143: archive] Error 1
    make[3]: Leaving directory '/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/examples/hello'
    make[3]: Entering directory '/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/nshlib'
    /bin/sh: -c: line 0: unexpected EOF while looking for matching `"'
    /bin/sh: -c: line 1: syntax error: unexpected end of file
    make[3]: *** [/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/Application.mk:143: archive] Error 1
    make[3]: Leaving directory '/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/nshlib'
    make[3]: Entering directory '/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/platform'
    /bin/sh: -c: line 0: unexpected EOF while looking for matching `"'
    /bin/sh: -c: line 1: syntax error: unexpected end of file
    make[3]: *** [/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/Application.mk:143: archive] Error 1
    make[3]: Leaving directory '/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/platform'
    make[3]: Entering directory '/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/system/nsh'
    /bin/sh: -c: line 0: unexpected EOF while looking for matching `"'
    /bin/sh: -c: line 1: syntax error: unexpected end of file
    make[3]: *** [/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/Application.mk:143: archive] Error 1
    make[3]: Leaving directory '/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/system/nsh'
    make[3]: Entering directory '/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/system/readline'
    /bin/sh: -c: line 0: unexpected EOF while looking for matching `"'
    /bin/sh: -c: line 1: syntax error: unexpected end of file
    make[3]: *** [/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/Application.mk:143: archive] Error 1
    make[3]: Leaving directory '/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/system/readline'
    make[2]: *** [Makefile:99: /cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/libapps.a] Error 2
    make[2]: Leaving directory '/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps'
    make[1]: *** [Makefile:53: all] Error 2
    make[1]: Leaving directory '/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps'
    make: *** [tools/LibTargets.mk:225: /cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/libapps.a] Error 2

patacongo, 2020-09-22:
This is a blocker; we cannot make any release with the Cygwin build broken.

patacongo, 2020-09-22:
With` make V=1` I can see this is in the archive target:

    ...
    make[3]: Leaving directory '/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/system/readline'
    for app in /cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/builtin /cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/examples/hello /cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/nshlib /cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/platform /cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/system/nsh /cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/system/readline; do \
            make -C "${app}" archive ; \
    done
    make[3]: Entering directory '/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/builtin'
    /bin/sh: -c: line 0: unexpected EOF while looking for matching `"'
    /bin/sh: -c: line 1: syntax error: unexpected end of file
    make[3]: *** [/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/Application.mk:143: archive] Error 1
    make[3]: Leaving directory '/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/builtin'   
    make[3]: Entering directory '/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/examples/hello'
    ...

patacongo, 2020-09-22:
Adding some debug output:

    diff --git a/Application.mk b/Application.mk
    index d1dba985..e3e250ac 100644
    --- a/Application.mk
    +++ b/Application.mk
    @@ -140,6 +140,8 @@ $(CXXOBJS): %$(SUFFIX)$(OBJEXT): %$(CXXEXT)
    
     archive:
     ifeq ($(CONFIG_CYGWIN_WINTOOL),y)
    +       $(warning BIN=$(BIN) -> "${shell cygpath -w $(BIN)}")
    +       $(warning OBJS=$(OBJS))
            $(call ARCHIVE_ADD, "${shell cygpath -w $(BIN)}", $(OBJS))
     else
            $(call ARCHIVE_ADD, $(BIN), $(OBJS))

Does not shed any light on this for me:

    make[3]: Entering directory '/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/builtin'
    /cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/Application.mk:143: BIN=/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/libapps.a -> "D:\Spuda\Documents\projects\nuttx\master\apps\libapps.a"
    /cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/Application.mk:144: OBJS=  builtin_list.cygdrive.d.Spuda.Documents.projects.nuttx.master.apps.builtin.o exec_builtin.cygdrive.d.Spuda.Documents.projects.nuttx.master.apps.builtin.o
    /bin/sh: -c: line 0: unexpected EOF while looking for matching `"'
    /bin/sh: -c: line 1: syntax error: unexpected end of file
    make[3]: *** [/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/Application.mk:145: archive] Error 1
    make[3]: Leaving directory '/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/builtin'

v01d, 2020-09-22:
I suspect it may be this line in ARCHIVE_ADD definition:
```
@echo "AR (add): $(notdir $1) $(2)"
```
Maybe the quotes in $1 are nested in the outer ones and bash expects these to be escaped

v01d, 2020-09-22:
I'm guessing that quotes are handled different in cygwin.

patacongo, 2020-09-22:
I am thinking that I should see the AR command if was executed.  Hence, I am thinking the problem my be in the notdir in Config.mk:

    define ARCHIVE_ADD
            @echo "AR (add): $(notdir $1) $(2)"
            $(Q) $(AR) $1 $(2)
    endef

In that case $1 will be a windows path and notdir is possibly failing on that?

Yes, if I just remove the notdir:

    diff --git a/tools/Config.mk b/tools/Config.mk
    index c3eb0303b0..7bfc30efcd 100644
    --- a/tools/Config.mk
    +++ b/tools/Config.mk
    @@ -317,7 +317,7 @@ endef
     #   CONFIG_WINDOWS_NATIVE - Defined for a Windows native build
    
     define ARCHIVE_ADD
    -       @echo "AR (add): $(notdir $1) $(2)"
    +       @echo "AR (add): $(1) $(2)"
            $(Q) $(AR) $1 $(2)
     endef

Then the build succeeds:

    make[3]: Entering directory '/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/builtin'
    /cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/Application.mk:143: BIN=/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/libapps.a -> "D:\Spuda\Documents\projects\nuttx\master\apps\libapps.a"
    /cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/Application.mk:144: OBJS=  builtin_list.cygdrive.d.Spuda.Documents.projects.nuttx.master.apps.builtin.o exec_builtin.cygdrive.d.Spuda.Documents.projects.nuttx.master.apps.builtin.o
    AR (add):  D:SpudaDocumentsprojectsnuttxmasterappslibapps.a builtin_list.cygdrive.d.Spuda.Documents.projects.nuttx.master.apps.builtin.o exec_builtin.cygdrive.d.Spuda.Documents.projects.nuttx.master.apps.builtin.o
    arm-none-eabi-ar rcs  "D:\Spuda\Documents\projects\nuttx\master\apps\libapps.a"    builtin_list.cygdrive.d.Spuda.Documents.projects.nuttx.master.apps.builtin.o exec_builtin.cygdrive.d.Spuda.Documents.projects.nuttx.master.apps.builtin.o
    make[3]: Leaving directory '/cygdrive/d/Spuda/Documents/projects/nuttx/master/apps/builtin'

Although the displayed path is wrong.

@v01d The notdir was added by your commit 34b34e2d451 and this breaks the Cygwin build.  What would you propose?

patacongo, 2020-09-22:
> 
> 
> I'm guessing that quotes are handled different in cygwin.

They shouldn't be different.  Same GNU make, same Bash.  The real difference is the Windows path.  I think that is what is making ARCHIVE_ADD puke.

v01d, 2020-09-22:
So you mean `notdir` does not handle backslashes?

patacongo, 2020-09-22:
> 
> 
> So you mean `notdir` does not handle backslashes?

That is my thought.  I am wondering why it is the full path to the tool?  I used to be only the tool name which is still arm-non-eabi-ar.  When why you add the full path, then you grow backslashes where not should be.

v01d, 2020-09-22:
The notdir applies to the library path, so that it just reads `AR (add): libapps.a`. It is purely cosmetic. You could remove it and let it show the full path or maybe it needs quotes like `$(notdir '$1')` or `$(notdir "$1")`. I'm guessing the backslash is interpreted by bash as an escape.

patacongo, 2020-09-22:
> The notdir applies to the library path, so that it just reads `AR (add): libapps.a`. It is purely cosmetic. You could remove it and let it show the full path or maybe it needs quotes like `$(notdir '$1')` or `$(notdir "$1")`. I'm guessing the backslash is interpreted by bash as an escape.

I don't think I an do that, the echo argument is already enclosed like:  `@echo "AR (create): $(notdir $1) $(2)"`

patacongo, 2020-09-22:
As a general rule, make functions will not accept Windows paths.  In this case, notdir breaks the build.  The last time it was broken, Xiang replaced the Bash dirname (which does handle windows paths) with the make dir command (which does not).

v01d, 2020-09-22:
Ok, if you think notdir is the issue, we can just remove it.

patacongo, 2020-09-22:
See PR #1871

patacongo, 2020-09-22:
> 
> 
> Ok, if you think notdir is the issue, we can just remove it.

yes, notdir with a Windows path is the issue.  I have a PR will submit in a moment which fixes the issue and retains the cosmetic output.

patacongo, 2020-09-22:
@v01d @btashton We really need to get some minimal verification of the Cygwin build into the PR checks.  I know that Brennan has done work on this and my understand is that there is hesitancy because the automated Cygwin build is very slow.  Is that true?

v01d, 2020-09-22:
I was thinking the same, that we should try adding it to CI, maybe just running one config.

btashton, 2020-09-22:
> I was thinking the same, that we should try adding it to CI, maybe just running one config. 

There is my PR. I'll clean it up and add an ARM target with it. Need to reread what you wrote about the toolchain. That's what held it up months ago, but you responded I just need to read.

closed 2020-09-22 by v01d