apache/nuttx #1880: Cygwin directories not being cleaned

patacongo, 2020-09-23:
Under Cygwin, the following directories are not being removed by the distclean target:

    arch/<arch>/src/board
    arch/<arch>/src/chip

I have several old board and and chip directories left over from unsuccessful distclean operations on different architectures.  These two lines in arch/arm/src/Makefile (and other arch Makefiles) are not apparaently not working under Cygwin:

         $(Q) $(DIRUNLINK) $(ARCH_SRC)/board
         $(Q) $(DIRUNLINK) $(ARCH_SRC)/chip

Cygwin supports is own style of symbolic links and these are used EXCEPT for the case where a Windows native toolchain is used such as the popular (https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm).

Those are Window native tools and cannot follow non-standard Cygwin symbolic links so in this case, the tools/dirlinks.sh script does a full directory copy and the corresponding unlink script does a directory removal.

Without knowing more, it looks like the DIRUNLINK is failing to remove the Cygwin symbolic link or the copied directory.

patacongo, 2020-09-23:
The dir unlink script seems to be working fine, however (at least using a relative path):

    $ find . -name chip
    ./arch/arm/include/chip
    ./arch/arm/src/chip
    ./arch/z80/include/chip
    ./include/arch/chip
    
    $ tools/unlink.sh ./arch/z80/include/chip
    
    $ find . -name chip
    ./arch/arm/include/chip
    ./arch/arm/src/chip
    ./include/arch/chip

patacongo, 2020-09-23:
I don't think this bug has been introduced recently.  I have not build the ez80 (under arch/z80) in many months.

patacongo, 2020-09-23:
I hate the close button being associated with the comment window!!!!!!!!!!

patacongo, 2020-09-23:
Hmmm... This is not so simple to replicate.  I don't know the incantation that creates the problem.  My simple tests seems to work.  But the existence of the old, stale, links and the problems described in apache/incubator-nuttx-testing#39 indicate that the problem is real.

patacongo, 2020-09-23:
The stranded directories are NOT the directory copies, but Cygwin symbolic links:

    $ ls -ld arch/arm/include/chip
    lrwxrwxrwx 1 spuda spuda 83 Aug 30 10:09 arch/arm/include/chip -> /cygdrive/d/Spuda/Documents/projects/nuttx/master/nuttx_fork/arch/arm/include/stm32

patacongo, 2020-09-23:
Aha... these bad directories are being created by configure.sh.

    $ make distclean (and verify that there are no chip or board directories)
    $ tools/configure.sh -c stm32f4discovery:nsh
    $ find . -name board
    ./arch/arm/include/board
    ./arch/arm/src/board
    ./boards/arm/stm32/common/board

@v01d @btashton This is a consequence of several changes to the Makefiles recently:

- The olddefconfig target used to invoke the clean_context target to remove all residuals from the configuration
- The symbolic links are (probably) created because CONFIG_CYGWIN_WINTOOL has not been selected yet in the configuration (it was initially a Linux configuration).
- The logic that removes these links was removed from its correct location in the clean_context target to the distclean target.

patacongo, 2020-09-23:
@btashton @hartmannathan Should this be marked a blocker?  I had it tagged that way, but decided not.

hartmannathan, 2020-09-23:
I don't think it needs to be a blocker, but maybe Good First Issue?

btashton, 2020-09-23:
My inclination is for it to be a blocker because I cannot enable windows CI without it and I would like to see that pass prior to the releases.

hartmannathan, 2020-09-23:
> My inclination is for it to be a blocker because I cannot enable windows CI without it and I would like to see that pass prior to the releases.

Okay in that case, make it a blocker. With the Windows CI working, we will find out about breakage much more quickly. That will be a good thing.

patacongo, 2020-09-23:
> I don't think it needs to be a blocker, but maybe Good First Issue?

Way to complex and risky for a first issue.

patacongo, 2020-09-23:
There could also be other issues in the created context.  Some configurations do much more than just setup links.  They may, for example, auto-generate files based on the current configuration.

When a defconfig is included by the Makefile as the .config, it is an incomplete and invalid configuration.  This could potentially cause other more serious configuration problems than just some extra bogus soft links.  It is just not a good behavior.

That is why clean_context was invoked before.  To eliminate this potentially bad configuration.

patacongo, 2020-09-24:
One solution that I would NOT recommend is modify the distclean target to clean more symbolic links.  That is a bandaid and not a really solution to the generic problem that a defconfig is not a valid .config file and can cause problems when used as one in the Makefile.

xiaoxiang781216, 2020-09-24:
@patacongo we are looking this problem.

patacongo, 2020-09-24:
There are multiple causes of the Cygwin build breakage.  This commit is also a factor:  bd656888f26c92e8832f0e76b395a5ece7704530  Although the other cited commits leave the system with a bad context setup, that would not have been noticed prior to this commit which changes the WINTOOL variable to the CYGWIN_WINTOOL Kconfig setting.

That is three commits that break the Cygwin CI build.

patacongo, 2020-09-24:
>  But [b5dc837](https://github.com/apache/incubator-nuttx/commit/b5dc83748c0d28e4310655177419a51412ade07d) is the real culprit because it allows bad context setups into the user build. That could introduce wide-reaching, difficult to analyze problems.

Actually I think I am mistaken about this:  that commit does not do the full context.  It only does dirlinks as far as I can tell.  If that is the case, then things are not so bad.  We just need to separate out a clean_links target to get rid of those links.

liuguo09, 2020-09-24:
@patacongo @btashton @xiaoxiang781216  It seems cygwin build using copydir.sh as DIRLINK in tools/Config.mk. In cygwin case, there are two copies: include/arch/chip and arch/arm/include/chip. And it finally only clean include/arch/chip. So could we switch to use link.sh as DIRLINK for cygwin build to keep one copy firstly? I have tested using link.sh locally and distclean verify ok. As to copydir.sh case, it may need separate out a clean_links target as @patacongo suggests.

patacongo, 2020-09-24:
> @patacongo @btashton @xiaoxiang781216 It seems cygwin build using copydir.sh as DIRLINK in tools/Config.mk. In cygwin case, there are two copies: include/arch/chip and arch/arm/include/chip. And it [...]

The Cygwin build does NOT use copydir.sh unless CONFIG_CYGWIN_WINTOOL is selected.   That is the problem.

- Then Make runs during configuration, it includes .config.  But that is not  complete configuration file.  It is a defconfig file.
- CONFIG_CYGWIN_WINTOOL is the default setting for the selected toolchain so it is NOT selected in the .config file
- As a result, the dirlinks target uses symbolic links to create the links (WRONG)
- After configuring, CONFIG_CYGWIN_WINTOOL will be be defined in the expanded, complete .config file
- The next time Make runs, copydir.sh re-creates all of links with directory copy copies EXCEPT for the two links at arch/arm/include/chip and arch/arm/include/board.  Those are NOT replaced because the directory is copied into include/arch/board and include/arch/chip (the symbolic links are probably copied into include/arch as well (ALSO WRONG)
- When 'make distclean' removes the links, the bad links at include/arch/board and include/arch/chip are not removed because the link target is include/arch/chip and  include/arch/board.  Hence, those two symbolic links are stranded and the CI build fails.

patacongo, 2020-09-24:
This is a consequence of:

- Replace the WINTOOL definition with the CONFIG_CYGWIN_WINTOOL configuration. WINTOOL was always defined previously.
- Removing the olddefconfig dependency on clean_context which removed the context (that removed too much, it is only necessary to remove the bad links)
- And, related, removing the logic that removes the links from the clean_context target.

patacongo, 2020-09-24:
The MYS2 build is similar in that it must copy directories since MSYS2 does not support any kind of symbolic linking.  However, it does not depend on CONFIG_CYGWIN_WINTOOL so should not suffer from this issue.

After you decide on the correct change and before releasing 10.0.0, I will verify the MSYS2 build as well.

liuguo09, 2020-09-27:
Thanks for your explanation, I'm working on it and make PR ASAP once fixed.

liuguo09, 2020-09-27:
@patacongo @btashton I have made a draft PR https://github.com/apache/incubator-nuttx/pull/1901 Could you help to review and verify it?

closed 2020-09-27 by patacongo