apache/nuttx #1918: Simulator Broken Under Cygwin?

patacongo, 2020-09-29:
It appears that the simulator is broken under Cygwin.  It is dying before I even get to main so it is not the usual stack size problem.  Does anyone have any insight into this?

I will mark this as a blocker but that might be a false alarm.

patacongo, 2020-09-29:
@davids5 Okay... I finally verified ASCII mode using examples/ipcfg.  Good to merge from my point of view.

btashton, 2020-09-29:
Greg did you see my comment in the CI PR about this?

patacongo, 2020-09-29:
You are referring to your comment which I replicate here:

I ran the Cygwin build locally and when I try to run the executable it crashes.  Strace shows this, any idea what is going on?  (nettest and nsh had the same result)

    --- Process 1892 created
    --- Process 1892 loaded C:\Windows\System32\ntdll.dll at 00007ffbca2c0000
    --- Process 1892 loaded C:\Windows\System32\kernel32.dll at 00007ffbca070000
    --- Process 1892 loaded C:\Windows\System32\KernelBase.dll at 00007ffbc75e0000
    --- Process 1892 thread 5812 created
    --- Process 1892 loaded C:\Windows\System32\ws2_32.dll at 00007ffbc9ad0000
    --- Process 1892 loaded C:\Windows\System32\rpcrt4.dll at 00007ffbc8370000
    --- Process 1892 loaded C:\Windows\System32\IPHLPAPI.DLL at 00007ffbc6680000
    --- Process 1892 thread 4800 created
    --- Process 1892 thread 9592 created
    --- Process 1892 loaded C:\tools\cygwin\bin\cygwin1.dll at 0000000180040000
    --- Process 1892 loaded C:\tools\cygwin\bin\cygz.dll at 00000003fbf80000
        5       5 [main] nuttx (1892) **********************************************
      369     374 [main] nuttx (1892) Program name: C:\tools\cygwin\home\btashton\sources\nuttx\nuttx.exe (windows pid 1892)
      241     615 [main] nuttx (1892) OS version:   Windows NT-10.0
      277     892 [main] nuttx (1892) **********************************************
    --- Process 1892 loaded C:\Windows\System32\advapi32.dll at 00007ffbc90e0000
    --- Process 1892 loaded C:\Windows\System32\msvcrt.dll at 00007ffbc99d0000
    --- Process 1892 loaded C:\Windows\System32\sechost.dll at 00007ffbc9f30000
    --- Process 1892 loaded C:\Windows\System32\cryptbase.dll at 00007ffbc6ba0000
    --- Process 1892 loaded C:\Windows\System32\bcryptprimitives.dll at 00007ffbc73c0000
    27939   28831 [main] nuttx (1892) sigprocmask: 0 = sigprocmask (0, 0x0, 0x180320DD0)
     1040   29871 [main] nuttx (1892) open_shared: name shared.5, n 5, shared 0x180030000 (wanted 0x180030000), h 0x10C, *m 6
      243   30114 [main] nuttx (1892) user_heap_info::init: heap base 0x800000000, heap top 0x800000000, heap size 0x20000000 (536870912)
      251   30365 [main] nuttx (1892) open_shared: name S-1-5-21-809113384-3503045568-2100469263-1000.1, n 1, shared 0x180020000 (wanted 0x180020000), h 0x108, *m 6
      238   30603 [main] nuttx (1892) user_info::create: opening user shared for 'S-1-5-21-809113384-3503045568-2100469263-1000' at 0x180020000
      250   30853 [main] nuttx (1892) user_info::create: user shared version AB1FCCE8
      280   31133 [main] nuttx (1892) fhandler_pipe::create: name \\.\pipe\cygwin-c3851c815b7231fc-1892-sigwait, size 11440, mode PIPE_TYPE_MESSAGE
      315   31448 [main] nuttx (1892) fhandler_pipe::create: pipe read handle 0x120
      277   31725 [main] nuttx (1892) fhandler_pipe::create: CreateFile: name \\.\pipe\cygwin-c3851c815b7231fc-1892-sigwait
     1715   33440 [main] nuttx (1892) fhandler_pipe::create: pipe write handle 0x124
     1706   35146 [main] nuttx (1892) dll_crt0_0: finished dll_crt0_0 initialization
    --- Process 1892 thread 9836 created
     6195   41341 [main] nuttx (1892) set_signal_mask: setmask 0, newmask FFFFFFFFFFFEFEFF, mask_bits 0
     1370   42711 [main] nuttx (1892) pthread_sigmask: 0 = pthread_sigmask(0, 0xFFFFCA30, 0xFFFFCA38)
     1850   44561 [main] nuttx (1892) set_signal_mask: setmask FFFFFFFFFFFEFEFF, newmask 0, mask_bits FFFFFFFFFFFEFEFF
     1851   46412 [sig] nuttx (1892) wait_sig: entering ReadFile loop, my_readsig 0x120, my_sendsig 0x124
     2396   48808 [main] nuttx (1892) pthread_sigmask: 0 = pthread_sigmask(0, 0xFFFFCA28, 0x0)
     6271   55079 [main] nuttx (1892) set_signal_mask: setmask 0, newmask FFFFFFFFFFFEFEFF, mask_bits 0
     4638   59717 [main] nuttx (1892) pthread_sigmask: 0 = pthread_sigmask(0, 0xFFFFC9C0, 0xFFFFC9C8)
     3305   63022 [main] nuttx (1892) set_signal_mask: setmask FFFFFFFFFFFEFEFF, newmask 0, mask_bits FFFFFFFFFFFEFEFF
     3421   66443 [main] nuttx (1892) pthread_sigmask: 0 = pthread_sigmask(0, 0xFFFFC9B8, 0x0)
      70955 [main] nuttx (1892) C:\tools\cygwin\home\btashton\sources\nuttx\nuttx.exe: *** fatal error - Internal error: Out of memory for new path buf.
     4512   70955 [main] nuttx (1892) C:\tools\cygwin\home\btashton\sources\nuttx\nuttx.exe: *** fatal error - Internal error: Out of memory for new path buf.
      75682 [main] nuttx (1892) cygwin_exception::open_stackdumpfile: Dumping stack trace to nuttx.exe.stackdump
     4727   75682 [main] nuttx (1892) cygwin_exception::open_stackdumpfile: Dumping stack trace to nuttx.exe.stackdump
    70579  146261 [main] nuttx (1892) proc_terminate: nprocs 0
      470  146731 [main] nuttx (1892) proc_terminate: leaving
    2490  152125 [main] nuttx (1892) pinfo::exit: Calling ExitProcess n 0x1, exitcode 0x100

Yes that is exactly the same problem.  At least same "fatal error - Internal error: Out of memory for new path buf" and subsequent stack dump.  If you put a breakpoint on main, it is never hit.  Adding --gdb to CFLAGS.  Addresses in stackdump are all in cygwin.dll.

So this is a real, showstopping problem!  It makes life difficult for me since I only use windows and my go-to hardware is mostly old and flaky now.

patacongo, 2020-09-29:
@btashton I do remember seeing that but then dismissing it because the next comment implies that you found a fix.  The README.txt does mention a situation that generates the same error.  The error itself is not informative.

In the case you in the README about the problem was old vs newer Cygwin tools.  Old tools decorated symbols with a leading _, the newer tools did not.  So, that is not the issue here but the same basic problem:  The error is occuring deep inside Cygwin before main is even called due to something wrong.

This will probably require a 'git bisect' to find the culprit.

patacongo, 2020-09-29:
@btashton @liuguo09 I did the bisection to determine which commit broke the Cygwin simulator build.  I had to do this manually in order to select the matching commit from apps/.  I did the bisection using only commits to arch/sim to simplify the job.  All tests performed with apps:

    commit d38351cfd07f6e0b52728e61d5f51c44d68b5dba
    Author: Brennan Ashton <bashton@brennanashton.com>
    Date:   Sat Aug 29 12:39:56 2020 -0700

The more appropriate apps/ commit of Sep 8 introduces compilation problems in nshlib (nsh_console.c) and could not be used.

This is the result of the bisection:

First corrupted Cygwin simulation build:

    commit 2cfb239a87263ffa55f49334bf00c4dbf6519f83
    Author: ligd <liguiding1@xiaomi.com>
    Date:   Tue Sep 8 11:20:16 2020 +0800

Last good Cywgin simulation build

    commit 42a1d45a8acdd1d77b58a121d56367ce1a3f2c80
    Author: ligd <liguiding1@xiaomi.com>
    Date:   Tue Sep 8 16:33:08 2020 +0800

I did the bisection using only commits to arch/sim.  It is possible that the some other commit could have cause the problem.  So, as a further test, I built with the current HEAD of the repositories after reverting commit 2cfb239a87263ffa55f49334bf00c4dbf6519f83.  It worked perfectly proving that that commit is the one that broke the Cygwin build.

patacongo, 2020-09-29:
@btashton So the error as described in the sim README.txt was do to bad symbols as is this one.  So the same error is not surprising.  The problem is easily solved by reverting https://github.com/apache/incubator-nuttx/commit/2cfb239a87263ffa55f49334bf00c4dbf6519f83

However, I would like @liuguo09 's opinion first.  Can we just revert this commit?

xiaoxiang781216, 2020-09-30:
@liguiding please take a look.

closed 2020-09-30 by patacongo