apache/nuttx #1952: Documentation: sphinx-build not found

saramonteiro, 2020-10-08:
When running the `make html` the result is:

``` shell
/bin/sh: 1: sphinx-build: not found
Makefile:42: recipe for target 'clean' failed
make: *** [clean] Error 127
```

It seems there's something missing in the Pipfile
@v01d 
@btashton

v01d, 2020-10-08:
Hi, sara, did you follow the steps outlined here?

https://nuttx.apache.org/docs/latest/contributing/documentation.html#building

If so, did you see sphinx being installed during `pipenv install`? `sphinx-build` is part of the sphinx python package.

saramonteiro, 2020-10-08:
Hi Matias,

Yes, I followed the steps from the contributing doc.
Well, when I run `pipenv install` is said it installed the dependencies from Pipfile.lock successfully but I really did not see the sphinx python package.

v01d, 2020-10-08:
If you go to the Documentation directory and run this, can you display the output:
```
$ pipenv clean
$ pipenv sync
```

saramonteiro, 2020-10-08:
Output:
`Installing dependencies from Pipfile.lock (f6636c)…
  🐍   ▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉ 28/28 — 00:01:28
All dependencies are now up-to-date!`

Then, I run again `make html` and the result was:
`/bin/sh: 1: sphinx-build: not found
Makefile:42: recipe for target 'clean' failed
make: *** [clean] Error 127`

btashton, 2020-10-08:
Did you do `pipenv shell` or use `pipenv run` to run the make command?

saramonteiro, 2020-10-08:
`pipenv shell`

v01d, 2020-10-09:
Did you manage to solve this?
If not, maybe paste the exact sequence of commands and output here. I feel that we're missing something or maybe there's something that is not fully configured in your system. pipenv shell should take care of defining the PATH to expose the sphinx-build script. This is how it looks on my end:

```
 ~/c/nuttx_doc/nuttx/Documentation ❯ pipenv shell                                 on   master *1
Launching subshell in virtual environment…
 . /home/v01d/.local/share/virtualenvs/Documentation-1KDjy31b/bin/activate
❯ which sphinx-build
/home/v01d/.local/share/virtualenvs/Documentation-1KDjy31b/bin/sphinx-build
 ~/c/nuttx_doc/nuttx/Documentation ❯ 
```

saramonteiro, 2020-10-14:
Yes, It was related to my system indeed. Conflicts with automatic activated virtual envs.
Now I got to build the HTML pages.
Thanks!

closed 2020-10-14 by saramonteiro