apache/nuttx #2001: Idle stack size is not checked.

fjpanag, 2020-10-16:
I am using NuttX v9.1.0, on an STM32F427.

As I see, (at least) on Cortex-M architectures, the idle stack is placed right after BSS.  
The top of stack is defined as:  
`#define IDLE_STACK      ((unsigned)&_ebss+CONFIG_IDLETHREAD_STACKSIZE-4)`  

This makes the assumption that there is enough memory after BSS to fit the idle stack. If BSS grows enough that there is less than `CONFIG_IDLETHREAD_STACKSIZE` free space, then the top of stack will fall outside the memory region.

Even worse, since this is not checked during build, there will be no warning or any indication of the problem. The system will just crash during initialization.

My proposition is to add an explicit memory region in the linker script to place the idle stack, so its position and size are well defined, and checked during build.

fjpanag, 2020-11-10:
I was thinking... In a typical bare-metal application, usually the top-of-stack is defined as a linker symbol, since its size its not known. There things are much simpler, there is only one stack which takes all available space after BSS.

However, NuttX is a different thing. It has multiple stacks, all of which have a defined size in compile-time. There is no way for the stack to grow to take all remaining space after linking (and there would be no reason to do so).

So, why have the IDLE stack defined like that? Why not make it a statically allocated buffer?
E.g.
```
uint8_t g_idle_topstack[CONFIG_IDLETHREAD_STACKSIZE];
```

This way the stack will be allocated within BSS. The linker will be able to check its size and whether it fits.

It will also give the ability to set it to a specific memory region. An users' custom linker scripts will be able to place the stack to whatever memory region they like. Just as discussed in #2002, there will be no assumptions on how stuff should be mapped into memory.

xiaoxiang781216, 2020-11-11:
No, you can't put g_idle_topstack to BSS section, since the startup code which zero out BSS section will corrupt your running(idle) stack.
Yes, you can define the idle stack as a global arrary, but you must put it into a special section(e.g. .noinit) in link script.

fjpanag, 2020-11-11:
Well, I thought of BSS because in most systems BSS is zero'ed (or painted) very soon during initialization, in a no-return function. So usually overwriting it, is not a problem, because you will never pop anything from the "corrupted" portion.

Nevertheless, you are right. It has to be done correctly.

What is the most appropriate section to place the stack? `.noinit`? Or define a special section (which requires changes in all linker-scripts though...)?

xiaoxiang781216, 2020-11-12:
> Well, I thought of BSS because in most systems BSS is zero'ed (or painted) very soon during initialization, in a no-return function. So usually overwriting it, is not a problem, because you will never pop anything from the "corrupted" portion.
> 

No, it may corrupt the variables you are using, the zero memory loop may never or premature terminate.

> Nevertheless, you are right. It has to be done correctly.
> 
> What is the most appropriate section to place the stack? `.noinit`? Or define a special section (which requires changes in all linker-scripts though...)?

.noinit is already used in several linker script, so it's a good candidate. Yes, you have to change all linker-scripts, so I don't suggest this direction.
The idle stack size overflow isn't a common problem, do you really hit this issue?