apache/nuttx #2002: Heap base shall be configurable.

fjpanag, 2020-10-16:
On STM32, heap base is defined as:  
`#define HEAP_BASE      ((uintptr_t)&_ebss + IDLE_STACKSIZE)`  

This makes the assumption that the heap will always be after BSS & Idle Stack, which is not always possible.  
On systems with multiple RAM regions, that are not contiguous, this is not going to work.

For example, I am using an STM32F427, that has:
- 64k @ 0x10000000
- 192k @ 0x20000000

This forces me to place everything in the 192k SRAM region (and thus lose the CCM), or place everything in CCM (64k), and not be able to take advantage of the full 192k RAM of the system.

Even worse, if I place BSS in 64k and select `CONFIG_STM32_CCMEXCLUDE`, **the heap will still be located within CCM**, and will extend till the end of the 192k RAM (of course with a "hole" in between).

I believe that we need a different way of defining the Heap.  
Examples/ideas:
- FreeRTOS heap_5 [uses a table](https://www.freertos.org/FreeRTOS_Support_Forum_Archive/July_2017/freertos_Using_heap_5.c_d1dfe891j.html). 
- Linker script regions or symbols may be defined for the heap.
- We may set HEAP_BASE (and size) in Kconfig.

xiaoxiang781216, 2020-10-25:
@fjpanag has you tried CONFIG_MM_REGIONS? this option let you specify multiple discontinuous heap regions as needed.

fjpanag, 2020-10-25:
Hi @xiaoxiang781216,

`CONFIG_MM_REGIONS`, allows you to configure multiple heaps, but this is not what I need. I only need one heap. The problem is that NuttX does not allow me to place the heap wherever I need. It just makes the assumption that it is just after BSS.

If I recall correctly the same goes for defining multiple heaps, with `CONFIG_MM_REGIONS`. It just places the two heaps at predetermined positions. If someone needs to modify this, they must make changes to the low-level drivers.

I believe that the memory map of the whole system shall be configurable, or at least governed by the board. The OS must not have hard-coded addresses, or make any assumptions about the end application.

xiaoxiang781216, 2020-10-26:
No, you have choice in arm_addregion:
1. Call ?mm_initialize to create the multiple heaps
2. Call ?mm_addregion to add the multiple regions into one heap
3. Combine item 1 and 2 to get the multiple heaps with multiple regions
BTW, how to find the heap location is controlled by chip specific code, so you are free to change the implementation to take the different approach without breaking the code in sched.

fjpanag, 2020-10-26:
Your proposition suggests modification to the `arch` files of (in my case) STM32.  
I don't want to have per-project forks of NuttX just for defining the memory map. That is why I am looking for a proper solution, allowing project-level configuration of the memory map.

Checking in `stm32_allocateheap.c`:

```c
/******************************************************************************
 * Name: up_allocate_heap
  ...
 *
 *   The following memory map is assumed for the flat build:
 *
 *     .data region.  Size determined at link time.
 *     .bss  region  Size determined at link time.
 *     IDLE thread stack.  Size determined by CONFIG_IDLETHREAD_STACKSIZE.
 *     Heap.  Extends to the end of SRAM.
 *
  ...
 *
 ******************************************************************************/
```
It assumes that the heap will always follow DATA -> BSS -> IDLE stack -> Heap.  
In this function the heap definition is explicitly coupled to the IDLE stack:
```
  /* Return the heap settings */

  board_autoled_on(LED_HEAPALLOCATE);
  *heap_start = (FAR void *)g_idle_topstack;
  *heap_size  = SRAM1_END - g_idle_topstack;
```

Furthermore, in the same file, the function `void arm_addregion(void)`, clearly uses a bunch of hard-coded values.  

Also note that `arm_addregion` is only compiled for `CONFIG_MM_REGIONS > 1`. In my case there is only one heap.

xiaoxiang781216, 2020-10-26:
mm_addregion is same as FreeRTOS heap_5 [uses a table] (https://www.freertos.org/FreeRTOS_Support_Forum_Archive/July_2017/freertos_Using_heap_5.c_d1dfe891j.html). 
What's the final layout do you want for STM32F427? one heap two regions, or two heaps?

fjpanag, 2020-10-29:
Hi @xiaoxiang781216,

I am not sure if I understand you correctly. Can you provide me with some details on how the heap is defined as a table?  
BTW I am using NuttX 9.1.

I want to have one heap, in a single continuous section. I just don't want to have this heap adjacent to the BSS/IDLE stack. This heap needs to be located to a different memory bank, completely unrelated to any other memory regions.

xiaoxiang781216, 2020-10-29:
> Hi @xiaoxiang781216,
>
> I am not sure if I understand you correctly. Can you provide me with some details on how the heap is defined as a table?
> BTW I am using NuttX 9.1.
>
> I want to have one heap, in [...]

so you want the global data in one memory region and the malloc from another memory region? But how do you utilize the remain memory in the first region?

fjpanag, 2020-10-30:
Yes, this is correct.  
CCM is rather small (64kb) comparing to the second RAM region (192kb). Since this is a port of an already working firmware, I know that DATA + BSS will completely fill the CCM, and I want the heap to take advantage of the whole 192kb.

Regardless of my specific application:  
CCM does not support DMA transfers. So one may want to exclude it from the heap, if they want to use DMA. BUT! The heap must always follow DATA/BSS which cases two problems:
- If I select CCM for DATA/BSS, the memory manager will fail. It will try to use an invalid memory region.
- The only way to fix the above, is to use the second region also for DATA & BSS. In this case CCM will not be able to be used, it will remain empty. And thus I won't be able to use the whole available RAM on my system.

xiaoxiang781216, 2020-10-31:
Since g_idle_topstack equals to ((uintptr_t)&_ebss + IDLE_STACKSIZE) and _ebss is controlled by the board link script, you can adjust _ebss to ensure the memory between _ebess and the end of CCM is little bigger than IDLE_STACKSIZE(to cover the idle stack and mm_heap_s structure. And enable CONFIG_MM_REGIONS to let arm_addregion add the second RAM.

xiaoxiang781216, 2020-10-31:
Or you can skip DMA or use bounce buffer if the memory inside CCM.

fjpanag, 2020-11-02:
Maybe I will manage to make it work with one of the above suggestions.

But it looks more than a work-around to me, that an proper solution, as it is usually needed in such systems.

This is why I am insisting that the heap boundaries shall be defined in a different way, that does not depend on other (unrelated) memories, and is easily configurable.

Personally, I believe that the best case would be to define the heap in the linker script.

fjpanag, 2020-11-02:
I would like to contribute on this, but I am not very confident at the moment to introduce such changes on my own.

If however someone can take the lead, I can spend some time helping out.

xiaoxiang781216, 2020-11-03:
Yes, you can add a new Kconfig option to adjust how stm32 find the heap memory.