apache/nuttx #2084: sim:bluetooth build broken

btashton, 2020-10-24:
This has not been triaged further, but this is the output running the sim target.

`sim:bluetooth`
```
❯ ./nuttx 
ERROR: btnull_register() failed: -116

NuttShell (NSH) NuttX-9.1.0
nsh> uname -a
NuttX NuttX-SIM 9.1.0 3285517910 Oct 24 2020 12:48:16 sim sim
```

v01d, 2020-10-24:
This is due to timeouts not waiting. The bluetooth CMD times out immediately. Also sleep 1 returns immediately.
@xiaoxiang781216 I believe you were working recently with timing in sim right?

sebastianene07, 2020-10-25:
> This is due to timeouts not waiting. The bluetooth CMD times out immediately. Also sleep 1 returns immediately.
> @xiaoxiang781216 I believe you were working recently with timing in sim right?

I think that `SIM_WALLTIME` should be enabled for this to run correctly with `SIM_WALLTIME_SIGNAL=y`

v01d, 2020-10-25:
Enabling SIM_WALLTIME indeed makes the wait function as expected (although HCI initialization still fails). I would say that either SIM_WALLTIME=n is broken or it makes no sense to have it disabled if timeouts will not work.

Regarding bluetooth stack itself, I get the following:

```
bt_initialize: btdev 0x5648f0b60080
bt_hci_cmd_create: opcode 0c03 param_len 0
bt_buf_alloc: buf 0x5648f0b63590 type 0 reserve 0
bt_hci_cmd_create: buf 0x5648f0b63590
bt_buf_extend: buf 0x5648f0b63590 len 3
bt_hci_cmd_send_sync: opcode 0c03 len 3
hci_tx_kthread: started
hci_tx_kthread: Sending command 0c03 buf 0x5648f0b63590 to driver
btnull_send: Bit bucket: length 3
btnull_send: CMD: 0c03
bt_buf_alloc: buf 0x5648f0b63560 type 1 reserve 0
btnull_send: Send CMD complete event
bt_hci_receive: buf 0x5648f0b63560 len 5
priority_rx_work: list 0x5648f0b63650
priority_rx_work: buf 0x5648f0b63560 type 1 len 5
bt_buf_consume: buf 0x5648f0b63560 len 2
hci_cmd_complete: opcode 0c03
bt_buf_consume: buf 0x5648f0b63560 len 3
hci_reset_complete: status 0
bt_buf_release: buf 0x5648f0b63560 ref 1 type 1
bt_buf_release: Buffer freed: 0x5648f0b63560
bt_buf_release: buf 0x5648f0b63590 ref 1 type 0
bt_buf_release: Buffer freed: 0x5648f0b63590
hci_initialize: ERROR: BT_HCI_OP_RESET failed: -116
bt_initialize: ERROR:  hci_initialize failed: -116
ERROR: btnull_register() failed: -116
```

sebastianene07, 2020-10-25:
'SIM_WALLTIME=n' is not broken, it tries to run the simulation as fast as possible by driving the tick events from Idle Task. I agree with you that this is not ideal, this is the reason why I implemented 'SIM_WALLTIME_SIGNAL=y'.

v01d, 2020-10-25:
I had this fix as part of my work on the nRF52 LL branch, while I was still using NuttX's stack. With the linked changed initialization now goes as far as:

```
bt_initialize: btdev 0x56126d485080
bt_hci_cmd_create: opcode 0c03 param_len 0
bt_buf_alloc: buf 0x56126d488590 type 0 reserve 0
bt_hci_cmd_create: buf 0x56126d488590
bt_buf_extend: buf 0x56126d488590 len 3
bt_hci_cmd_send_sync: opcode 0c03 len 3
hci_tx_kthread: started
hci_tx_kthread: Sending command 0c03 buf 0x56126d488590 to driver
btnull_send: Bit bucket: length 3
btnull_send: CMD: 0c03
bt_buf_alloc: buf 0x56126d488560 type 1 reserve 0
btnull_send: Send CMD complete event
bt_hci_receive: buf 0x56126d488560 len 5
priority_rx_work: list 0x56126d488650
priority_rx_work: buf 0x56126d488560 type 1 len 5
bt_buf_consume: buf 0x56126d488560 len 2
hci_cmd_complete: opcode 0c03
bt_buf_consume: buf 0x56126d488560 len 3
hci_reset_complete: status 0
bt_buf_addref: buf 0x56126d488560 (old) ref 1 type 1
bt_buf_release: buf 0x56126d488560 ref 2 type 1
bt_buf_release: Remaining references: 1
bt_buf_release: buf 0x56126d488590 ref 1 type 0
bt_buf_release: Buffer freed: 0x56126d488590
bt_buf_release: buf 0x56126d488560 ref 1 type 1
bt_buf_release: Buffer freed: 0x56126d488560
bt_hci_cmd_create: opcode 1003 param_len 0
bt_buf_alloc: buf 0x56126d488560 type 0 reserve 0
bt_hci_cmd_create: buf 0x56126d488560
bt_buf_extend: buf 0x56126d488560 len 3
bt_hci_cmd_send_sync: opcode 1003 len 3
hci_tx_kthread: Sending command 1003 buf 0x56126d488560 to driver
btnull_send: Bit bucket: length 3
btnull_send: CMD: 1003
bt_buf_alloc: buf 0x56126d488590 type 1 reserve 0
btnull_send: Send CMD complete event
bt_hci_receive: buf 0x56126d488590 len 14
priority_rx_work: list 0x56126d488650
priority_rx_work: buf 0x56126d488590 type 1 len 14
bt_buf_consume: buf 0x56126d488590 len 2
hci_cmd_complete: opcode 1003
bt_buf_consume: buf 0x56126d488590 len 3
hci_cmd_complete: Unhandled opcode 1003
bt_buf_addref: buf 0x56126d488590 (old) ref 1 type 1
bt_buf_release: buf 0x56126d488590 ref 2 type 1
bt_buf_release: Remaining references: 1
bt_buf_release: buf 0x56126d488560 ref 1 type 0
bt_buf_release: Buffer freed: 0x56126d488560
read_local_features_complete: status 0
bt_buf_release: buf 0x56126d488590 ref 1 type 1
bt_buf_release: Buffer freed: 0x56126d488590
bt_hci_cmd_create: opcode 1001 param_len 0
bt_buf_alloc: buf 0x56126d488590 type 0 reserve 0
bt_hci_cmd_create: buf 0x56126d488590
bt_buf_extend: buf 0x56126d488590 len 3
bt_hci_cmd_send_sync: opcode 1001 len 3
hci_tx_kthread: Sending command 1001 buf 0x56126d488590 to driver
btnull_send: Bit bucket: length 3
btnull_send: CMD: 1001
bt_buf_alloc: buf 0x56126d488560 type 1 reserve 0
btnull_send: Send CMD complete event
bt_hci_receive: buf 0x56126d488560 len 5
priority_rx_work: list 0x56126d488650
priority_rx_work: buf 0x56126d488560 type 1 len 5
bt_buf_consume: buf 0x56126d488560 len 2
hci_cmd_complete: opcode 1001
bt_buf_consume: buf 0x56126d488560 len 3
[... 222 lines cut ...]
bt_hci_cmd_send_sync: opcode 2017 len 35
hci_tx_kthread: Sending command 2017 buf 0x56126d488530 to driver
btnull_send: Bit bucket: length 35
btnull_send: CMD: 2017
bt_buf_alloc: buf 0x56126d488560 type 1 reserve 0
btnull_send: Send CMD complete event
bt_hci_receive: buf 0x56126d488560 len 5
priority_rx_work: list 0x56126d488650
priority_rx_work: buf 0x56126d488560 type 1 len 5
bt_buf_consume: buf 0x56126d488560 len 2
hci_cmd_complete: opcode 2017
bt_buf_consume: buf 0x56126d488560 len 3
hci_cmd_complete: Unhandled opcode 2017
hci_cmd_done: WARNING: status 207
bt_buf_release: buf 0x56126d488560 ref 1 type 1
bt_buf_release: Buffer freed: 0x56126d488560
bt_hci_cmd_send_sync: ERROR:  Failed get return parameters
bt_buf_release: buf 0x56126d488530 ref 1 type 0
bt_buf_release: Buffer freed: 0x56126d488530
aes_test: ERROR: Test aes-cmac0: Failed
smp_self_test: ERROR: SMP AES-CMAC self tests failed
ERROR: btnull_register() failed: -1
```

I'm not really sure what is the failure about, maybe some feature that needs enabling.

v01d, 2020-10-25:
> 'SIM_WALLTIME=n' is not broken, it tries to run the simulation as fast as possible by driving the tick events from Idle Task. I agree with you that this is not ideal, this is the reason why I implemented 'SIM_WALLTIME_SIGNAL=y'. @xiaoxiang781216 maybe we should make 'SIM_WALLTIME' option enabled by default. By looking at the bluetooth stack logs, it seems that you receive a cmd response for HCI_RESET. Why did it fail though ?

But you have to realize this is not the normal case and of course many things will break. I would set SIM_WALLTIME=y by default.

xiaoxiang781216, 2020-10-25:
If the driver interact with the real device and can't work without the correct host timing, it should select SIM_WALLTIME in Kconfig like what SIM_X11FB and SIM_NETDEV have done.

v01d, 2020-10-25:
But why not making it the default? There's more risk in default disabled than default enabled doesn't it?

v01d, 2020-10-25:
Consider the large number of configurations that require this to work, you would have to go one by one and check.

xiaoxiang781216, 2020-10-25:
> But why not making it the default? There's more risk in default disabled than default enabled doesn't it?

I don't know why SIM_WALLTIME isn't default. Maybe SIM_WALLTIME is added later, and the owner want to keep the old behavior.

> Consider the large number of configurations that require this to work, you would have to go one by one and check.

But we don't need to check all defconfig in boards/sim/, because if you add 'select SIM_WALLTIME' inside 'config SIM_HCISOCKET', kbuild will automatically enable CONFIG_SIM_WALLTIME if some defconfig enable CONFIG_SIM_HCISOCKET.

v01d, 2020-10-25:
@sebastianene07 do you see a problem in having this enabled by default?
If SIM_WALLTIME=n means `sleep 1` returns immediately I would change this to `y` by default. Otherwise if with `n` sleep should still work, then we have another problem.

sebastianene07, 2020-10-25:
> @sebastianene07 do you see a problem in having this enabled by default?
> If SIM_WALLTIME=n means `sleep 1` returns immediately I would change this to `y` by default. Otherwise if with `n` sleep should still work, then we have another problem.

I think that we should keep this enabled by default. There are many features that depend on this timing.

btashton, 2020-10-25:
I also don't see any reason in this case to not change the default to enabled. The worst impact is a test runs long and we can explicitly disable it for that test.  Im thinking an OS interface test were time bounds might be being tested.

v01d, 2020-10-25:
It should be easy to create a test that kill nuttx in that case.

@sebastianene07 could you open a PR to change the default and refresh all sim configs?

sebastianene07, 2020-10-26:
> It should be easy to create a test that kill nuttx in that case.
> 
> @sebastianene07 could you open a PR to change the default and refresh all sim configs?

I created a pull request here https://github.com/apache/incubator-nuttx/pull/2116
This enabled the option by default from Kconfig.

btashton, 2020-11-05:
> Enabling SIM_WALLTIME indeed makes the wait function as expected (although HCI initialization still fails). I would say that either SIM_WALLTIME=n is broken or it makes no sense to have it disabled [...]

@v01d I think something is broken in the stack because I get this running via the HCISOCKET:
```
= Close Index: 60:57:18:EE:03:FD                             [hci0] 3591.587962
@ MGMT Event: Index Removed (0x0005) plen 0         {0x0002} [hci0] 3591.588456
@ MGMT Event: Index Removed (0x0005) plen 0         {0x0001} [hci0] 3591.588456
= Open Index: 60:57:18:EE:03:FD                              [hci0] 3591.588505
= Index Info: 60:57:18:EE:03:FD (Intel Corp.)                [hci0] 3591.588507
@ RAW Close: nuttx                                         {0x0003} 3591.588510
@ USER Open: nuttx (privileged) version 2.22        {0x0003} [hci0] 3591.588512
< HCI Command: Reset (0x03|0x0003) plen 0              #2759 [hci0] 3591.589814
= bluetoothd: Endpoint unregistered: sender=:1.67 path=/MediaEn..   3591.590811
= bluetoothd: Endpoint unregistered: sender=:1.67 path=/MediaEn..   3591.590819
> HCI Event: Command Complete (0x0e) plen 4            #2760 [hci0] 3591.603696
      Reset (0x03|0x0003) ncmd 2
        Status: Success (0x00)
```
```
bt_initialize: btdev 0x44b960
bt_hci_cmd_create: opcode 0c03 param_len 0
bt_buf_alloc: buf 0x466250 type 0 reserve 0
bt_hci_cmd_create: buf 0x466250
bt_buf_extend: buf 0x466250 len 3
bt_hci_cmd_send_sync: opcode 0c03 len 3
hci_tx_kthread: started
hci_tx_kthread: Sending command 0c03 buf 0x466250 to driver
bt_buf_release: buf 0x466250 ref 1 type 0
bt_buf_release: Buffer freed: 0x466250
bt_buf_alloc: buf 0x466250 type 0 reserve 1
bthcisock_loop: Recv BT_EVT
bt_hci_receive: buf 0x466250 len 6
priority_rx_work: list 0x466310
priority_rx_work: buf 0x466250 type 1 len 6
bt_buf_consume: buf 0x466250 len 2
hci_cmd_complete: opcode 0c03
bt_buf_consume: buf 0x466250 len 3
hci_reset_complete: status 0
hci_cmd_done: ERROR:  Unexpected completion of opcode 0x0c03
bt_buf_alloc: buf 0x466220 type 0 reserve 1
bt_buf_release: buf 0x466250 ref 1 type 1
bt_buf_release: Buffer freed: 0x466250
hci_initialize: ERROR: BT_HCI_OP_RESET failed: -116
bt_initialize: ERROR:  hci_initialize failed: -116
ERROR: bthcisock_register() failed: -116
```

It seems like it is aware that reset completed, but then still errors out on cmd_done.

btashton, 2020-11-05:
Yeah I reverted back to commit 8934f2ed79c0e2fe6e94b4116f924444cc7dbe12 and it works fine, so I think some of the bluetooth stack changes after broke this.

v01d, 2020-11-05:
Ok, I'll take a look

btashton, 2020-11-14:
@v01d Have you had a chance to look into this?  I think the BLE issues are some of the last remaining that I know of for the 10.0.0 release.

v01d, 2020-11-15:
Sorry, no. I'll see what I can find later today

v01d, 2020-11-15:
@btashton I'm confused, did you apply the fix I linked to this PR? It went much further ahead during initialization for me than what you last copied.

v01d, 2020-11-15:
From what I'm seeing this is something that will not ever work with bt_null. The command that is failing is LE_ENCRYPT, which should encrypt the passed data and return it as part of the return parameters of the CMD COMPLETE HCI command. bt_null simply returns generic CMD COMPLETE events without return parameters for any unhandled opcode, so the BT stack chokes when this does not contain the expected response. I don't see how this could have worked before, unless something else was wrong and hiding this problem.
I'm not sure what would be the best way to deal with this, I don't think it makes much sense to be testing such a stub driver.

btashton, 2020-11-15:
Sorry I think I'm mixing the two ble tickets (I think they are related).  The HCI socket is still broken on master as I posted above and works fine if I revert back the new bluetooth changes.

v01d, 2020-11-15:
Well, regarding this issue, as I mentioned I think we may have to close it with "wontfix" and maybe even remove that config as it will never work as is.

Regarding the bthcisock, 10.0 works for me. Maybe there's something strange with your BT stack or BT adapter.

btashton, 2020-11-15:
There is nothing wrong with my adapter. Look at the log I posted it is not handling the most basic completion event. Plus it works prior to the recent changes.

v01d, 2020-11-15:
I was confused because that issue was marked as blocker for 10.0 and on 10.0 there's no problem, but indeed I see it on master. I'll update any findings there (you can remove the blocker tag in the mean time).

btashton, 2020-11-15:
This smp functionality can be disabled by a config flag.  This PR disables it now https://github.com/apache/incubator-nuttx/pull/2305

v01d, 2020-11-15:
Ah, that's better.

PR #2137 (2020-10-27): [BACKPORT #2116] arch/sim: Enable SIM_WALLTIME option by default 
## Summary
Backport #2116

## Impact
Partially resolve #2084 but there seems to be other issues as well.

## Testing
PR #2305 (2020-11-15): Disable smp selftest for bt_null simulation configuration
## Summary
The bt_null driver does very limited emulation of commands for testing the stack so it should not be expected to handle more complex modes like encryption.

This disables this self test from the sim:bluetooth configuration that previously would result in this.  I thought about changing the Kconfig to do this automatically, but I think it is worth it to allow this mode to be selected if we want to be able to test the smp flow in the future.  Plus this area is very much in flux right now.
This resolves #2084 

```
le_encrypt: e [... cut]
PR #2307 (2020-11-15): [BACKPORT #2305] Disable smp selftest for bt_null driver
## Summary
This backports #2305 

## Impact
Resolves #2084 

## Testing
Verified Bluetooth device is registered.
closed 2020-11-15 by Ouss4