apache/nuttx #2085: sim:bthcisock build broken

btashton, 2020-10-24:
The `sim:bthcisock` build is broken failing to register the HCI interface

```
❯ sudo btmon
@ RAW Open: nuttx (privileged) version 2.22                  {0x0002} 22.803159
@ MGMT Event: Index Removed (0x0005) plen 0           {0x0001} [hci0] 22.803185
@ MGMT Event: Index Added (0x0004) plen 0             {0x0001} [hci0] 22.803198
@ RAW Close: nuttx        
```

```
❯ sudo setcap 'cap_net_raw,cap_net_admin=eip' ./nuttx
❯ ./nuttx 
ERROR: bthcisock_register() failed: -1

NuttShell (NSH) NuttX-9.1.0
nsh> 
```

btashton, 2020-10-24:
@v01d would you mind taking a look at these two BLE issues on the 10.0 release.  I am working through functional test of the branch, so it will be a while before a circle back to debug.

v01d, 2020-10-24:
I'll have a look. Last time it was working for me.

v01d, 2020-10-24:
```
❯ sudo setcap 'cap_net_raw,cap_net_admin=eip' nuttx/nuttx
❯ nuttx/nuttx

NuttShell (NSH) NuttX-9.1.0
nsh> ifconfig
bnep0	Link encap:UNSPEC at DOWN

nsh> ifup bnep0
ifup bnep0...OK
nsh> ifconfig
bnep0	Link encap:UNSPEC at UP

nsh> 
```
Maybe you don't have your BT dongle up? Or you are already using it from some other app?

btashton, 2020-10-24:
Nope, the code should be taking it down anyway and then opening it as a user socket. You tested the releases/10.0 branch right?  I can take a look later and see if I can figure out whats up.  Running into fair amount of partially broken configurations :/

v01d, 2020-10-24:
Yes, this is for 10.0

sebastianene07, 2020-10-31:
Can I help on this ? I have some experience with bluetooth devices.

btashton, 2020-10-31:
@sebastianene07 I think it is something with my config since I wrote the code for it recently and @v01d has been using it a lot for his BLE work.  I'll be digging into it today.

v01d, 2020-11-15:
Found the issue, see related PR. This is only on master though, so it does not need to block 10.0 release.

btashton, 2020-11-15:
Hmm not this seems to be working fine on the 10.0 branch now.  I am not sure what was going on, maybe I accidentally ended up back on master.  I'll close this out.  Sorry for the extra noise @v01d

closed 2020-11-15 by btashton