apache/nuttx #2086: sim:cxx Build Broken

btashton, 2020-10-24:
The build is broken for `sim:cxxtest`

```
make[2]: Leaving directory '/home/bashton/nuttx/apache/incubator-nuttx/boards/sim/sim/sim/src'
CPP:  nuttx-names.in-> nuttx-names.dat
cc: warning: nuttx-names.in: linker input file unused because linking not done
LD:  nuttx
ld: cannot open linker script file nuttx.ld: No such file or directory
make[1]: *** [Makefile:295: nuttx] Error 1
make[1]: Leaving directory '/home/bashton/nuttx/apache/incubator-nuttx/arch/sim/src'
make: *** [tools/Makefile.unix:415: nuttx] Error 2
```

Also the NOTES section in readme is no long accurate for this target and should be updated to correctly represent the requirements of this.

xiaoxiang781216, 2020-10-25:
@btashton I will fix this issue by moving c++ specified setting to arch/**/Toolchain.defs.

btashton, 2020-10-25:
Thanks @xiaoxiang781216 sorry some of these reports are brief, I just wanted to try to get some quality of testing done. I'm happy to go back and test any fixes. (just slower during the work week unfortunately)

xiaoxiang781216, 2020-10-25:
The idea to enable the automation test in pre-check is an important method of quality guarantee. @ttnie could help on this.

btashton, 2020-10-25:
I had started to write some tests with robotframework which is also what renode uses. AntMicro who is behind renode had offered to help support making sure there emulation frameworks is working with NuttX.

ttnie, 2020-10-26:
Hi Btashton, 
I've apply PR about adding LTP test support scripts. As LTP is ready, this script could run LTP case by case on sim. 
my PR: https://github.com/apache/incubator-nuttx-testing/pull/70

> I had started to write some tests with robotframework which is also what renode uses. AntMicro who is behind renode had offered to help support making sure there emulation frameworks is working with NuttX.

btashton, 2020-11-07:
@xiaoxiang781216 Have you been able to look into this?

btashton, 2020-11-12:
Going to reopen until we get the backport in so we don't forget.

xiaoxiang781216, 2020-11-13:
Fixed, let's close it.

PR #2290 (2020-11-13): [BACKPORT #2265/#2267/#2281] sim: Remove cxxtest specifial Make.defs
## Summary
Backports #2265 #2267 #2281 

## Impact
Fixes #2086 sim:cxxtest build failure

## Testing
sim:cxxtesting
closed 2020-11-13 by xiaoxiang781216