apache/nuttx #2087: sim:duktape Broken Build

btashton, 2020-10-24:
The build is broken for `sim:cxxtest`

```
❯ ./nuttx
login: admin
password: 
User Logged-in!

NuttShell (NSH) NuttX-9.1.0
MOTD: username=admin password=Administrator
nsh> duk
Segmentation fault (core dumped)
```

xiaoxiang781216, 2020-10-25:
@ no1wudi could you take a look the break?

no1wudi, 2020-10-26:
> @no1wudi could you look at the break?

This program need 32K stack on sim, and about 8k on thumbv7m platform.

@btashton @xiaoxiang781216 Should we let a large enough stack (e.g. 64K) as default stack size for all platform ?

xiaoxiang781216, 2020-10-26:
it's better to change duktape stack to 64KB in sim's defconfig.

no1wudi, 2020-10-26:
it's ok, but may same broken also exist on other platform (like riscv)

xiaoxiang781216, 2020-10-26:
But, no config in riscv enable duktape. if some one enable duktape in new config, he has to it set the stack size explicitly. Maybe, we should remove the default stack from duktape.

btashton, 2020-10-26:
I'll close when I create the backport.

PR #2123 (2020-10-26): sim/duktape: Increase stack size to fix broken
## Summary
Fix #2087 
## Impact
sim:duktape
## Testing
sim
PR #2136 (2020-10-27): [BACKPORT #2123] sim/duktape: Increase stack size to fix broken
## Summary
Backports #2123 

This resolves #2087 

## Impact

## Testing
closed 2020-10-30 by btashton