apache/nuttx #2088: docs: sim:loadable README does not explain how to test

btashton, 2020-10-24:
This configuration is working, but the readme does not explain what is going on with it or how to test it.  We should add this to the the readme.

This is the key part of the configuration:
```
CONFIG_PATH_INITIAL="/system/bin"
CONFIG_USER_INITPATH="/system/bin/nsh"
```
the shell is loaded from the elf, but you can also run any of the ELFs that are in `/system/bin` as they are on the "PATH"

```
❯ ./nuttx 

NuttShell (NSH) NuttX-9.1.0
nsh> help
help usage:  help [-v] [<cmd>]

  .         cd        echo      hexdump   mh        pwd       test      usleep    
  [         cp        exec      kill      mount     rm        time      xd        
  ?         cmp       exit      ls        mv        rmdir     true      
  basename  dirname   false     mb        mw        set       uname     
  break     dd        free      mkdir     poweroff  sleep     umount    
  cat       df        help      mkrd      ps        source    unset     
nsh> ls /system/bin
/system/bin:
 hello
 sh
 ..
 .
 nsh
nsh> hello
Hello, World!!
nsh> 
```

xiaoxiang781216, 2020-10-25:
@liuguo09 please take a look.

liuguo09, 2020-10-27:
I have made PR https://github.com/apache/incubator-nuttx/pull/2134 to update sim loadable README with @btashton shown info : ) Also update sotest and module configs readme.

closed 2020-10-27 by btashton