apache/nuttx #2092: sim:nxffs Test failure

btashton, 2020-10-24:
For the `sim:nxffs` configuration the self test is failing

Here is the first portion of the logs:
```
=== FILLING 1 =============================
Filled file system
  Number of files: 61
  Number deleted:  0
Directory:
 1. Type[8]: File  Name: JzjHDOyw5F6R/j8kmxeOM1/78Jzp
 2. Type[8]: File  Name: FdzG0jEy2NQlSI8RA/sg0qX9IDFExGR0x
 3. Type[8]: File  Name: SVjnrSFqb9CxFARVDgIxAUEvPUDaE4Ro4
 4. Type[8]: File  Name: BlhzQ9Br/d8RLulhpbOiFD6AbyPs/MBPt
 5. Type[8]: File  Name: DjDUaAna/ZJiqxuL7fqOJZLxb/NFyO/KI
 6. Type[8]: File  Name: SKVKqGgNrIcv9XdWMkfxI04QVNqzKPCC2
 7. Type[8]: File  Name: C
 8. Type[8]: File  Name: z7znO4HBnXIlDbjCay5qryGaVYffIsa9Q
 9. Type[8]: File  Name: 4WE00wK6I8CZWLKM0tu3mthuDiqooPR1z
10. Type[8]: File  Name: ZJlzbyZdrw0X7wukqiRW3vIDBpP3foLxA
11. Type[8]: File  Name: KAierUkgMiRd0omFrpHtXtoawi9Xo32Ej
12. Type[8]: File  Name: EnmzEwZ62DqOZOxMgE6N4ysXaVW0rnWni
13. Type[8]: File  Name: FReqW/q7PXIZXqUqibaP8YhiAxZDyTcab
14. Type[8]: File  Name: vGCorVt9z7MLSP5ufguJUUnZesvom0Vk1
15. Type[8]: File  Name: eGOW7bthGZsfwF
16. Type[8]: File  Name: AIsl8kQIy3j//jjqbpqRK8FCQJls8LoaD
17. Type[8]: File  Name: 0lPDaAqD8ndI2DP7l6SLTyhMKFE3Osa6S
18. Type[8]: File  Name: sch9TW6KdVK2sc6t9ZhlRuCI1cS2nZqR4
19. Type[8]: File  Name: RoB8bCDpdLOLYTFIJ8eqUtqlnzL4735Co
20. Type[8]: File  Name: uea
21. Type[8]: File  Name: YwtkcA8LqB0XHb6toMtWGmdyRPtg41CJe
22. Type[8]: File  Name: 5XF16qxUSMm5F3mhxIKPKbKzd4RuHj2P
23. Type[8]: File  Name: UyLbppM2F3ASuuaVVE/EzsSy8GQUsuLKi
24. Type[8]: File  Name: 6taxcnPNDbtOSl3qOUC1pGYThSDhrnvss
25. Type[8]: File  Name: hFjdeyXHle56PZAQ36D/iwo6OeV8QPO44
26. Type[8]: File  Name: 5DfR5f4Din2jehP/
27. Type[8]: File  Name: 2AXNn4WLjx1eYwvJ1RFRiOfsa7CP2tgxZ
28. Type[8]: File  Name: OB8hWs4qtRy5dWAAcAjnj2xu2Oe/F1EYR
29. Type[8]: File  Name: uLX9cO5a5PEi2Ra0X7WWy8fKiWNIgyW/V
30. Type[8]: File  Name: Yfub4Evmec4xPXspp/9CHYWc0TKFpEWJ
31. Type[8]: File  Name: 9HKRHjwFF0fmocZeSvzSLsVLSwp9aS0d6
32. Type[8]: File  Name: q9HE5YeXVimH1iqH9lDYGPe18SZCk0ASJ
ERROR: Failed to open file for reading: 28
  File name: /mnt/nxffs/EVwni/b4A/BqjDc42RCtKvjJ6DInifZ0tME26XGRXWl5ZAE
  File size: 2680
ERROR: Failed to read a file: 0
  File name: /mnt/nxffs/EVwni/b4A/BqjDc42RCtKvjJ6DInifZ0tME26XGRXWl5ZAE
  File size: 2680
ERROR: Failed to verify files
  Number of files: 61
  Number deleted:  0

=== DELETING 1 ============================
ERROR: Unlink 1 failed: 28
  File name:  /mnt/nxffs/7yee3fRJj3gXpKOEuQZMygaVcS3pSUUzIArhnEH7Bfv3yhg3G5HvbRezu06Xonr9JK
  File size:  1038
  File index: 7
ERROR: Unlink 1 failed: 28
  File name:  /mnt/nxffs/kPmXquTSrz
  File size:  7811
  File index: 8
ERROR: Unlink 1 failed: 28
  File name:  /mnt/nxffs/jpfZNjnGapJ
  File size:  2835
  File index: 9
ERROR: Unlink 1 failed: 28
  File name:  /mnt/nxffs/3Ki0kkm/8JgEHA/YMkXqdDJPL5PXR1LKqSZcZfelqZVfb7MTE
  File size:  2280
  File index: 10
ERROR: Unlink 1 failed: 28
  File name:  /mnt/nxffs/NrxbRqA9HY1JnY5yijLr5XvrklQyiA1EQlNK2
  File size:  3706
  File index: 11
ERROR: Unlink 1 failed: 28
  File name:  /mnt/nxffs/rRPpZVdXxJZFxc
  File size:  5998
  File index: 12
ERROR: Unlink 1 failed: 28
  File name:  /mnt/nxffs/Dr0dlhT/YF5uamAqw8AcNDvmDQIdOOvfnO549TdDfZdwNOYSR4b2o32IcYesXUnT8NStVcN1
  File size:  435
  File index: 13
ERROR: Unlink 1 failed: 28
  File name:  /mnt/nxffs/ycvIAKPGXU7EG6AdXTsCxrSiAmruJRy143K/j50nmvF9ENVIbqGis9O8jNUG
  File size:  5383
  File index: 14
ERROR: Unlink 1 failed: 28
  File name:  /mnt/nxffs/3WQeK17gTMTXBx3k/0RLXT7fFCStyffivV9u9P/PsDvwn0KccM0DqzOeTRyS5q0VrjC/ss5CC7yW9WROjjDMxogSov0c1StiiOIq5Blsq0W
  File size:  4242
  File index: 15
ERROR: Unlink 1 failed: 28
  File name:  /mnt/nxffs/0NzfziosNPM3OoHX6t3ED5ZY3iXKkTefv6eL3tBA64YX45A
  File size:  3349
  File index: 16
ERROR: Unlink 1 failed: 28
  File name:  /mnt/nxffs/jbcdNTqSQJs2s3hywoZyTtgDJmceVeI4Mfezqf4CKqoZKR/9wBjBA7nx7YsRqDYNzX/0BztrDpXQ4EybF9s0r36gitpy8YSHiORJfj
  File size:  7789
  File index: 17
ERROR: Unlink 1 failed: 28
  File name:  /mnt/nxffs/GZ0LvjwIHYnP3JtbMZ0OXyjgzwzAminV9BKRFewF/IlI1hLTW7n/hu7H9N02mVWR4cVzjdPAgx/kJZIL7SYN3EZrIqc0J5NRsEkmOZi/qiodV7
  File size:  3607
  File index: 18
ERROR: Unlink 1 failed: 28
  File name:  /mnt/nxffs/HJE1CxiqBFnzeuay1dqBakh9/NOsOu5dCw30U7LwGCztV2UKhCtYi3Wf9nLD1ze3UuTBS2LPean5cWe89eZmJPSABz
  File size:  3323
```

btashton, 2020-11-11:
This should not have been closed quite yet.

btashton, 2020-11-15:
I think there are very bad things happening here.  I cleaned up some configuration settings so the initial file writing works, but I am seeing what looks like inodes being overwritten.

We can see `EVwni0b4` verified, but after the next round of files are written it seems to have been overwritten.

```
Directory:
 1. Type[8]: File  Name: EVwni0b4
 2. Type[8]: File  Name: iDrgzlIPo4OF
 3. Type[8]: File  Name: Jr7euPt7BEA5R
 4. Type[8]: File  Name: txgGQpEb
 5. Type[8]: File  Name: 5lCi
 6. Type[8]: File  Name: PVd33
 7. Type[8]: File  Name: 8An2DVQZMEHq75SV9uzh
 8. Type[8]: File  Name: g0x7Sj4a8Vw1D0g6qmq
 9. Type[8]: File  Name: olO4Qw5g8KLnuD2DR3jZ
10. Type[8]: File  Name: QctLaNi7rqQi
11. Type[8]: File  Name: GXPLOKsvH0VAw
12. Type[8]: File  Name: aqPFCVxkRabuT
13. Type[8]: File  Name: m
14. Type[8]: File  Name: SUR5KNn9hLHhMEq2Bav
15. Type[8]: File  Name: c3s50WlUBrMjfvm7I6
16. Type[8]: File  Name: 8a
17. Type[8]: File  Name: pFM1IwB6GmecRJATcrbT
18. Type[8]: File  Name: WyD7w
19. Type[8]: File  Name: fD29PA0Z793z1eBJ
20. Type[8]: File  Name: l0TBNGiFmhLeSa
21. Type[8]: File  Name: K9UG0W0pUiI
22. Type[8]: File  Name: mDP5chi
23. Type[8]: File  Name: ufAsEPLpg
24. Type[8]: File  Name: FF9ZJ
25. Type[8]: File  Name: SY0qvj4Do8jsL3upQHIM
26. Type[8]: File  Name: pMZ5Ren
  Verifed file /mnt/nxffs/EVwni0b4
Deleted file 1 OK
Deleted file 2 OK
Deleted file 3 OK
Deleted file 4 OK
Deleted file 5 OK
Deleted file 6 OK
  Verifed file /mnt/nxffs/iDrgzlIPo4OF
Deleted file 8 OK
Deleted file 9 OK
Deleted file 10 OK
Deleted file 11 OK
Deleted file 12 OK
Deleted file 13 OK
Deleted file 14 OK
Deleted file 15 OK
Deleted file 16 OK
Deleted file 17 OK
Deleted file 18 OK
Deleted file 19 OK
Deleted file 20 OK
Deleted file 21 OK
Deleted file 22 OK
Deleted file 23 OK
Deleted file 24 OK
Deleted file 25 OK
Deleted file 26 OK
Deleted file 27 OK
Deleted file 28 OK
Deleted file 29 OK
Deleted file 30 OK
Deleted file 31 OK
Deleted file 32 OK
Deleted file 33 OK
Deleted file 34 OK
Deleted file 35 OK
Deleted file 36 OK
Deleted file 37 OK
[... 191 lines cut ...]
41. Type[8]: File  Name: FA0frl9XCfVer
42. Type[8]: File  Name: e2c547
43. Type[8]: File  Name: LOX
44. Type[8]: File  Name: orkByhdTUvnGLoO
45. Type[8]: File  Name: A9eABe39d52Rj
46. Type[8]: File  Name: 41BD3vzzNn1AiVfA
47. Type[8]: File  Name: DhZ5bn
48. Type[8]: File  Name: JWlCZ5WnW2yN0FJqo
49. Type[8]: File  Name: j1PojXROB
50. Type[8]: File  Name: MK1bz
51. Type[8]: File  Name: 1C1V2pv
52. Type[8]: File  Name: Tf0flG00CHZHf
53. Type[8]: File  Name: Mq0tjUw0SI0
54. Type[8]: File  Name: 9y2ohs1rSvNbT55k3b0I
55. Type[8]: File  Name: N62fy
ERROR: Failed to open file for reading: 2
  File name: /mnt/nxffs/EVwni0b4
  File size: 6394
ERROR: Failed to read a file: 0
  File name: /mnt/nxffs/EVwni0b4
  File size: 6394
ERROR: Failed to verify files
  Number of files: 100
  Number deleted:  0
```

btashton, 2020-11-15:
This is also highly suspicious `EVwni0b4` should be the first inode.  It looks like this is getting totally corrupted when the new files are written.
Before the writing of the new files
```
NXFFS Dump:
  BLOCK:OFFS  TYPE  STATE   LENGTH
      0:0     BLOCK ERASED    512
      1:5     INODE OK       6394
      1:39    DATA  OK        463
      2:5     DATA  OK        497
      3:5     DATA  OK        497
```
After the writing of the new files
```
NXFFS Dump:
  BLOCK:OFFS  TYPE  STATE   LENGTH
      0:5     INODE OK       3283
      0:32    DATA  OK        470
      1:5     DATA  OK        497
      2:5     DATA  OK        497
      3:5     DATA  OK        497
      4:5     DATA  OK        497
      5:5     DATA  OK        497
      6:5     DATA  OK        117
      6:132   DATA  OK        211
```

btashton, 2020-11-15:
When nxffs_mediacheck is called on block 0 and the filesystem looks like this, it will fail thinking that there are no inodes.  This is not correct.  It kind of looks like `nxffs_nextentry` is hitting that empty space and not looking far enough to see block 1 before giving up and saying the rest of the flash must be zero.

```
NXFFS Dump:
  BLOCK:OFFS  TYPE  STATE   LENGTH
      0:0     BLOCK ERASED    512
      1:5     INODE DELETED  6394
      1:39    DATA  OK        463
      2:5     DATA  OK        497
      3:5     DATA  OK        497
      4:5     DATA  OK        497
      5:5     DATA  OK        497
      6:5     DATA  OK        497
      7:5     DATA  OK        497
      8:5     DATA  OK        497
      9:5     DATA  OK        497
     10:5     DATA  OK        497
     11:5     DATA  OK        497
     12:5     DATA  OK        497
     13:5     DATA  OK        464
     13:479   INODE DELETED   339
```
If I force `nxffs_mediacheck` to start at block 1 instead of 0 this will pass.

patacongo, 2020-11-15:
I am not sure if this related, but there is documented SPIFFS bug in the TODO list (should be a GIT issue):

```
  Title:       SILENT SPIFFS FILE TRUNCATION
  Description: Under certain corner case conditions, SPIFFS will truncate
               files.  All of the writes to the file will claim that the
               data has been written but after the file is closed, it may
               be a little shorter than expected.

               This is due to how the caching is implemented in SPIFFS:

               1. On each write, the data is not written to the FLASH but
                  rather to an internal cache in memory.
               2. When the a write causes the cache to become full, the
                  content of cache is flushed to memory.  If that flush
                  fails because the FLASH has become full, write will
                  return the file system full error (ENOSPC).
               3. The cache is also flushed when the file is closed (or
                  when fsync() is called).  These will also fail if the
                  file system becomes full.

               The problem is when the file is closed, the final file
               size could be smaller than the number of successful writes
               to the file.

               This error is probably not so significant in a real world
               file system usage:  It requires that you write continuously
               to SPIFFS, never deleting files or freeing FLASH resources
               in any way.  And it requires the unlikely circumstance that
               the final file written has its last few hundred bytes in
               cache when the file is closed but there are even fewer bytes
               available on the FLASH.  That would be rare with a cache
               size of a few hundred bytes and very large serial FLASH.

               This issue does cause the test at apps/testing/fstest to
               fail.  That test fails with a "Partial Read" because the
               file being read is smaller than number bytes written to the
               file.  That test does write small files continuously until
               file system is full and even the the error is rare.  The
               boards/sim/sim/sim/configs/spiffs test can used to
               demonstrate the error.
  Status:      Open
  Priority:    Medium.  It is certain a file system failure, but I think that
               the exposure in real world uses cases is very small.
```

I think that the real world exposure of this problem is rare since it depends on a condition where the physical media becomes full while there is still cached data.  However, the spiffs test does show this failure.  It would probably go away with a different random seed, but I wanted to keep the problem in place and repeatable so that it could be fixed sometime.

I think that the fix would be to reserve sectors that are at least the size of the cache so that this condition could not occur.

patacongo, 2020-11-15:
This bug is in the SPIFFS base code.  It would be interesting to see if they solved it.  The SPIFFS project has gone a different way, however, now focusing on security.  So nothing would be easily backported.

btashton, 2020-11-15:
I think the issue is different. This happens when pack is required, it does not find the first inode because it finds all the empty flags in the first block.  Shouldn't it be looking at the start of blocks not just incrementing byte by byte? 

This only happens once we have written and deleted enough that we need to circle back to the beginning to reclaim space marked deleted.

btashton, 2020-11-15:
I will have to look later today, but this seems suspect https://github.com/apache/incubator-nuttx/pull/1431  That might explain why we have the empty block in the first place.

btashton, 2020-11-19:
Looks like my analysis was correct here.  The linked PR did cause the first block to be skipped when formatting the filesystem which caused the pack routine to fail.  That PR needs to be reverted or fixed, this will cause corruption of the filesystem.

patacongo, 2020-11-19:
> 
> 
> Looks like my analysis was correct here. The linked PR did cause the first block to be skipped when formatting the filesystem which caused the pack routine to fail. That PR needs to be reverted or fixed, this will cause corruption of the filesystem.

These sim tests are very valuable.  We should encourage a suite of sim tests to verify all major features.  That would be a first step toward some automated regression testing.

Regression testing is sorely needed.  NuttX is gaining a reputation for being unstable.  Alan pointer this out to me:  https://www.embarcados.com.br/blink-led-no-esp32-com-o-rtos-nuttx/#comment-10549  It is in Portuguese, but translates to:

"Professional readers who may be curious about using NuttX, don't bother.  The latest versions are fundamentally broken, the howtos are out of date.  The email list is practically dead… it's a shame because it was a very interesting concept .. a posix in a small footprint..."

patacongo, 2020-11-19:
What would it take to automate sim:nxffs in CI?  Would it be (1) building the sim, (2) running the test, and (2) checking for a successful result in the output?

btashton, 2020-11-19:
Yeah this is something that I mentioned in an email a while back about the 10.0.0 release. I wrote some scripting already using the Robot Framework that starts to run some of these tests and it's my intention to have it in place for the next release. We also have the LTP work that I intended to wire up.

This is actually one of the reasons that the release got delayed, I ran all the sim configurations as well as all the configurations for multiple boards and found multiple issues. It's my intention to run all these again this weekend as well as the 15 or so supported boards that I have. There is a spreadsheet I shared that I used.

Getting these to be part of that suite of tests is also something I want to get working as well.

Additionally I have a webapp that I started (it's in my GitHub) that takes the build artifacts from our CI runs and makes it really easy for people to test them and submit an issue against a given build. I'm hoping that this will make it easy for people to periodically run a quick test against a board they have.  It also will do some automatic work like keeping track of the build size over time.

closed 2020-11-20 by xiaoxiang781216