apache/nuttx #2093: sim:nxwm Test failure

btashton, 2020-10-24:
For the `sim:nxwm` target the simulation crashes
```
❯ ./nuttx 
nxwm_main: Initialize the NSH library
ERROR: Failed to register oneshot at /dev/oneshot: -17
createTaskbar: Create CTaskbar instance
createTaskbar: Connect CTaskbar instance to the NX server
createTaskbar: Initialize CTaskbar instance
Segmentation fault (core dumped)
```

xiaoxiang781216, 2020-10-25:
May relate to my timer reconstruction, let's take a look.

sebastianene07, 2020-10-31:
I investigated this a little bit, here are some logs:

`Program received signal SIGSEGV, Segmentation fault.
                                                    0x00005555555920b5 in nxmu_sendwindow (wnd=0x0, msg=0x7ffff3ccfdb0, msglen=16) at nxmu/nxmu_sendwindow.c:87
87	  if (!NXBE_ISBLOCKED(wnd))
(gdb) bt
#0  0x00005555555920b5 in nxmu_sendwindow (wnd=0x0, msg=0x7ffff3ccfdb0, msglen=16) at nxmu/nxmu_sendwindow.c:87
#1  0x000055555559248d in nx_getposition (hwnd=0x0) at nxmu/nx_getposition.c:94
#2  0x000055555556fe13 in NXWidgets::CWidgetControl::waitGeoData (this=0x7ffff3cd5038) at src/cwidgetcontrol.cxx:950
#3  0x00005555555a8ec8 in NXWidgets::CWidgetControl::getWindowSize (this=0x7ffff3cd5038, size=0x7ffff3ccfe6c)
    at /home/seb/repos/incubator-nuttx-apps/include/graphics/nxwidgets/cwidgetcontrol.hxx:730
#4  0x000055555556d367 in NXWidgets::CNxWindow::getSize (this=0x7ffff3cd51b0, pSize=0x7ffff3ccfe6c) at src/cnxwindow.cxx:165
#5  0x0000555555564a3a in NxWM::CTaskbar::createBackgroundImage (this=0x7ffff3cd01f0) at src/ctaskbar.cxx:1001
#6  0x0000555555563dad in NxWM::CTaskbar::initWindowManager (this=0x7ffff3cd01f0) at src/ctaskbar.cxx:224
#7  0x000055555556359a in createTaskbar () at src/nxwm_main.cxx:181
#8  0x000055555556395c in nxwm_main (argc=1, argv=0x7ffff3ccff60) at src/nxwm_main.cxx:524
#9  0x000055555556110a in nxtask_startup (entrypt=0x5555555638fd <nxwm_main(int, char**)>, argc=1, argv=0x7ffff3ccff60)
    at sched/task_startup.c:165
#10 0x000055555555d34f in nxtask_start () at task/task_start.c:144
#11 0x0000000000000000 in ?? ()
(gdb) f 0
#0  0x00005555555920b5 in nxmu_sendwindow (wnd=0x0, msg=0x7ffff3ccfdb0, msglen=16) at nxmu/nxmu_sendwindow.c:87
87	  if (!NXBE_ISBLOCKED(wnd))
(gdb) p wnd
$1 = (struct nxbe_window_s *) 0x0
`

sebastianene07, 2020-10-31:
I have a fix for this

xiaoxiang781216, 2020-10-31:
@sebastianene07 thanks to find the fix.

btashton, 2020-11-01:
This is not fully resolved until https://github.com/apache/incubator-nuttx-apps/pull/456
Is merged correct?

sebastianene07, 2020-11-01:
> This is not fully resolved until [apache/incubator-nuttx-apps#456](https://github.com/apache/incubator-nuttx-apps/pull/456)
> Is merged correct?

I confirm

btashton, 2020-11-02:
App side backport that will close this issue out https://github.com/apache/incubator-nuttx-apps/pull/458

@sebastianene07 thanks for taking care of this one.

PR #2185 (2020-10-31): arch/sim: Update defconfig for NXWM and disable CONFIG_NSH_ARCHINIT
fixes #2093 

 ## Summary of changes

This patch addresses the first issue described in the linked ticket:
- prevent the oneshot registration error to be shown up at bootup
- the second patch against `incubator-nuttx-apps` fixes the issue with the crash: https://github.com/apache/incubator-nuttx-apps/pull/456

Disable CONFIG_NSH_ARCHINIT to prevent the boardctl() logic to call
multiple times the oneshot_register() function. The boardctl logic is
already called from nxwm_main.

## Testing

Verified that the error message:`ERROR [... cut]
closed 2020-11-05 by btashton