apache/nuttx #2094: sim:rpserver segfault

btashton, 2020-10-24:
The `sim:rpserver` configuration segfaults
```
❯ sudo ./nuttx
Segmentation fault
```

This prevents also testing the `sim:rpproxy` configuration.

xiaoxiang781216, 2020-10-25:
This is regressed by the change: https://github.com/apache/incubator-nuttx/pull/650. CONFIG_SIM_M32=y could fix this problem, but the change can't pass macOS build. We need enhance testbuild.sh and datlist to skip some special config on a particular host.

btashton, 2020-10-25:
Is there a reason we know why it only works in 32bit mode?

xiaoxiang781216, 2020-10-25:
This is the initial remoteproc design limitation:
https://lists.openampproject.org/pipermail/tsc/2019-October/000009.html
Since remoteproc/rpmsg is deployed on many OS(Linux, NuttX, Zephyr, FreeRTOS...), OpenAMP community need time to find a good mitigation path.

btashton, 2020-10-26:
@xiaoxiang781216 If you can look at my PR over in the test repo, with that we can disable the rpserver/proxy targets as disabled for macOS and then change the config to M32=y which should resolve this issue.  The PR does not yet add these targets to the tweak list for macOS, so let me know if you want me to make that change.

closed 2020-10-28 by btashton