apache/nuttx #2096: sim:smp Self Test Failure

btashton, 2020-10-24:
The `sim:smp` configuration is frequently failing:

```
❯ ./nuttx 

NuttShell (NSH) NuttX-9.1.0
nsh>smp
  Main[0]: Running on CPU0
  Main[0]: Initializing barrier
  Main[0]: Thread 1 created
  Main[0]: Now running on CPU1
Thread[2]: Started
  Main[0]: Thread 2 created
Thread[2]: Running on CPU1
  Main[0]: Now running on CPU2
Thread[3]: Started
<<LOCKED>>

❯ ./nuttx 

NuttShell (NSH) NuttX-9.1.0
nsh>smp
  Main[0]: Running on CPU0
  Main[0]: Initializing barrier
  Main[0]: Thread 1 created
  Main[0]: Now running on CPU1
Thread[2]: Started
  Main[0]: Thread 2 created
Thread[2]: Running on CPU1
  Main[0]: Now running on CPU2
Thread[3]: Started
  Main[0]: Thread 3 created
Thread[3]: Running on CPU2
  Main[0]: Now running on CPU3
Thread[4]: Started
Thread[4]: Running on CPU3
  Main[0]: Thread 4 created
  Main[0]: Now running on CPU2
Thread[5]: Started
Thread[5]: Running on CPU3
Thread[4]: Now running on CPU1
  Main[0]: Thread 5 created
Thread[2]: Now running on CPU3
  Main[0]: Now running on CPU3
Thread[5]: Now running on CPU1
Thread[6]: Started
Thread[4]: Now running on CPU2
  Main[0]: Thread 6 created
Thread[6]: Running on CPU2
Thread[2]: Now running on CPU1
  Main[0]: Now running on CPU2
Thread[5]: Now running on CPU3
Thread[4]: Now running on CPU3
Segmentation fault (core dumped)

❯ ./nuttx 

NuttShell (NSH) NuttX-9.1.0
nsh>smp
  Main[0]: Running on CPU0
  Main[0]: Initializing barrier
  Main[0]: Thread 1 created
  Main[0]: Now running on CPU1
Thread[2]: Started
  Main[0]: Thread 2 created
Thread[2]: Running on CPU1
  Main[0]: Now running on CPU2
Thread[3]: Started
  Main[0]: Thread 3 created
Thread[3]: Running on CPU2
  Main[0]: Now running on CPU3
Thread[4]: Started
Thread[4]: Running on CPU3
  Main[0]: Thread 4 created
  Main[0]: Now running on CPU2
Thread[5]: Started
Thread[2]: Now running on CPU3
  Main[0]: Thread 5 created
Thread[5]: Running on CPU1
<<BLOCKED>>
```

btashton, 2020-10-24:
@masayuki2009 You have been doing a lot of work with the SMP support can you take a look?

masayuki2009, 2020-10-24:
> @masayuki2009 You have been doing a lot of work with the SMP support can you take a look?

@btashton 
I finally found that the following PR caused this issue.
https://github.com/apache/incubator-nuttx/pull/1914
Also, I confirmed that hello, smp, and ostest work by reverting the commit.

@xiaoxiang781216 
Do you have any comments on this issue and the PR #1914

masayuki2009, 2020-10-24:
According to the PR #1914, it seems that sim/ostest was tested.
However, the PR affects smp as well, because up_simsmp.c was modified.

sebastianene07, 2020-10-25:
> > @masayuki2009 You have been doing a lot of work with the SMP support can you take a look?
> 
> @btashton
> I finally found that the following PR caused this issue.
> #1914
> Also, I confirmed that hello, smp, and ostest work by reverting the commit.
> 
> @sebastianene07 @xiaoxiang781216
> Do you have any comments on this issue and the PR #1914

Thanks for bisecting this. I think the issue is from some other place but the quick solution is to revert this.

masayuki2009, 2020-10-25:
>Thanks for bisecting this. I think the issue is from some other place but the quick solution is to revert this.

@sebastianene07 
I will create a new PR to revert #1914 .

PR #2112 (2020-10-26): [BACKPORT #2109] Revert "arch/sim: Make the SIGUSR1 host signal to use the NuttX irq logic
##  Summary
This is a backport of #2109 which should resolve #2096 

This reverts commit d6210fcd84dfd7b3bc3bac6cd6d3ee00c4f92239.

## Impact
SMP should be functional in the simulator again

## Testing
sim:smp  (ostest, smp)
closed 2020-10-26 by btashton