apache/nuttx #2099: qemu-i486 build broken

btashton, 2020-10-24:
The `qemu-i486:nsh` and `qemu-i486:ostest` configurations do not seem to build any more due to a linker error looking for `libm`

```
CC:  qemu_boot.c
CC:  qemu_appinit.c
AR (create): libboard.a   qemu_boot.o qemu_appinit.o 
make[2]: Leaving directory '/home/bashton/nuttx/apache/incubator-nuttx/boards/x86/qemu/qemu-i486/src'
LD: nuttx.elf
ld: cannot find -lm
make[1]: *** [Makefile:135: nuttx.elf] Error 1
make[1]: Leaving directory '/home/bashton/nuttx/apache/incubator-nuttx/arch/x86/src'
make: *** [tools/Makefile.unix:415: nuttx.elf] Error 2
```

masayuki2009, 2020-10-25:
@btashton 

Oh really?
I can build qemu-i486:nsh and qemu-i486:ostest on ubuntu18.04 x86_64.

```
make V=1
...
LD: nuttx.elf
ld --entry=__start -m elf_i386 -T/mnt/m2ssd/opensource/RTOS/nuttx-releases/nuttx/boards/x86/qemu/qemu-i486/scripts/qemu.ld -static -L"/mnt/m2ssd/opensource/RTOS/nuttx-releases/nuttx/staging" -L"/mnt/m2ssd/opensource/RTOS/nuttx-releases/nuttx/arch/x86/src/board" -L "/usr/lib/gcc/x86_64-linux-gnu/7/32/" -L "/usr/lib/gcc/x86_64-linux-gnu/7/../../../../lib32/" \
	-o "/mnt/m2ssd/opensource/RTOS/nuttx-releases/nuttx/nuttx.elf" qemu_head.o  \
	--start-group -lsched -ldrivers -lboards -lc -lmm -larch -lapps -lfs -lbinfmt -lboard -lgcc -lm --end-group
nm "/mnt/m2ssd/opensource/RTOS/nuttx-releases/nuttx/nuttx.elf" | \
grep -v '\(compiled\)\|\(\.o$\)\|\( [aUw] \)\|\(\.\.ng$\)\|\(LASH[RL]DI\)' | \
sort > /mnt/m2ssd/opensource/RTOS/nuttx-releases/nuttx/System.map
make[1]: Leaving directory '/mnt/m2ssd/opensource/RTOS/nuttx-releases/nuttx/arch/x86/src'
if [ -w /tftpboot ] ; then \
	cp -f nuttx.elf /tftpboot/nuttx.elf.x86; \
fi
```

btashton, 2020-10-25:
I'll try building it again with verbose logs, but it did not work correctly for me in either config.

btashton, 2020-10-26:
@masayuki2009 
```
LD: nuttx.elf
ld --entry=__start -m elf_i386 -g -T/home/bashton/nuttx/apache/incubator-nuttx/boards/x86/qemu/qemu-i486/scripts/qemu.ld -static -L"/home/bashton/nuttx/apache/incubator-nuttx/staging" -L"/home/bashton/nuttx/apache/incubator-nuttx/arch/x86/src/board" -L "/usr/lib/gcc/x86_64-redhat-linux/10/32/" -L "./" \
	-o "/home/bashton/nuttx/apache/incubator-nuttx/nuttx.elf" qemu_head.o  \
	--start-group -lsched -ldrivers -lboards -lc -lmm -larch -lapps -lfs -lbinfmt -lboard -lgcc -lm --end-group
ld: cannot find -lm
make[1]: *** [Makefile:135: nuttx.elf] Error 1
make[1]: Leaving directory '/home/bashton/nuttx/apache/incubator-nuttx/arch/x86/src'
make: *** [tools/Makefile.unix:415: nuttx.elf] Error 2
```
Do we really want to be using the system math library here?  That seems wrong to me.

btashton, 2020-10-26:
Ok looks like I was wrong here and it is intentional.  Including the `glibc-static.i686` fixes this, I did not even have the 64-bit version on my system, for the x86_64 port we have `LIBM=y` so this is not required.

closed 2020-10-26 by btashton