apache/nuttx #2102: stm32h7: Incorrect voltage output scale configuration

btashton, 2020-10-25:
There is an issue with the way the voltage output scale configuration is handled for the stm32h7.

In `stm32h7x7xx_rcc.c`  the code block is `#if 0` disabled, but it is still enabled in the  `stm32h7x3xx_rcc.c`.
This causes the nucleo-h743(5) to frequently hang on
`while ((getreg32(STM32_PWR_D3CR) & STM32_PWR_D3CR_VOSRDY) == 0)`

When this block is disabled as is done in the stm32H7x7xx_rcc the board boots correctly.  This power logic should be resolved.

btashton, 2020-10-25:
@davids5 Looks like you had added this logic almost a year ago.

btashton, 2020-10-25:
Hmm the board I was using is actually the h745 which is technically different, this power section still looks suspect to me and we should make sure we handle it correctly across the full family.

davids5, 2020-10-25:
@btashton - wow! Hats off to you for all the release efforts and all the testing! It is very impressive!

What HW is the H745 on? How it is powered?

I recall an issues with powering some of these devices on some of the Necleo dev boards.  

We have, I think 4 different HW platforms with H7{4|5}3 under test on the PX4 HW test rack and no issues, with RCC start up. that I am aware of.  

I do not have any HW with H777 to test with so I am not sure why the code is if def-ed out. 

If I can buy the same HW you have online I can see what is happening with it.

raiden00pl, 2020-10-25:
@btashton Could you please provide more details? 
I just tested the nucleo-h743zi/nsh configuration from master on nucleo-h745zi-q and nucleo-h743zi boards and it boots correctly.

raiden00pl, 2020-10-25:
Maybe the problem lies with the ST-LINK-V3 which is present on nucleo-h745zi-q. I noticed that it doesn't want to work properly when the USB cable is too long, which was never a problem with older ST-LINK versions.

raiden00pl, 2020-10-25:
OK, I switched to my previous setup with long usb cable connected to ST-LINK-V3 and I was able to reproduce the problem. Then I switched to the setup that previously worked and ... it no longer works :)

btashton, 2020-10-25:
> OK, I switched to my previous setup with long usb cable connected to ST-LINK-V3 and I was able to reproduce the problem. Then I switched to the setup that previously worked and ... it no longer works :)

This is exactly what I saw. It worked then I built the pwm config and then they both did not work.

davids5, 2020-10-25:
> This is exactly what I saw. It worked then I built the pwm config and then they both did not work.

 Is this the issue that's noted in the code that it requires of power cycle?

btashton, 2020-10-25:
> Is this the issue that's noted in the code that it requires of power cycle?

Power cycling, reflashing, changing USB ports, none of this made it work again. The original test was on a thin USB cable that did drop voltage but I could not get it working again even with a short fat cable on my quality powered hub _and_ made the code change.  The STLINK3 was showing 3.28V

davids5, 2020-10-25:
@btashton @raiden00pl - what HW are you testing on?

btashton, 2020-10-25:
NUCLEO-H745ZI-Q

raiden00pl, 2020-10-25:
This issue only affects NUCLEO-H745ZI-Q.
NUCLEO-H743ZI works fine for me.

There is no support for H745ZI in the current upstream, and we shouldn't assume that H743ZI configuration will work for it.

btashton, 2020-10-25:
Yeah I realized that after I started testing, but then I saw that we have the logic if 0 in the other rcc (which should be removed or fixed) so I was less convinced, but you are right I need to compare the reference manual.

btashton, 2020-11-22:
I am going to close this out.  I have been banging on the `NUCLEO-H743ZI` all day and have not had any issues.  There likely is some incompatibility.

closed 2020-11-22 by btashton