apache/nuttx #2104: avr: teensy 2.0 Non-Functional

btashton, 2020-10-25:
The AVR Teensy 2.0 configuration  (the only AVR target I have) does not seem functional.  The two targets that supposedly should work are:

* hello --  `File "nuttx.hex". 30828 bytes, 96% used`  This seems really high, and I do not get anything on the UART
* nsh -- `File "nuttx.hex". 53729 bytes, 167% used`  Clearly non-functional.

We really need to decide if we keep these targets or not, because right now we are keeping them building but not making sure they are functional. I worry that the `hc` targets are in a similar place, there is some nostalgia for me to that port even though I have not used it with NuttX because the hcs12 was my first microcontroller, but I am not going to take ownership of that.

btashton, 2020-10-25:
@patacongo I know this is going to frustrate you, but maybe you can help debug this?  I don't know what AVR hardware you might still have.

patacongo, 2020-10-25:
I had a medical  emergency  last week am still in the hospital.  I will not be able to help now.  Sorry.Sent from Samsung tablet.

Ouss4, 2020-10-25:
We might get this to work if we temper with tasks' stacks and disable feature that are not needed for a basic NSH bring up.

patacongo, 2020-10-25:
I had a medical emergency last week am still in the hospital.  I will not be able to help now.  Sorry.There may be options to reconfigure Harvard memory usage, moving strings into program memory.  Or maybe we don't need Te3nsy-2.0 anymore?Sent from Samsung tablet.

patacongo, 2020-10-25:
There was also a PR from  Xiao Xiang the removed and reorganized support for progmem stings.  Of course that was never testedSent from Samsung tablet.

btashton, 2020-10-25:
@patacongo take care of yourself Greg this is not urgent. I'll see if I can figure out a little more about what is going on. Maybe I'll track down some other AVR hardware that I can actually debug. This was just convenient and cheep.