apache/nuttx #2163: sim: terminal is garbled when nuttx is killed

v01d, 2020-10-29:
If you kill nuttx sim binary (even with SIGTERM), the terminal is garbled. Probably something needs to be done in some exit/signal handler. 

[image]

xiaoxiang781216, 2020-10-30:
This is because sim use atexit to restore the state:
https://github.com/apache/incubator-nuttx/blob/master/arch/sim/src/sim/up_simuart.c#L112
https://github.com/apache/incubator-nuttx/blob/master/arch/sim/src/sim/up_x11framebuffer.c#L262
which need change to signal handler, but I guess x11 can restore automatically without the special action.

xiaoxiang781216, 2021-03-26:
Should be fixed by: https://github.com/apache/incubator-nuttx/pull/2998. @v01d please try it.

v01d, 2021-03-26:
It would seem so, but I'm confused as to why that PR fixes it. As you mentioned before, there was cleanup required to be done on sim exit.

xiaoxiang781216, 2021-03-26:
> It would seem so, but I'm confused as to why that PR fixes it. As you mentioned before, there was cleanup required to be done on sim exit.

Since the PR doesn't disable OPOST, \n will always translate to \r\n.

v01d, 2021-03-26:
Ah, I understand: it was garbled due to changing this bit from default. I assume other changes do not require any undoing?

v01d, 2021-04-22:
I think that this needs further work since I find now that CTRL+C does not work anymore when nuttx is killed. The cleanest would probably be to get the flags before sim starts and then reapply them in the end.

xiaoxiang781216, 2021-04-23:
Yes, the input process is turned off, we need to find a reliable way to restore it.  #2998 just fix the output:(.

v01d, 2021-04-23:
I'll reopen for tracking