apache/nuttx #2197: tms570 problems

longhuarst, 2020-11-02:
is newest nuttx supoort tms570 series?
when i use buildroot as the toolchain, when i run make command, i got error info as follows:

/nuttx/staging/libfs.a(fs_munmap.o): compiled for a big endian system and target is little endian

how can i solve this problems?

acassis, 2020-11-04:
Hi @longhuarst the tms570 port was done a long time ago. But I didn't see nobody interested on it. Only the basic port it present. It means that if you want SPI, I2C, etc you need to add the driver. Are you brave enough? ;-)
I think TMS570 is interesting because it is a of few Big Endian arch supported by NuttX.

longhuarst, 2020-11-15:
> Hi @longhuarst the tms570 port was done a long time ago. But I didn't see people interested on it. Only the basic port it present. It means that if you want SPI, I2C, etc you need to add the driver. Are you brave enough? ;-)
> I think TMS570 is interesting because it is one of the few Big Endian arch supported by NuttX.

i use tms570 with ti ccs for a long time， the HALCOGEN tool can generator code for freertos， but i want to run nuttx on tms570，i use the toolchain build by ti， and can nuttx use the toolchain build by ti？