apache/nuttx #2203: Hardfault on boot when only usb console is used

lukegluke, 2020-11-02:
I've moved from NuttX 8.2 to NuttX 10.0.0.  If uart console is used booting is ok. But if only usb console is used (CONFIG_CDCACM_CONSOLE) I get hardfault on each boot.

Digging in debug showed that it occurs on first use of stdout or stderr!
As I see with commit b0797263ca2817cf536be8b843ac5b5aac5d4d9c stdout and stderr are now gotten from dynamic stream list
https://github.com/apache/incubator-nuttx/blob/d987dd2f5aa5d26f814c66d01eb4509b1efb1672/include/stdio.h#L81-L83
 I supposed due to latter usb console initialization in my case `nxsched_get_streams()->sl_head` is null, so `nxsched_get_streams()->sl_head->fs_next` causes hardfault.

stdout, stderr could be easily used in board_app_initialize, but stderr is certainly used in

apps/nshlib/nsh_console.c:
https://github.com/apache/incubator-nuttx-apps/blob/3b884cfb870d21d02542d7e8434d286228e359f6/nshlib/nsh_console.c#L510

on usb console initialization in apps/nshlib/nsh_usbconsole.c
https://github.com/apache/incubator-nuttx-apps/blob/3b884cfb870d21d02542d7e8434d286228e359f6/nshlib/nsh_usbconsole.c#L269

before `cdcacm_initialize()` is called. I haven't dig further, but I suppose stream allocation is made with `/dev/console` registration, right?

(If it is matters, I'm working with custom board and firmware based on latest PX4. But issue definitely related to NuttX)

So it looks like deferred console is special case that should be treated carefully.
@xiaoxiang781216, hope for your help, thanks in advance.

xiaoxiang781216, 2020-11-03:
Do you have these patch which should fix you problem:
https://github.com/apache/incubator-nuttx/pull/1993
https://github.com/apache/incubator-nuttx/pull/1997
https://github.com/apache/incubator-nuttx-apps/pull/425

lukegluke, 2020-11-03:
> Do you have these patch which should fix you problem:

No, I didn't have these patches. Now I have, but it doesn't help.
As I see patches were aimed on fclose of the builtin streams, while in my case they are not yet allocated on init in nsh_newconsole and board_app_initialize.

xiaoxiang781216, 2020-11-03:
But, there is a special patch for usb console:
https://github.com/apache/incubator-nuttx-apps/pull/425/commits/2b9d283ed547a6bb6049f1142dea00772c6ac88b
@masayuki2009 hit the similar issue with stm32f4discovery:usbnsh and fixed by the above patch.

lukegluke, 2020-11-03:
> But, there is a special patch for usb console:
> [apache/incubator-nuttx-apps@2b9d283](https://github.com/apache/incubator-nuttx-apps/commit/2b9d283ed547a6bb6049f1142dea00772c6ac88b)
> @masayuki2009 hit the similar issue with stm32f4discovery:usbnsh and fixed by the above patch.

Patch is certainly here, but `nsh_configstdio` func is called further in `nsh_consolemain` func while hardfault is occurring in `nsh_newconsole`

xiaoxiang781216, 2020-11-03:
Here is a part of the task initialization process:
https://github.com/apache/incubator-nuttx/blob/master/sched/group/group_setupstreams.c#L71

lukegluke, 2020-11-03:
> > Here is a part of the task initialization process:
> https://github.com/apache/incubator-nuttx/blob/master/sched/group/group_setupstreams.c#L71

Thanks! In this place inside `fs_fdopen` the `fs_checkfd` func returns ENOENT, because inode is null.

davids5, 2020-11-03:
@lukegluke See https://github.com/PX4/PX4-Autopilot/blob/master/platforms/nuttx/src/px4/common/px4_init.cpp#L45 

This is how we fixed the changes in upstream NuttX that are caused by the dynamic allocation of the streams. When there is no console.

davids5, 2020-11-03:
The root cause in that the 3 fds are not initialized in sched then the stream leaves the stream initialized. 

See group_setupidlefiles.c 

If the CONFIG_DEV_CONSOLE is not defined 
https://github.com/apache/incubator-nuttx/blob/8c464b0e7a35bf6ae94fcecdccc31dd31af9cd29/sched/group/group_setupidlefiles.c#L98-L125

this https://github.com/apache/incubator-nuttx/blob/8c464b0e7a35bf6ae94fcecdccc31dd31af9cd29/sched/group/group_setupidlefiles.c#L129-L131

does nothing.

lukegluke, 2020-11-03:
> @lukegluke See https://github.com/PX4/PX4-Autopilot/blob/master/platforms/nuttx/src/px4/common/px4_init.cpp#L45
> 
> This is how we fixed the changes in upstream NuttX that are caused by the dynamic allocation of the streams. When there is no console.

Thanks for feedback, I have this code and call px4_platform_init in my board_app_initialize, but it doesn't get into px4_platform_console_init. CONFIG_DEV_CONSOLE is not defined. And CONFIG_DEV_NULL also, but according to configuration docs it is set by default, but looks like it doesn't. I'm trying to investigate this further.

lukegluke, 2020-11-03:
> @lukegluke See https://github.com/PX4/PX4-Autopilot/blob/master/platforms/nuttx/src/px4/common/px4_init.cpp#L45
> 
> This is how we fixed the changes in upstream NuttX that are caused by the dynamic allocation of the streams. When there is no console.

Thanks for feedback! In fact I already had this code and call px4_platform_init in my board_app_initialize. It was just my misconfiguration (I missed that CONFIG_DEV_CONSOLE is set by default, and I need to unset it explicitly). Now this workaround works for me.

But it is still the issue for NuttX itself. I suppose it is not a good thing to fall in hadrdfault if there is no console, or it is failed to open.

v01d, 2020-11-04:
As I suspected, I'm getting bit by the same bug. In my case I disabled console since I'm working with a device with inaccessible UART. The moment I do a printf() from my app it crashes. I'm finding that group->nxsched_get_streams also has a NULL head.
I'm not fully understanding if this is a recent problem or something that was always problematic. @davids5 you are using a workaround in PX4, was this reported here? @xiaoxiang781216 what can we do to fix this properly on NuttX?
Note that I'm using master

v01d, 2020-11-04:
Ok, I avoided the crash if I do enable DEV_CONSOLE but enable console->syslog redirection. It seems the problem is missing /dev/console. Maybe this shouldn't be an option?
BTW, the config is really confusing, at first glance it appears as if you would be able to make console go to syslog and make syslog go to console simultaneously. The symbol names are even more confusing: CONFIG_CONSOLE_SYSLOG and CONFIG_SYSLOG_CONSOLE. If you enable CONFIG_CONSOLE_SYSLOG, the other option is still selectable. Seems a `depends on` is missing.

xiaoxiang781216, 2020-11-04:
@v01d , Ok, I will provide a patch to preallocate stdin/stdout/stderr, so the printf don't crash at least.

v01d, 2020-11-04:
Great. Maybe this could be added as an option to have the console go to /dev/null. Don't know what is easier.

davids5, 2020-11-04:
> @davids5 you are using a workaround in PX4, was this reported here? @xiaoxiang781216 what can we do to fix this properly on NuttX?

@v01d Yes - years before ASF. But for a different reason. The one you found printf had a side effect. It would write to a file if  0,1,2 are not stdxxxx. Then 0,1,2 becomes the first, second and third file opened in a group. printf corrupts the file. @patacongo rejected my fix at the time. He expected that we should ifdef all the printfs. I disagreed and implemented what I posted.  

When https://github.com/apache/incubator-nuttx/pull/1755 came in it broke systems with out a /dev/console. as shown https://github.com/apache/incubator-nuttx/issues/2203#issuecomment-721146799 

All the nuttx bootloders in PX4 do not use a console and where hardfaulting in `sercon` 's printf.  

Since my concept was rejected, I saw no point in rehashing the issue.  

I agree @xiaoxiang781216 soult fix it properly and we need to test better (we being me too, i missed the bootloader test case when I approved the #1755) 

I agree that CONFIG_CONSOLE_* is confusing and we should have a doc that explains all the console options and what is mutually exclusive and why. There should also be code to enforce it.  I think this was cased by organic growth. Unfortunately I have lost context and would have to reverse engineer it and still might misconstrue the intent.

xiaoxiang781216, 2020-11-04:
Not really need, it is enough to initialize the preallocated FILE structure to zero, because stdio function will find the flag is invalid and return directly. On the other hand, /dev/null isn't reliable because user can disable /dev/null too.

davids5, 2020-11-04:
> Not really need, it is enough to initialize the preallocated FILE structure to zero, because stdio function will find the flag is invalid and return directly. On the other hand, /dev/null isn't reliable because user can disable /dev/null too.

@xiaoxiang781216 - that sounds like the way to go.  But please test it.

v01d, 2020-11-04:
I will test it as well in my config

closed 2020-11-11 by davids5

btashton, 2020-11-13:
I believe this backport PR resolve this issue on 10.0 branch https://github.com/apache/incubator-nuttx/pull/2293