apache/nuttx #2219: Recent modifications on testbuild.sh makes it difficult to know if the compilation succeed or failed

acassis, 2020-11-04:
Previously the NuttX compilation was finished clearly:

LD: nuttx
make[1]: Leaving directory 'xxxx'
CP: nuttx.hex
CP: nuttx.bin

Now it is completely messed:

LD: nuttx
make[1]: Leaving directory 'xxxx'
echo nuttx > nuttx.manifest
printf "%s\n" *.map >> nuttx.manifest
CP: nuttx.hex
echo nuttx.hex >> nuttx.manifest
CP: nuttx.bin
echo nuttx.bin >> nuttx.manifest

I suggest we don't show these garbage to the users.

Also should be interesting to show:
"NuttX was compiled with Success!"

This way the user will know everything worked correctly.

btashton, 2020-11-04:
I would not call it "completely messed" it's just the output of a build. It's also not part of testbuild it is the make files generating the manifest.

If you don't like the lines then just add a @ infront of them. We have lots of loud parts of the build besides this (like wget and curl which should be silent)

v01d, 2020-11-04:
Better to use $(Q) so that it can appear when building with V=1

acassis, 2020-11-04:
Hi @btashton, I agree that there are many parts that are verbose now by default. but this modification makes it hard to the end user know if everything compiled correct. My fault I just "git grep" nuttx.manifest. I will submit a patch adding $(Q) as @v01d suggested.

closed 2020-11-05 by btashton