apache/nuttx #2238: make menuconfig got a lot of syntax error.

yjdwbj, 2020-11-07:
Hi, guys. after run `git pull` then `make menuconfig`, got a lot of error like below:
```sh
arch/arm/src/imxrt/Kconfig:1212: syntax error
arch/arm/src/imxrt/Kconfig:1211: invalid option
arch/arm/src/stm32l4/Kconfig:5236: syntax error
arch/arm/src/stm32l4/Kconfig:5185: missing end statement for this entry
arch/arm/src/stm32l4/Kconfig:5053: missing end statement for this entry
arch/arm/src/stm32l4/Kconfig:8: missing end statement for this entry
arch/arm/Kconfig:1017: missing end statement for this entry
arch/arm/Kconfig:7: missing end statement for this entry
Kconfig:1794: missing end statement for this entry
arch/arm/src/stm32l4/Kconfig:5235: invalid statement
arch/arm/src/stm32l4/Kconfig:5245: unexpected end statement
arch/arm/src/stm32l4/Kconfig:5248: syntax error
arch/arm/src/stm32l4/Kconfig:5247: invalid statement
arch/arm/src/stm32l4/Kconfig:5257: unexpected end statement
arch/arm/src/stm32l4/Kconfig:5260: syntax error
arch/arm/src/stm32l4/Kconfig:5259: invalid statement
arch/arm/src/stm32l4/Kconfig:5269: unexpected end statement
arch/arm/src/stm32l4/Kconfig:5271: unexpected end statement
arch/arm/src/stm32l4/Kconfig:5273: unexpected end statement
arch/arm/src/stm32l4/Kconfig:6097: unexpected end statement
arch/arm/Kconfig:1018: unexpected end statement
arch/arm/Kconfig:1032: unexpected end statement
Kconfig:1796: unexpected end statement
make: *** [tools/Makefile.unix:492: olddefconfig] Error 1
ERROR: failed to refresh
```
I checked the error place, find the errors such as that ` if <param> = <num>`. how to resolve the problem, thanks.

yjdwbj, 2020-11-07:
Now . I using the `kconfig-mconf` also got the same error.

```sh
github/apache/nuttx$ kconfig-mconf Kconfig
Kconfig:8:warning: environment variable APPSDIR undefined
arch/arm/src/imxrt/Kconfig:1212: syntax error
arch/arm/src/imxrt/Kconfig:1211: invalid option
arch/arm/src/stm32l4/Kconfig:5236: syntax error
arch/arm/src/stm32l4/Kconfig:5185: missing end statement for this entry
arch/arm/src/stm32l4/Kconfig:5053: missing end statement for this entry
arch/arm/src/stm32l4/Kconfig:8: missing end statement for this entry
arch/arm/Kconfig:1017: missing end statement for this entry
arch/arm/Kconfig:7: missing end statement for this entry
Kconfig:1794: missing end statement for this entry
arch/arm/src/stm32l4/Kconfig:5235: invalid statement
arch/arm/src/stm32l4/Kconfig:5245: unexpected end statement
arch/arm/src/stm32l4/Kconfig:5248: syntax error
arch/arm/src/stm32l4/Kconfig:5247: invalid statement
arch/arm/src/stm32l4/Kconfig:5257: unexpected end statement
arch/arm/src/stm32l4/Kconfig:5260: syntax error
arch/arm/src/stm32l4/Kconfig:5259: invalid statement
arch/arm/src/stm32l4/Kconfig:5269: unexpected end statement
arch/arm/src/stm32l4/Kconfig:5271: unexpected end statement
arch/arm/src/stm32l4/Kconfig:5273: unexpected end statement
arch/arm/src/stm32l4/Kconfig:6097: unexpected end statement
arch/arm/Kconfig:1018: unexpected end statement
arch/arm/Kconfig:1032: unexpected end statement
Kconfig:1796: unexpected end statement
Kconfig:1859: can't open file "/Kconfig"
```
```

Ouss4, 2020-11-07:
Just pulled the latest upstream and built a couple of configurations with no errors.
What config are you trying to build?
Do you have all the dependencies?

yjdwbj, 2020-11-08:
> Just pulled the latest upstream and built a couple of configurations with no errors.
> What config are you trying to build?
> Do you have all the dependencies?

Hi, Ouss4, thanks you replay me.  I already known how to resolve the problem, I changed the target file  like below:

```sh
~$git diff arch/arm/src/stm32l4/Kconfig
diff --git a/arch/arm/src/stm32l4/Kconfig b/arch/arm/src/stm32l4/Kconfig
index e8a831cdac..af71b5d568 100644
--- a/arch/arm/src/stm32l4/Kconfig
+++ b/arch/arm/src/stm32l4/Kconfig
@@ -5232,7 +5232,7 @@ config STM32L4_ADC3_INJ_CHAN
        ---help---
                Number of configured ADC3 injected channels.
 
-if STM32L4_ADC1_INJ_CHAN > 0
+if STM32L4_ADC1_INJ_CHAN 
 
 config STM32L4_ADC1_JTIMTRIG
        int "ADC1 external trigger for injected channels"
@@ -5244,7 +5244,7 @@ config STM32L4_ADC1_JTIMTRIG
 
 endif
 
-if STM32L4_ADC2_INJ_CHAN > 0
+if STM32L4_ADC2_INJ_CHAN 
 
 config STM32L4_ADC2_JTIMTRIG
        int "ADC2 external trigger for injected channels"
@@ -5256,7 +5256,7 @@ config STM32L4_ADC2_JTIMTRIG
 
 endif
 
-if STM32L4_ADC3_INJ_CHAN > 0
+if STM32L4_ADC3_INJ_CHAN 
 
 config STM32L4_ADC3_JTIMTRIG
        int "ADC3 external trigger for injected channels"

```
I just want to known that use  express such as  `if STM32L4_ADC2_INJ_CHAN > 0` is wrong  or  my way is  wrong.

Ouss4, 2020-11-08:
What board are you trying to build?
I just tried one STM32L4 it's building fine.
Please do the following:
```
./tools/configure.sh -E stm32l476vg-disco:nsh
make menuconfig # Change whatever you want
make
```

yjdwbj, 2020-11-08:
my board is STM32F1/4/7 , I mean that after `git pull` will affect the original project configure.

raiden00pl, 2020-11-09:
You need to update your kconfig-frontends. 'if <param> = <num>' syntax doesn't work for older versions.

v01d, 2020-11-09:
> You need to update your kconfig-frontends. 'if = ' syntax doesn't work for older versions.

Yes, please follow the documentation on how to setup kconfig:

https://nuttx.apache.org/docs/latest/quickstart/quickstart.html

closed 2020-11-10 by yjdwbj

yjdwbj, 2020-11-26:
> You need to update your kconfig-frontends. 'if = ' syntax doesn't work for older versions.

my kconfig-frontends is already the newest version (4.11.0.1+dfsg-2).