apache/nuttx #2248: nucleo-144:f767-netnsh  no debug output

yjdwbj, 2020-11-08:
Hi, guys , I am test a nucleo-f767zi board, initialize  with  `tools/configure.sh  nucleo-144:f767-netnsh`, and enabled all of debug options , but not have any debug info output.  I hope someone who can fixed problem , Thanks.
```sh
~$ grep "DEBUG" .config
# CONFIG_ARCH_DEBUG_H is not set
CONFIG_DEBUG_ALERT=y
CONFIG_DEBUG_FEATURES=y
CONFIG_DEBUG_ERROR=y
CONFIG_DEBUG_WARN=y
CONFIG_DEBUG_INFO=y
CONFIG_DEBUG_ASSERTIONS=y
# CONFIG_DEBUG_BINFMT is not set
# CONFIG_DEBUG_CRYPTO is not set
# CONFIG_DEBUG_FS is not set
# CONFIG_DEBUG_GRAPHICS is not set
# CONFIG_DEBUG_LIB is not set
CONFIG_DEBUG_MM=y
CONFIG_DEBUG_MM_ERROR=y
# CONFIG_DEBUG_MM_WARN is not set
# CONFIG_DEBUG_MM_INFO is not set
# CONFIG_DEBUG_NET is not set
# CONFIG_DEBUG_POWER is not set
# CONFIG_DEBUG_SCHED is not set
CONFIG_DEBUG_IRQ=y
# CONFIG_DEBUG_IRQ_ERROR is not set
# CONFIG_DEBUG_IRQ_WARN is not set
# CONFIG_DEBUG_IRQ_INFO is not set
# CONFIG_DEBUG_LEDS is not set
# CONFIG_DEBUG_GPIO is not set
# CONFIG_DEBUG_MEMCARD is not set
CONFIG_DEBUG_SPI=y
CONFIG_DEBUG_SPI_ERROR=y
CONFIG_DEBUG_SPI_WARN=y
CONFIG_DEBUG_SPI_INFO=y
# CONFIG_DEBUG_TIMER is not set
CONFIG_DEBUG_SYMBOLS=y
# CONFIG_DEBUG_NOOPT is not set
# CONFIG_DEBUG_CUSTOMOPT is not set
CONFIG_DEBUG_FULLOPT=y
CONFIG_ARCH_HAVE_HARDFAULT_DEBUG=y
CONFIG_DEBUG_HARDFAULT_ALERT=y
CONFIG_DEBUG_HARDFAULT_INFO=y
CONFIG_ARCH_HAVE_MEMFAULT_DEBUG=y
# CONFIG_DEBUG_MEMFAULT is not set
# CONFIG_NETDEV_PHY_DEBUG is not set
# CONFIG_NET_TCP_WRBUFFER_DEBUG is not set
# CONFIG_NET_UDP_WRBUFFER_DEBUG is not set
# CONFIG_IOB_DEBUG is not set
# CONFIG_NETINIT_DEBUG is not set
```

btashton, 2020-11-08:
If I remember on this board with the configuration the link has to be up for it to get to the shell. Do you have ethernet cable plugged in? You are likely not seeing any debug info because you do not have the network debug flags turned on.

yjdwbj, 2020-11-08:
> If I remember on this board with the configuration the link has to be up for it to get to the shell. Do you have ethernet cable plugged in? You are likely not seeing any debug info because you do not have the network debug flags turned on.

Hi, Yes, I did have ethernet cache plugged in. and I got a nsh shell from the `ttyACM0`:

```sh
NuttShell (NSH) NuttX-9.1.0                                                                                                                                                                   
nsh>                                                                                                                                                                                          
nsh>                                                                                                                                                                                          
nsh> uname -a                                                                                                                                                                                 
NuttX stntest 9.1.0 5bb4eb39f2-dirty Sep 28 2020 18:33:47 arm nucleo-144                                                                                                                      
nsh> ifconfig                                                                                                                                                                                 
eth0    Link encap:Ethernet HWaddr 96:0f:40:c2:b0:45 at UP                                                                                                                                    
        inet addr:10.0.0.2 DRaddr:10.0.0.1 Mask:255.255.255.0                                                                                                                                 
        inet6 addr: 2409:8a55:2416:2500:940f:40ff:fec2:b045/64                                                                                                                                
        inet6 DRaddr: fe80::2276:93ff:fe00:ac08/64                                                                                                                                            
                                                                                                                                                                                              
lo      Link encap:Local Loopback at UP                                                                                                                                                       
        inet addr:127.0.0.1 DRaddr:127.0.0.1 Mask:255.0.0.0                                                                                                                                   
        inet6 addr: ::1/128                                                                                                                                                                   
        inet6 DRaddr: ::1/128                                                                                                                                                                 
                                                                                                                                                                                              
             IPv4  IPv6   TCP   UDP  ICMP  ICMPv6                                                                                                                                             
Received     0000  0004  0000  0000  0000  0001                                                                                                                                               
Dropped      0000  0003  0000  0000  0000  0000                                                                                                                                               
  IPv4        VHL: 0000   Frg: 0000                                                                                                                                                           
  IPv6        VHL: 0000                                                                                                                                                                       
  Checksum   0000  ----  0000  0000  ----  ----                                                                                                                                               
  TCP         ACK: 0000   SYN: 0000                                                                                                                                                           
              RST: 0000  0000                                                                                                                                                                 
  Type       0000  0000  ----  ----  0000  0000                                                                                                                                               
Sent         0000  0006  0000  0000  0000  0006                                                                                                                                               
  Rexmit     ----  ----  0000  ----  ----  ----                                                                                                                                               
nsh> 

```

yjdwbj, 2020-11-09:
The following I tried that turn on the network debug flag or disabled network ,also still not  seeing any debug info from the nsh shell.

yjdwbj, 2020-11-09:
Sorry, the debug everything is fine, this is my mistake to configured.

closed 2020-11-09 by yjdwbj