apache/nuttx #2276: Atmel sam4s-xplained-pro board can't using built-in SD Card and nand flash

yjdwbj, 2020-11-10:
Hi, guys , I have one Atmel sam4s-xplained-pro board, when I using the nuttx os find some issues. the built-in
[image]
[image]

 MMCSD can't work. following is my testing steps.

```sh
~$ tools/configure.sh sam4s-xplained-pro:nsh

~$ make && openocd -f board/atmel_sam4s_xplained_pro.cfg -c init -c "reset halt" -c "flash write_image erase nuttx.bin 0x00400000" -c "reset"

Open On-Chip Debugger 0.10.0+dev-01447-g91c4c83f4-dirty (2020-11-04-12:11)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: JTAG Supported
Info : CMSIS-DAP: FW Version = 1.0
Info : CMSIS-DAP: Serial# = ATML1803040200001055
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 500 kHz
Info : SWD DPIDR 0x2ba01477
Info : ATSAM4SD32C.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for ATSAM4SD32C.cpu on 3333
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x004000cc msp: 0x20001d00
Info : sam4 does not auto-erase while programming (Erasing relevant sectors)
Info : sam4 First: 0x00000000 Last: 0x0000000d
Info : Erasing sector: 0x00000000
Info : Erasing sector: 0x00000001
Info : Erasing sector: 0x00000002
Info : Erasing sector: 0x00000003
Info : Erasing sector: 0x00000004
Info : Erasing sector: 0x00000005
Info : Erasing sector: 0x00000006
Info : Erasing sector: 0x00000007
Info : Erasing sector: 0x00000008
Info : Erasing sector: 0x00000009
Info : Erasing sector: 0x0000000a
Info : Erasing sector: 0x0000000b
Info : Erasing sector: 0x0000000c
Info : Erasing sector: 0x0000000d
auto erase enabled
wrote 114688 bytes from file nuttx.bin in 5.806748s (19.288 KiB/s)

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : ATSAM4SD32C.cpu: external reset detected
Info : ATSAM4SD32C.cpu: external reset detected
```
Don't have any errors  during  did above steps. but the board reset each 5 seconds. Then I review the .config find content like a s below:

```sh
~$ grep "WDT" .config
CONFIG_SAM34_WDT=y
CONFIG_WDT_ENABLED_ON_RESET=y
# CONFIG_WDT_DISABLE_ON_RESET is not set
CONFIG_WDT_TIMEOUT=5000
CONFIG_WDT_MINTIME=2500
CONFIG_WDT_THREAD=y
CONFIG_WDT_THREAD_NAME="wdog"
CONFIG_WDT_THREAD_INTERVAL=2500
CONFIG_WDT_THREAD_PRIORITY=200
CONFIG_WDT_THREAD_STACKSIZE=512
```
And, I changed the config and then rebuild it.
[... 132 lines cut ...]
mmcsd_decode_csd:   CCC: 1461                                                                                                                                                                 
mmcsd_decode_csd:   READ_BL_LEN: 9 READ_BL_PARTIAL: 0                                                                                                                                         
mmcsd_decode_csd:   WRITE_BLK_MISALIGN: 0 READ_BLK_MISALIGN: 0                                                                                                                                
mmcsd_decode_csd:   DSR_IMP: 0                                                                                                                                                                
mmcsd_decode_csd:   SD Block Addressing:                                                                                                                                                      
mmcsd_decode_csd:     C_SIZE: 15199 SD_ER_BLK_EN: 1                                                                                                                                           
mmcsd_decode_csd:     SD_SECTOR_SIZE: 127 SD_WP_GRP_SIZE: 0                                                                                                                                   
mmcsd_decode_csd:   WP_GRP_EN: 0 MMC DFLT_ECC: 0 (MMC) R2W_FACTOR: 2                                                                                                                          
mmcsd_decode_csd:   WRITE_BL_LEN: 9 WRITE_BL_PARTIAL: 0                                                                                                                                       
mmcsd_decode_csd:   FILE_FORMAT_GROUP: 0 COPY: 0                                                                                                                                              
mmcsd_decode_csd:   PERM_WRITE_PROTECT: 0 TMP_WRITE_PROTECT: 0                                                                                                                                
mmcsd_decode_csd:   FILE_FORMAT: 0 ECC: 0 (MMC) CRC: 58                                                                                                                                       
mmcsd_decode_csd: Capacity: 7782400Kb, Block size: 512b, nblocks: 15564800 wrprotect: 0                                                                                                       
mmsd_decode_scr: SCR:                                                                                                                                                                         
mmsd_decode_scr:   SCR_STRUCTURE: 0 SD_VERSION: 2                                                                                                                                             
mmsd_decode_scr:   DATA_STATE_AFTER_ERASE: 0 SD_SECURITY: 3 SD_BUS_WIDTHS: 5                                                                                                                  
mmsd_decode_scr:   Manufacturing data: 00000001                                                                                                                                               
mmcsd_widebus: Wide bus operation selected                                                                                                                                                    
                                                                                                                                                                                              
nsh>                                                                                                                                                                                          
nsh> mount -t vfat /dev/mmcsd0 /mnt                                                                                                                                                           
find_blockdriver: pathname="/dev/mmcsd0"                                                                                                                                                      
mmcsd_open: Entry                                                                                                                                                                             

``

adamfeuer, 2020-11-10:
Hi,

It looks like the sam4s-xplained-pro can talk with the SD Card to get the
cardinfo structure. So that's working. That seems to be happening in 1 bit
bus mode. The switch to wide bus operation may not be working correctly.

Can you debug and find out where in mount it's hanging? Or use info()
logging statements to help you find out what's hanging? When debugging SD
Card drivers I also found that having a logic analyzer helped so I could
see the communication between the board and the SD Card. Do you have one?

I used one of these to attach the logic analyzer:
https://www.sparkfun.com/products/12941

-adam

adamfeuer, 2020-11-10:
Whoops sorry, I meant to link to this micro SD Card sniffer rather than the
breakout board:

https://www.sparkfun.com/products/9419

Here's a full size sniffer if you need that:

https://www.sparkfun.com/products/11468

I mainly have micro SD cards, so just used the micro SD sniffer with an
adapter.

-adam

yjdwbj, 2020-11-11:
Thank you  @adamfeuer  reply to me, I having a logic analyzer, and connected to the right PIOD interface. 
[image]

And I enabled the sam_sendcmd output like below:
```sh
BD                                                                                                                                                                                            
initializing HSMCI                                                                                                                                                                            
sam_hsmci_initialize: Initializing SDIO                                                                                                                                                       
sdio_initialize: slotno: 0                                                                                                                                                                    
mmcsd_slotinitialize: minor: 0                                                                                                                                                                
mmcsd_hwinitialize: DMA supported: 1                                                                                                                                                          
mmcsd_hwinitialize: Attached MMC/SD interrupts                                                                                                                                                
sam_registercallback: Register 0x413e65(0x200042e0)                                                                                                                                           
sam_callbackenable: eventset: 02                                                                                                                                                              
sam_callback: Callback 0x413e65(0x200042e0) cbevents: 02 cdstatus: 00                                                                                                                         
sam_callback:  No... return without performing the callback                                                                                                                                   
mmcsd_slotinitialize: MMC/SD slot is empty                                                                                                                                                    
sam_cardinserted: Slot 0 inserted: YES                                                                                                                                                        
sdio_mediachange: cdstatus OLD: 00 NEW: 01                                                                                                                                                    
sam_callback: Callback 0x413e65(0x200042e0) cbevents: 02 cdstatus: 01                                                                                                                         
sam_callback: Callback to 0x413e65(0x200042e0)                                                                                                                                                
mmcsd_mediachange: arg: 0x200042e0                                                                                                                                                            
mmcsd_probe: type: 0 probed: 0                                                                                                                                                                
mmcsd_removed: type: 0 present: 1                                                                                                                                                             
sam_sendcmd: cmd: 00000000 arg: 00000000 regval: 00000000                                                                                                                                     
sam_sendcmd: cmd: 00000000 arg: 00000000 regval: 00000000                                                                                                                                     
sam_sendcmd: cmd: 00008101 arg: 40200000 regval: 00001041                                                                                                                                     
sam_waitresponse: ERROR: cmd: 00008101 events: 009b0001 SR: 0410c0e4                                                                                                                          
mmcsd_sendcmdpoll: ERROR: Wait for response to cmd: 00008101 failed: -116                                                                                                                     
mmcsd_cardidentify: ERROR: CMD1 RECVR3: -22                                                                                                                                                   
sam_sendcmd: cmd: 00000208 arg: 000001aa regval: 00001048                                                                                                                                     
mmcsd_cardidentify: SD V2.x card                                                                                                                                                              
[... cut ...]
sam_sendcmd: cmd: 00000077 arg: 12340000 regval: 00001077                                                                                                                                     
sam_sendcmd: cmd: 00000046 arg: 00000002 regval: 00001046                                                                                                                                     
mmcsd_widebus: Wide bus operation selected                                                                                                                                                    
```

The CMD and CLK  that it's looks like right.
[image]

adamfeuer, 2020-11-11:
Ok, CMD and CLK look ok, but that trace doesn't show response data from the
card on the DATn lines.

The next thing to do is to try to capture some traces from when you try to
do a mount. You should see NuttX send a command to read data, and then see
data come back from the card on the DATn lines. Can you try this?

Also, do you have memory card and file system logging turned on (docs link
<https://nuttx.apache.org/docs/latest/quickstart/debugging.html>)? If the
data transfers are failing, and you have file system error warn and info on
you should see messages the describe the failure. Examining the logic
analyzer output along with the logging should give you a clue as to what's
going on. If you try this and post some more logic analyzer captures and
logs I'll try to help.

-adam

yjdwbj, 2020-11-30:
@adamfeuer  Thanks your reply, I have a SPI over SD sniffer, and installed it to test, at least the worked for a TOSHIBA 8GB SDHC Memory Card.  But instert another KINGMAX 256MB SD Card , it's report like below , looks said the CMD9 failed.
```sh
Registering CDC/ACM serial driver                                                                                                                                                             
sam_automount_initialize: Initializing automounter(s)                                                                                                                                         
automount_initialize: lower=0x41e404                                                                                                                                                          
sam_cardinserted_internal: Slot 0 inserted: NO                                                                                                                                                
Mounting procfs to /proc                                                                                                                                                                      
sam_sdinitialize: Initializing SERCOM SPI0                                                                                                                                                    
sam_sdinitialize: Successfully initialized SPI0                                                                                                                                               
sam_sdinitialize: Binding SPI0 to MMC/SD slot 0                                                                                                                                               
mmcsd_mediainitialize: Send CMD0                                                                                                                                                              
mmcsd_sendcmd: CMD0[00000000] R1=01                                                                                                                                                           
mmcsd_mediainitialize: Card is in IDLE state                                                                                                                                                  
mmcsd_mediainitialize: Send CMD8                                                                                                                                                              
mmcsd_sendcmd: CMD8[000001aa] R1=03 R7=ffffffff                                                                                                                                               
mmcsd_mediainitialize: Send CMD55/ACMD41                                                                                                                                                      
mmcsd_sendcmd: CMD55[00000000] R1=01                                                                                                                                                          
mmcsd_sendcmd: CMD41[00000000] R1=01                                                                                                                                                          
mmcsd_mediainitialize: Identified SD ver1 card                                                                                                                                                
mmcsd_mediainitialize: 0. Send CMD55/ACMD41                                                                                                                                                   
mmcsd_sendcmd: CMD55[00000000] R1=01                                                                                                                                                          
mmcsd_sendcmd: CMD41[00000000] R1=00                                                                                                                                                          
mmcsd_mediainitialize: Get CSD                                                                                                                                                                
mmcsd_sendcmd: CMD9[00000000] R1=00                                                                                                                                                           
mmcsd_getcardinfo: 0. SPI send returned ff                                                                                                                                                    
mmcsd_getcardinfo: 1. SPI send returned fc                                                                                                                                                    
mmcsd_getcardinfo: 2. SPI send returned 00                                                                                                                                                    
mmcsd_getcardinfo: 3. SPI send returned fc                                                                                                                                                    
mmcsd_getcardinfo: 4. SPI send returned 00                                                                                                                                                    
[... cut ...]
mmcsd_geometry: ERROR: mmcsd_getcsd returned -5                                                                                                                                               
nx_mount: ERROR: Bind method failed: -19                                                                                                                                                      
ERROR: Failed to mount the FAT filesystem: -1 (19)                                                                                                                                            
 ```                                        
[image]
[image]

yjdwbj, 2020-11-30:
> Thank you @adamfeuer reply to me, I having a logic analyzer, and connected to the right PIOD interface.
> [image]
> 
> And I enabled the sam_sendcmd output like below:
> 
> ```shell
> BD                                                                                                                                                                                            
> initializing HSMCI                                                                                                                                                                            
> sam_hsmci_initialize: Initializing SDIO                                                                                                                                                       
> sdio_initialize: slotno: 0                                                                                                                                                                    
> mmcsd_slotinitialize: minor: 0                                                                                                                                                                
> mmcsd_hwinitialize: DMA supported: 1                                                                                                                                                          
> mmcsd_hwinitialize: Attached MMC/SD interrupts                                                                                                                                                
> sam_registercallback: Register 0x413e65(0x200042e0)                                                                                                                                           
> sam_callbackenable: eventset: 02                                                                                                                                                              
> sam_callback: Callback 0x413e65(0x200042e0) cbevents: 02 cdstatus: 00                                                                                                                         
> sam_callback:  No... return without performing the callback                                                                                                                                   
> mmcsd_slotinitialize: MMC/SD slot is empty                                                                                                                                                    
> sam_cardinserted: Slot 0 inserted: YES                                                                                                                                                        
> sdio_mediachange: cdstatus OLD: 00 NEW: 01                                                                                                                                                    
> sam_callback: Callback 0x413e65(0x200042e0) cbevents: 02 cdstatus: 01                                                                                                                         
> sam_callback: Callback to 0x413e65(0x200042e0)                                                                                                                                                
> mmcsd_mediachange: arg: 0x200042e0                                                                                                                                                            
> mmcsd_probe: type: 0 probed: 0                                                                                                                                                                
> mmcsd_removed: type: 0 present: 1                                                                                                                                                             
> sam_sendcmd: cmd: 00000000 arg: 00000000 regval: 00000000                                                                                                                                     
> sam_sendcmd: cmd: 00000000 arg: 00000000 regval: 00000000                                                                                                                                     
> sam_sendcmd: cmd: 00008101 arg: 40200000 regval: 00001041                                                                                                                                     
> sam_waitresponse: ERROR: cmd: 00008101 events: 009b0001 SR: 0410c0e4                                                                                                                          
> mmcsd_sendcmdpoll: ERROR: Wait for response to cmd: 00008101 failed: -116                                                                                                                     
> mmcsd_cardidentify: ERROR: CMD1 RECVR3: -22                                                                                                                                                   
> sam_sendcmd: cmd: 00000208 arg: 000001aa regval: 00001048                                                                                                                                     
> mmcsd_cardidentify: SD V2.x card                                                                                                                                                              
[... cut ...]
> sam_sendcmd: cmd: 00000046 arg: 00000002 regval: 00001046                                                                                                                                     
> mmcsd_widebus: Wide bus operation selected                                                                                                                                                    
> ```
> 
> The CMD and CLK that it's looks like right.
> [image]

I use the GDB to debug and find out the program is freeze at  `usleep(MMCSD_CLK_DELAY)`, .

```sh
mmcsd_mmcinitialize (priv=<optimized out>) at mmcsd/mmcsd_sdio.c:2780
2780	  usleep(MMCSD_CLK_DELAY);
```

closed 2020-11-30 by yjdwbj