apache/nuttx #2289: Compile error of qemu port, Ho to deal with this error?

13824125580, 2020-11-13:
Hi folks:
   my procedure is this:
 1.clone the nuttx and apps project and let then in the same directory.
 2.  ./tools/configure.sh qemu-intel64:nsh and got success.
 3. execute "make" in top nuttx dir, and got failure.
~/WorkSpace/smp-nuttx/nuttx-upstream/clean-code/nuttx$ make 
make[1]: Entering directory '/home/czl/WorkSpace/smp-nuttx/nuttx-upstream/clean-code/nuttx/tools'
make[1]: Leaving directory '/home/czl/WorkSpace/smp-nuttx/nuttx-upstream/clean-code/nuttx/tools'
make[1]: Entering directory '/home/czl/WorkSpace/smp-nuttx/nuttx-upstream/clean-code/nuttx'
make[2]: Entering directory '/home/czl/WorkSpace/smp-nuttx/nuttx-upstream/clean-code/nuttx/boards'
make[2]: Leaving directory '/home/czl/WorkSpace/smp-nuttx/nuttx-upstream/clean-code/nuttx/boards'
make[2]: Entering directory '/home/czl/WorkSpace/smp-nuttx/nuttx-upstream/clean-code/apps'
make[3]: Entering directory '/home/czl/WorkSpace/smp-nuttx/nuttx-upstream/clean-code/apps/platform'
make[3]: Leaving directory '/home/czl/WorkSpace/smp-nuttx/nuttx-upstream/clean-code/apps/platform'
make[3]: Entering directory '/home/czl/WorkSpace/smp-nuttx/nuttx-upstream/clean-code/apps/builtin'
make[3]: Leaving directory '/home/czl/WorkSpace/smp-nuttx/nuttx-upstream/clean-code/apps/builtin'
make[2]: Leaving directory '/home/czl/WorkSpace/smp-nuttx/nuttx-upstream/clean-code/apps'
make[2]: Entering directory '/home/czl/WorkSpace/smp-nuttx/nuttx-upstream/clean-code/nuttx/graphics'
make[3]: Entering directory '/home/czl/WorkSpace/smp-nuttx/nuttx-upstream/clean-code/nuttx/graphics/nxglib'
make[3]: Leaving directory '/home/czl/WorkSpace/smp-nuttx/nuttx-upstream/clean-code/nuttx/graphics/nxglib'
make[3]: Entering directory '/home/czl/WorkSpace/smp-nuttx/nuttx-upstream/clean-code/nuttx/graphics/nxglib'
make[3]: Leaving directory '/home/czl/WorkSpace/smp-nuttx/nuttx-upstream/clean-code/nuttx/graphics/nxglib'
make[3]: Entering directory '/home/czl/WorkSpace/smp-nuttx/nuttx-upstream/clean-code/nuttx/graphics/nxglib'
make[3]: Leaving directory '/home/czl/WorkSpace/smp-nuttx/nuttx-upstream/clean-code/nuttx/graphics/nxglib'
make[2]: Leaving directory '/home/czl/WorkSpace/smp-nuttx/nuttx-upstream/clean-code/nuttx/graphics'
make[2]: Entering directory '/home/czl/WorkSpace/smp-nuttx/nuttx-upstream/clean-code/nuttx/arch/x86_64/src'
make[2]: Nothing to be done for 'clean_context'.
make[2]: Leaving directory '/home/czl/WorkSpace/smp-nuttx/nuttx-upstream/clean-code/nuttx/arch/x86_64/src'
make[1]: Leaving directory '/home/czl/WorkSpace/smp-nuttx/nuttx-upstream/clean-code/nuttx'
Create .version
error: unknown option `sort=taggerdate'
usage: git tag [-a|-s|-u <key-id>] [-f] [-m <msg>|-F <file>] <tagname> [<head>]
   or: git tag -d <tagname>...
   or: git tag -l [-n[<num>]] [--contains <commit>] [--points-at <object>] 
		[<pattern>...]
   or: git tag -v <tagname>...

    -l, --list            list tag names
    -n[<n>]               print <n> lines of each tag message
    -d, --delete          delete tags
    -v, --verify          verify tags

Tag creation options
    -a, --annotate        annotated tag, needs a message
    -m, --message <message>
                          tag message
    -F, --file <file>     read message from file
    -s, --sign            annotated and GPG-signed tag
    --cleanup <mode>      how to strip spaces and #comments from message
    -u, --local-user <key id>
                          use another key to sign the tag
    -f, --force           replace the tag if exists
    --column[=<style>]    show tag list in columns

Tag listing options
    --contains <commit>   print only tags that contain the commit
    --points-at <object>  print only tags of the object

Missing versioning information
USAGE: tools/version.sh [-d|-h] [-b <build>] [-v <major.minor.patch>] <outfile-path>
Try 'tools/version.sh -h' for more information
tools/Makefile.unix:230: recipe for target '/home/czl/WorkSpace/smp-nuttx/nuttx-upstream/clean-code/nuttx/.version' failed
make: *** [/home/czl/WorkSpace/smp-nuttx/nuttx-upstream/clean-code/nuttx/.version] Error 1

so, any one can tell me what is the rootcause?
thank you!

btashton, 2020-11-13:
Looks like you have a very old version of git and should probably upgrade. But you can get around this by running the tool directly and supplying a version like 9.1.0 `./tools/version.sh -h`

13824125580, 2020-11-13:
> Looks like you have a very old version of git and should probably upgrade. But you can get around this by running the tool directly and supplying a version like 9.1.0 `./tools/version.sh -h`

it worked, thank you! :)

closed 2020-11-13 by btashton