apache/nuttx #2310: which configuration should choose if wanna to run nuttx on qemu?

13824125580, 2020-11-16:
HI folks:
  how to configuration the sdk if want to run nuttx on qemu?
thank you!

btashton, 2020-11-16:
All the QEMU configurations build I think you are facing some issues due to running Ubuntu 14 which is really really old. The 64bit version is a little more complicated to set up because it also requires using grub. 

It would be best to first understand what you want to do.  The simulator might work better depending.

13824125580, 2020-11-17:
> All the QEMU configurations build I think you are facing some issues due to running Ubuntu 14 which is really really old. The 64bit version is a little more complicated to set up because it also [...]

 I am planing to use nuttx as my project kernel. so first i want to use qemu,i to familiar nuttx first. and then i need to verify the
stability of SMP  mechanism.
as i know, the Nuttx SMP it not stable at present.

btashton, 2020-11-17:
SMP should be quite stable especially in master now. @masayuki2009 has done a lot of work on it. You can also run the simulator in SMP mode. See `./tools/configure.sh sim:smp`

The K210 also runs in QEMU with SMP. 

https://github.com/apache/incubator-nuttx/tree/master/boards/risc-v/k210/maix-bit

xiaoxiang781216, 2020-11-17:
you can try sabre-6quad with QEMU:
https://github.com/apache/incubator-nuttx/blob/master/boards/arm/imx6/sabre-6quad/configs/smp/defconfig
@masayuki2009 has done many SMP work on it. And here is the setup instruction for QEMU:
https://github.com/apache/incubator-nuttx/blob/master/boards/arm/imx6/sabre-6quad/README.txt#L590

13824125580, 2020-11-17:
It seem s the K210 default configuration not support SMP ＠btashton
 236 # CONFIG_SPINLOCK is not set
 237 # CONFIG_SPINLOCK_IRQ is not set
 238 # CONFIG_IRQCHAIN is not set
 239 # CONFIG_IRQCOUNT is not set
 240 # CONFIG_SMP is not set                                                                                                                                                                   
 241 # CONFIG_INIT_NONE is not set
 242 CONFIG_INIT_ENTRYPOINT=y

qemu is just a emulator, not simulator, As my experience with qemu , on SMP scenario,some timing related issue cant be reproduced on Qemu. And its effective MIPS benchmark it too low.  so qumu running SMP did not has too much meanings.
real board should be more meaningful on SMP. @xiaoxiang781216

btashton, 2020-11-17:
> It seem s the K210 default configuration not support SMP @btashton
>  236 # CONFIG_SPINLOCK is not set
>  237 # CONFIG_SPINLOCK_IRQ is not set
>  238 # CONFIG_IRQCHAIN is not set
>  239 # CONFIG_IRQCOUNT [...]

https://www.github.com/apache/incubator-nuttx/tree/master/boards%2Frisc-v%2Fk210%2Fmaix-bit%2Fconfigs%2Fsmp

> qemu is just a emulator, not simulator, As my experience with qemu , on SMP scenario,some timing related issue cant be reproduced on Qemu. And its effective MIPS benchmark it too low.  so qumu running SMP did not has too much meanings.
> real board should be more meaningful on SMP. @xiaoxiang781216 

Sorry I don't understand what you are asking for there are multiple configurations we have pointed you at for ARM and RISC-V that will run in both QEMU and hardware as well as the non QEMU simulator.

xiaoxiang781216, 2020-11-17:
But your question is how to run nuttx on qemu, is it right? if you want to run on real device, just buy the board which has SMP supported by NuttX. The following boards are good candidate:
https://github.com/apache/incubator-nuttx/tree/master/boards/arm/cxd56xx/spresense
https://github.com/apache/incubator-nuttx/blob/master/boards/arm/imx6/sabre-6quad
Because @masayuki2009 has done many SMP on them recently and can give you help if you hit some issue.

13824125580, 2020-11-17:
Ok, I see, thank you! @xiaoxiang781216 @btashton

masayuki2009, 2020-11-17:
Hi @13824125580,

>as i know, the Nuttx SMP it not stable at present.

It's much stable than before but I need more work to fix some problems.

>It seem s the K210 default configuration not support SMP @btashton

You can use maix-bit:smp configuration with QEMU as well as a real board.
However, please note that OpenOCD with maix-bit is sometimes unstable.

>As my experience with qemu , on SMP scenario,some timing related issue cant be reproduced on Qemu.

Exactly. That's why I've started to use the sabre-6quad board recently to fix L1 d-cache issue.
With QEMU, we can test some simple use cases such as ostest, but we can not test cache coherency issues.

Also, real use-cases (e.g. W-Fi audio streaming and so on) are much more complicated,
because many drivers (e.g. SPI, DMA, I2S, ...) are involved. To test such use-cases, I've been using Sony Spresense board.

xiaoxiang781216, 2020-11-23:
It seems that the question has been addressed. Let's close it now.

closed 2020-11-23 by xiaoxiang781216