apache/nuttx #2314: chip/stm32_otghost.c doesn't build , have C errors

kernelgurumeditation, 2020-11-16:
tested with master version:

In file included from chip/stm32_otghost.c:73:0:
chip/stm32_otg.h:135:34: warning: 'struct usbdev_s' declared inside parameter list
 void stm32_usbsuspend(FAR struct usbdev_s *dev, bool resume);
                                  ^
chip/stm32_otg.h:135:34: warning: its scope is only this definition or declaration, which is probably not what you want
chip/stm32_otghost.c:78:2: warning: #warning OTG host not tested for STM32H7! [-Wcpp]
 #warning OTG host not tested for STM32H7!
  ^
chip/stm32_otghost.c:1235:33: error: expected ';', ',' or ')' before 'uint8_t'
                                 uint8_t speed,
                                 ^
chip/stm32_otghost.c: In function 'stm32_epfree':
chip/stm32_otghost.c:4245:3: error: 'ret' undeclared (first use in this function)
   ret = stm32_takesem_noncancelable(&priv->exclsem);
   ^
chip/stm32_otghost.c:4245:3: note: each undeclared identifier is reported only once for each function it appears in
chip/stm32_otghost.c: At top level:
chip/stm32_otghost.c:348:12: warning: 'stm32_ctrlchan_alloc' used but never defined
 static int stm32_ctrlchan_alloc(FAR struct stm32_usbhost_s *priv,
            ^
chip/stm32_otghost.c: In function 'stm32_epfree':
chip/stm32_otghost.c:4275:1: warning: control reaches end of non-void function [-Wreturn-type]
 }

Ouss4, 2020-11-17:
Thanks for reporting the error, #2322 should fix this.

kernelgurumeditation, 2020-11-17:
Have you noticed the:

chip/stm32_otghost.c:1235:33: error: expected ';', ',' or ')' before 'uint8_t'

Not sure if it was fixed in a previous commit

PR #2322 (2020-11-17): arch/arm/src/stm32h7/stm32_otghost.c: Fix undeclared "ret".
## Summary
fix #2314 
## Impact

## Testing
closed 2020-11-17 by davids5