apache/nuttx #2323: Howto to build external apps(elf)?

kernelgurumeditation, 2020-11-17:
Hello...
I have read all the howtos, videos...,  tried with symbols, without symbols.
Gone back to 9. ... 8... versions
and it never works....
I enabled elf loader debug in the Nuttx, and it seems that it fails to parse the  elf file(the text, data section size is read as zero..)
It seems that I'm not creating the elf files correctly but followed the instructions in the official wiki...

Does this work? currently? if so can someone point me to how to do it?

I just want to build an elf hello world, put it on a filesystem, and run it... I have seen videos from Alan.. it worked at some time...

Any help will be very appreciated

acassis, 2020-11-19:
hi @kernelgurumeditation I tested it in the middle of 2018 at that time version was 7.25! I need to tested it to verify if someone break it in this mean time.

btashton, 2020-11-19:
Alan also your tutorial required people to supply things that are already in the `scripts/Make.defs` file that is exported (maybe that file did not exist at the time?). We should probably show people how to make use of that rather than supply on the command line. 

Personally I find that the improvements made for including external folders for apps / board / arch much easier to work with.

btashton, 2020-11-19:
@kernelgurumeditation also is this the same issue you opened in the Apps repo that we commented on? If so we should probably close this and keep the conversation there.

kernelgurumeditation, 2020-11-19:
@btashton, unfortunately, no, the other issue is an error linking.

This is about building a standalone elf file, with or without symbols,  as described: 

https://cwiki.apache.org/confluence/pages/viewpage.action?pageId=139629543

In this case, It builds, but when I try to run it on the board it fails.

I have enabled the elf loader debug and I'm seeing very strange things when it loads the elf section headers.... all the values seem to be wrong regarding endianness....  but the elf header is read correctly and the values are correct in the file...
Very strange

closed 2020-11-20 by xiaoxiang781216

xiaotailang, 2023-06-26:
hi @btashton @acassis @kernelgurumeditation I also want to compile an ELF format hello program. I have read your discussion and successfully compiled the hello program, which can run normally. However, when I compile my own application into ELF format and run it on the board, I encounter the issue of "myapp:: command not found". I have investigated the problem and found that it is related to the compilation options or linking conditions. For example, when I turn off certain linking options, myapp can run normally. Therefore, I suspect that there is an error in the system when loading or executing myapp, but the system does not provide specific information at this time and always gives the "myapp:: command not found" prompt. I want to try debugging my ELF file myapp with GDB. Which configuration options should I enable during compilation?In addition, myapp file is executed from the SPI flash. My NuttX version is 8.2.