apache/nuttx #2366: debug: nuttx-gdbinit not producing backtraces on armv7e-m

btashton, 2020-11-22:
When I try to dump the threads with `nxthread_bt_all` nothing is shown and then it crashes the target. I am using this with STM32H7 and the ARM Embedded 2020-q2-update toolchain:
```
GNU gdb (GNU Arm Embedded Toolchain 9-2020-q2-update) 8.3.1.20191211-git
```

```
nx_start () at init/nx_start.c:832
832	      up_idle();
(gdb) bt
#0  nx_start () at init/nx_start.c:832
#1  0x080003ce in __start () at chip/stm32_start.c:426
(gdb) nxthread
target examined 
_target_arch.name=armv7e-m
$_target_has_fpu : 0 
$_target_has_smp : 0 
saved current_tcb (pid=0) 
Missing argument 0 in user function.
(gdb) nxthread_all_bt 
* 0 Thread 0x24000504  (Name: Idle Task, State: Running, Priority: 0) 0x80025c6 in nx_start()
#0  nx_start () at init/nx_start.c:832
#1  0x080003ce in __start () at chip/stm32_start.c:426
  1 Thread 0x38000490  (Name: hpwork, State: Waiting,Signal, Priority: 224) 0x0 in No()
#0  0x00000000 in ?? ()
  2 Thread 0x38001130  (Name: lpwork, State: Waiting,Signal, Priority: 100) 0x0 in No()
#0  0x00000000 in ?? ()
  3 Thread 0x38001dd0  (Name: init, State: Waiting,Semaphore, Priority: 100) 0x0 in No()
#0  0x00000000 in ?? ()
  4 Thread 0x38002b60  (Name: usbhost, State: Waiting,Semaphore, Priority: 100) 0x0 in No()
#0  0x00000000 in ?? ()
  5 Thread 0x38003420  (Name: Telnet daemon, State: Waiting,Semaphore, Priority: 100) 0x0 in No()
#0  0x00000000 in ?? ()
```

btashton, 2020-11-22:
@masayuki2009 any idea what is going on here? The `Missing argument 0 in user function.` does not look good.

masayuki2009, 2020-11-22:
> @masayuki2009 any idea what is going on here? The `Missing argument 0 in user function.` does not look good.

What happens if you call ```info_nxthread``` before calling ```nxthread``` ?

btashton, 2020-11-22:
Still looks like it is not grabbing the stackframes correctly:
```c
(gdb) c
Continuing.
^C
Program received signal SIGINT, Interrupt.
stm32_phyread (phydevaddr=0, phyregaddr=1, value=0x38002866) at chip/stm32_ethernet.c:3228
3228	      if ((stm32_getreg(STM32_ETH_MACMDIOAR) & ETH_MACMDIOAR_MB) == 0)
(gdb) info_nxthreads 
  0 Thread 0x240004a4  (Name: Idle Task, State: Ready, Priority: 0) 0x0 in No()
  1 Thread 0x38000490  (Name: hpwork, State: Waiting,Signal, Priority: 224) 0x0 in No()
  2 Thread 0x38001130  (Name: lpwork, State: Ready, Priority: 100) 0x0 in No()
  3 Thread 0x38001dd0  (Name: init, State: Running, Priority: 100) 0x0 in No()
  4 Thread 0x38002b60  (Name: usbhost, State: Ready, Priority: 100) 0x0 in No()
  5 Thread 0x38003420  (Name: Telnet daemon, State: Ready, Priority: 100) 0x0 in No()
(gdb) bt
#0  stm32_phyread (phydevaddr=0, phyregaddr=1, value=0x38002866) at chip/stm32_ethernet.c:3228
#1  0x0800a264 in stm32_phyinit (priv=0x240068e0 <g_stm32ethmac>) at chip/stm32_ethernet.c:3517
#2  0x0800a80c in stm32_ethconfig (priv=0x240068e0 <g_stm32ethmac>) at chip/stm32_ethernet.c:4312
#3  0x08009c90 in stm32_ifup (dev=0x2400693c <g_stm32ethmac+92>) at chip/stm32_ethernet.c:2580
#4  0x08032034 in netdev_ifup (dev=0x2400693c <g_stm32ethmac+92>) at netdev/netdev_ioctl.c:1833
#5  0x08031bce in netdev_ifr_ioctl (psock=0x38002164, cmd=1818, req=0x3800297c) at netdev/netdev_ioctl.c:907
#6  0x08031f6e in psock_vioctl (psock=0x38002164, cmd=1818, ap=...) at netdev/netdev_ioctl.c:1681
#7  0x08032006 in netdev_vioctl (sockfd=8, cmd=1818, ap=...) at netdev/netdev_ioctl.c:1810
#8  0x0801edfa in nx_vioctl (fd=8, req=1818, ap=...) at vfs/fs_ioctl.c:151
#9  0x0801ef38 in ioctl (fd=8, req=1818) at vfs/fs_ioctl.c:259
#10 0x0800add4 in netlib_ifup (ifname=0x8038ee4 "eth0") at netlib_setifstatus.c:97
#11 0x0800acb8 in netinit_net_bringup () at netinit.c:603
#12 0x0800ad60 in netinit_configure () at netinit.c:701
#13 0x0800ad6c in netinit_bringup () at netinit.c:1045
#14 0x0800aabe in nsh_consolemain (argc=1, argv=0x38002a58) at nsh_consolemain.c:104
#15 0x0800aa6c in nsh_main (argc=1, argv=0x38002a58) at nsh_main.c:168
#16 0x08006a00 in nxtask_startup (entrypt=0x800aa0d <nsh_main>, argc=1, argv=0x38002a58) at sched/task_startup.c:165
#17 0x08002b36 in nxtask_start () at task/task_start.c:144
#18 0x00000000 in ?? ()
(gdb) 
```

masayuki2009, 2020-11-22:
> Still looks like it is not grabbing the stackframes correctly:

Hmm, it's strange...
```
$ arm-none-eabi-gdb --version
GNU gdb (GNU Arm Embedded Toolchain 9-2020-q2-update) 8.3.1.20191211-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

I've just tried ```stm32f4discovery:rndis```

```
(gdb) target extended-remote:3333                                                                                                                                                                                              
Remote debugging using :3333                                                                                                                                                                                                   
up_idle () at chip/stm32_idle.c:208                                                                                                                                                                                            
208       END_IDLE();                                                                                                                                                                                                          
(gdb) info_nxthreads                                                                                                                                                                                                           
target examined                                                                                                                                                                                                                
_target_arch.name=armv7e-m                                                                                                                                                                                                     
$_target_has_fpu : 0                                                                                                                                                                                                           
$_target_has_smp : 0                                                                                                                                                                                                           
saved current_tcb (pid=0)                                                                                                                                                                                                      
* 0 Thread 0x20000308  (Name: Idle Task, State: Running, Priority: 0) 0x80051c6 in up_idle()                                                                                                                                   
  1 Thread 0x20004a80  (Name: lpwork, State: Waiting,Signal, Priority: 100) 0x8005778 in arm_switchcontext()                                                                                                                   
  2 Thread 0x200056c0  (Name: init, State: Waiting,Semaphore, Priority: 100) 0x8005778 in arm_switchcontext()                                                                                                                  
  3 Thread 0x20007820  (Name: Telnet daemon, State: Waiting,Semaphore, Priority: 100) 0x8005778 in arm_switchcontext()    
```

masayuki2009, 2020-11-22:
>Still looks like it is not grabbing the stackframes correctly:

@btashton 
Which board and configuration are you using now?

btashton, 2020-11-22:
I am using `nucleo-h743zi:otg_fs_host` which was just merged in, the only change I made is setting the thread name limit to 31 instead of 0.  When I used `stm32f429i-disco:nsh` everything was fine.

masayuki2009, 2020-11-22:
@btashton 
Perhaps, I found the root cause.

Actually, ```nucleo-.h743zi:otg_fs_host``` uses FPU but it was not detected by the nuttx-gdbinit script.
Currently, it just checks the symbol named ```fpuconfig``` but ```stm32_fpuconfig``` was used instead.
For the interim solution, could you try it by replacing fpuconfig with stm32_fpuconfig?

btashton, 2020-11-22:
@masayuki2009 I had to make an additional change to use `lookup_symbol` instead of `lookup_global_symbol` which returns a tuple with the first value as the symbol (or None).

`python if (type(gdb.lookup_symbol("stm32_fpuconfig")[0]) is gdb.Symbol) : \`

There are 22 different definitions of this {}_fpuconfig symbol so I am not quite sure what the change should be here.

masayuki2009, 2020-11-22:
>There are 22 different definitions of this {}_fpuconfig symbol so I am not quite sure what the change should be here.

@btashton 
I came up with a new idea to check if the target uses FPU.
It just checks the size of tcb.xcp.regs[].
I will try this idea later.

closed 2020-11-23 by btashton