apache/nuttx #2388: nuttx-gdbinit uses symbol from unrelated application

v01d, 2020-11-23:
When trying to use nuttx-gdbinit, it complains that `g_statenames` is not defined. I found this symbol in `note_main.c` from note app. nuttx-gdbinit facility should not depend an a symbol from an unrelated app. The symbol should either by defined in nuttx itself or inside nuttx-gdbinit.

v01d, 2020-11-23:
Actually, this array is static so I'm not even sure if this is the place where it is supposed to be defined. Also, this same information is defined in three places:

```
nuttx/fs/procfs/fs_procfsproc.c:static FAR const char *g_statenames[] =
apps/examples/thttpd/content/tasks/tasks.c:static const char *g_statenames[] =
apps/system/sched_note/note_main.c:static FAR const char *g_statenames[] =
```

I think the best is to define this information in one place, which will make it available to all required places, including openocd which can be made to retrieve it instead of define it itself.

v01d, 2020-11-23:
@masayuki2009 it seems you authored this file. Where is this variable defined for you?

btashton, 2020-11-23:
I think procfs is where is was supposed to be pulled from which is used by the `ps` command.

v01d, 2020-11-23:
But it is static there as well

v01d, 2020-11-23:
Ah, the symbol is still visible. I assumed it wouldn't. So maybe this place isn't such a bad place to have it defined I guess.
I will document that PROCFS needs to be enabled for this to work. Also, I would update openocd to use this then.

closed 2020-11-23 by v01d