apache/nuttx #2401: Did #2335 break something?

Ouss4, 2020-11-25:
After #2335 I can't build esp32-core:wapi.  I see however that the CI builds it correctly, but I can't locally.

```
make V=1

...

make[1]: Entering directory '/home/ouss4/NuttX/nuttx/mm'
make[1]: Nothing to be done for 'depend'.
make[1]: Leaving directory '/home/ouss4/NuttX/nuttx/mm'
make[1]: Entering directory '/home/ouss4/NuttX/nuttx/libs/libxx'
make[1]: Nothing to be done for 'depend'.
make[1]: Leaving directory '/home/ouss4/NuttX/nuttx/libs/libxx'
make[1]: Entering directory '/home/ouss4/NuttX/nuttx/wireless'
make makedepfile
make[2]: Entering directory '/home/ouss4/NuttX/nuttx/wireless'
cat   >  Make.dep
```
The build hangs at the last command.

Reverting #2335 the build continues with no issues.

Can't build in 2 Arch based computers. Building in a Ubuntu based docker image passes correctly, however... Was it only tested on Debian based distros? We had few changes before that broke other non Debian distros. 
Some tools used in the build system are implementation dependant and change from distro to distro.  Last time we had an issue with different implementations of `touch`.

v01d, 2020-11-25:
I tested on my Ubuntu, that is the only OS I can test myself. 

The make rule being executed there is:
```
makedepfile: $(CSRCS:.c=.ddc) $(ASRCS:.S=.dds)
	$(call CATFILE, Make.dep, $^)
	$(call DELFILE, $^)
```
which means that CSRCS/ASRCS appears to be empty, which is strange.

I just reproduced the issue locally, will see what it is about.

v01d, 2020-11-25:
@Ouss4 is CONFIG_WIRELESS really needed for this config? It seems that while WIRELESS is turned on, not WIRELESS_* option is enabled so nothing is actually added to the build. I disabled it and it builds OK, but I'm not sure if the option is required
for any other reason.
I think that would be one thing to fix. 

As to having empty CSRCS/ARSCS, I'm not sure if that is a normal thing to have and consider or not. I could make the dep rule not run if it is empty somehow, but maybe that would hide an unexpected situation.

Ouss4, 2020-11-25:
> @Ouss4 is CONFIG_WIRELESS really needed for this config?

I think it's necessary to include the wireless directory into the build.
https://github.com/apache/incubator-nuttx/blob/cb8d2672305c4a2f8740a2c561dd71a16ab3b7a8/tools/Directories.mk#L147-L151
https://github.com/apache/incubator-nuttx/blob/cb8d2672305c4a2f8740a2c561dd71a16ab3b7a8/tools/FlatLibs.mk#L135-L139

v01d, 2020-11-25:
OK, I will change CATFILE definition so that it does not hang on empty list of input files (a sane expectation). I'm not sure how to adapt the windows case, though. This is how the change looks for the Linux case so far:

```Make
# CATFILE - Cat a list of files
#
# USAGE: $(call CATFILE,dest,src1,src2,src3,...)

ifeq ($(CONFIG_WINDOWS_NATIVE),y)
define CATFILE
	$(Q) type $(2) > $1
endef
else
define CATFILE
	$(Q) if [ -z $(2) ]; then echo '' > Make.dep; else cat $(2) > $1; fi
endef
endif
```

v01d, 2020-11-25:
BTW, I'm perplexed as to why this didn't fail on CI. My guess is that stdin is not available and thus "cat > Make.dep" would return immediately.

patacongo, 2020-11-26:
> 
> 
> OK, I will change CATFILE definition so that it does not hang on empty list of input files (a sane expectation). I'm not sure how to adapt the windows case, though. This is how the change looks for the Linux case so far:
> 
> ```makefile
> # CATFILE - Cat a list of files
> #
> # USAGE: $(call CATFILE,dest,src1,src2,src3,...)
> 
> ifeq ($(CONFIG_WINDOWS_NATIVE),y)
> define CATFILE
> 	$(Q) type $(2) > $1
> endef
> else
> define CATFILE
> 	$(Q) if [ -z $(2) ]; then echo '' > Make.dep; else cat $(2) > $1; fi
> endef
> endif
> ```

Shouldn't cause a hang.  It should generate a (uncaptured) error message and generate an empty file:

    C:\Users\spuda>type > foobar
    The syntax of the command is incorrect.
    
    C:\Users\spuda>dir foobar
     Volume in drive C has no label.
     Volume Serial Number is 947F-7817
    
     Directory of C:\Users\spuda
    
    11/25/2020  06:05 PM                 0 foobar
                   1 File(s)              0 bytes
                   0 Dir(s)  943,090,741,248 bytes free

v01d, 2020-11-26:
Ok, then, I'll submit the above change. Thanks for the input Greg.

PR #2402 (2020-11-26): Config.mk: fix CATFILE macro to work when file list is empty (Linux)
## Summary

The change makes CATFILE work as expected for an empty list. Fixes #2401 

## Impact

Affected build (make depend) on Linux when CSRCS/ASRCS was empty.

## Testing

esp32-core:wapi on Linux
closed 2020-11-26 by Ouss4