apache/nuttx #2405: tools/configure.sh  <board> a lot of syntax error,kconfig-frontends-v4.11

yjdwbj, 2020-11-26:
Hi, I had have apply the same of issues, I  was think that I am  did wrong way. so I make a folder and clone it. But it's got this error. 
```sh
~$  tools/configure.sh  nucleo-l152re:nsh
Copy files
  Select CONFIG_HOST_LINUX=y
  Refreshing...
[.....]
arch/arm/src/imxrt/Kconfig:1212: syntax error
arch/arm/src/imxrt/Kconfig:1211: invalid option
arch/arm/src/stm32l4/Kconfig:5236: syntax error
arch/arm/src/stm32l4/Kconfig:5185: missing end statement for this entry
arch/arm/src/stm32l4/Kconfig:5053: missing end statement for this entry
arch/arm/src/stm32l4/Kconfig:8: missing end statement for this entry
arch/arm/Kconfig:1017: missing end statement for this entry
arch/arm/Kconfig:7: missing end statement for this entry
Kconfig:1809: missing end statement for this entry
arch/arm/src/stm32l4/Kconfig:5235: invalid statement
arch/arm/src/stm32l4/Kconfig:5245: unexpected end statement
arch/arm/src/stm32l4/Kconfig:5248: syntax error
arch/arm/src/stm32l4/Kconfig:5247: invalid statement
arch/arm/src/stm32l4/Kconfig:5257: unexpected end statement
arch/arm/src/stm32l4/Kconfig:5260: syntax error
arch/arm/src/stm32l4/Kconfig:5259: invalid statement
arch/arm/src/stm32l4/Kconfig:5269: unexpected end statement
arch/arm/src/stm32l4/Kconfig:5271: unexpected end statement
arch/arm/src/stm32l4/Kconfig:5273: unexpected end statement
arch/arm/src/stm32l4/Kconfig:6097: unexpected end statement
arch/arm/Kconfig:1018: unexpected end statement
arch/arm/Kconfig:1032: unexpected end statement
Kconfig:1811: unexpected end statement
boards/sim/sim/sim/Kconfig:53: syntax error
boards/sim/sim/sim/Kconfig:52: invalid option
drivers/note/Kconfig:62: syntax error
drivers/note/Kconfig:61: invalid option
drivers/sensors/Kconfig:227: syntax error
drivers/sensors/Kconfig:226: invalid option
make: *** [tools/Makefile.unix:492: olddefconfig] Error 1
ERROR: failed to refresh
```

First of all, I installed the **`kconfig-frontends    4.11.0.1+dfsg-2`**. I found this syntax error have the same of  characteristic，the character '>` in the Kconfig.  I filter it out below:

```sh
~$grep -n '>' arch/arm/src/imxrt/Kconfig
1211:	depends on IMXRT_DTCM > 0

~$ grep -n '>' arch/arm/src/stm32l4/Kconfig 
5235:if STM32L4_ADC1_INJ_CHAN > 0
5247:if STM32L4_ADC2_INJ_CHAN > 0
5259:if STM32L4_ADC3_INJ_CHAN > 0

~$ grep -n '>' boards/sim/sim/sim/Kconfig
52:	depends on SENSORS_WTGAHRS2 && SIM_UART_NUMBER > 0

~$ grep -n '>' drivers/note/Kconfig
61:	default 256 if TASK_NAME_SIZE > 0
```

Finally, I removed these character '>' by a bash shell, then everything is fine.

```sh
~$ make menuconfig 2>&1 | grep "Kconfig"  | awk -F ':' '{print $1}' | uniq  | xargs sed -i 's/ >.*//g'
```

davids5, 2020-11-26:
@btashton , @v01d - I have seen this issue mentioned before. I believe it is a version-itas issue.  Do we have a "Prerequisite Section" somewhere that lists the minimum versions of tools we require?

v01d, 2020-11-26:
The version report is the same as I have on 20.04 (just a different build: 4.11.0.1+dfsg-3).
@yjdwbj what OS are you using? I'm suspecting there is something wrong in your environment.

@davids5 this is our documentation right now, distinguishing Ubuntu 20.04 from other cases: file:///home/v01d/coding/nuttx_doc/nuttx/Documentation/_build/html/quickstart/quickstart.html
Which means that we are indirectly specifying a specific kconfig version in either case.

yjdwbj, 2020-11-27:
@v01d  My OS is Debian 10(Buster) , I used Linux kernel is my custom build.
```sh
uname -a
Linux debian 5.8.14-20200909-lcy #1 SMP Wed Oct 14 10:00:45 CST 2020 x86_64 GNU/Linux
```
Which means I use `make menuconfig` for `latest Linux kernel` have nothing syntax error. So ,  I am sure my environment and cross-tools is right.  Because I removed these character '>', then can build the final nuttx.bin , and then flash to the target board. and the board can  startup run.

btashton, 2020-11-27:
You should definitely not be removing the '>' .  If you are getting a valid build you are lucky and likely only because you are not using one of the configuration that actually uses it.

Technically this is the build of kconfig-frontends that we actually support and test against
`https://bitbucket.org/nuttx/tools/src/master/kconfig-frontends/`

Can you try using that?

yjdwbj, 2020-11-27:
> You should definitely not be removing the '>' . If you are getting a valid build you are lucky and likely only because you are not using one of the configuration that actually uses it.
>
> Technically [...]

Here is my `kconfig-frontends"

```sh
~$ dpkg -l | grep "kconfig-frontends"
ii  kconfig-frontends                                           4.11.0.1+dfsg-2                                           amd64        Standalone implementation of the Linux Kconfig parser and frontend
```
I check out the `https://bitbucket.org/nuttx/tools/src/master/kconfig-frontends/` code .  The latest commit at 2018-02-14 that it said "kconfig-configs: Update .version and remove old files so that is an exact copy of 4.11.0.1".  Do you means  the nuttx must use the `Special kconfig-frontends`, the other OS release maybe not working, or have some else reason?

v01d, 2020-11-27:
> @v01d My OS is Debian 10(Buster) , I used Linux kernel is my custom build.
> 
> ```shell
> uname -a
> Linux debian 5.8.14-20200909-lcy #1 SMP Wed Oct 14 10:00:45 CST 2020 x86_64 GNU/Linux
> ```
> 
> Which means I use `make menuconfig` for `latest Linux kernel` have nothing syntax error. So , I am sure my environment and cross-tools is right. Because I removed these character '>', then can build the final nuttx.bin , and then flash to the target board. and the board can startup run.

Comparing the kconfig-frontends package between Debian 10 and Ubuntu 20.04 they appear to use the same version. At most there's some configuration difference or different patching being done, so I'm not sure what could be happening. Maybe some dependency is different?

v01d, 2020-11-27:
> I check out the `https://bitbucket.org/nuttx/tools/src/master/kconfig-frontends/` code . The latest commit at 2018-02-14 that it said "kconfig-configs: Update .version and remove old files so that [...]

The problem is that there doesn't seem to be an authoritative source for kconfig-frontends source-code anymore so now there are many forks and each distro uses different patches. But I think that it is worth it to try to support kconfig-frontends as packaged from major OSs so that bootstrapping NuttX is simpler.

btashton, 2020-11-27:
I really think this is something broken in the environment.  I just grabbed the `debian:10` docker container and ran this:
```
apt install make gcc git kconfig-frontends
git clone https://github.com/apache/incubator-nuttx --depth 1 nuttx
git clone https://github.com/apache/incubator-nuttx-apps --depth 1 apps
cd nuttx
./tools/configure.sh sim:nsh
```
and

yjdwbj, 2020-11-27:
Yeah, I am sure about the question point is that must use the  `https://bitbucket.org/nuttx/tools/src/master/kconfig-frontends/` . The Debian's release  of `kconfig-frontends 4.11.0.1+dfsg-2` is can't working.

closed 2020-11-27 by yjdwbj

btashton, 2020-11-27:
@yjdwbj  but did you see my comment I used a clean Debian container with the kconfig-frontends from the distribution.

yjdwbj, 2020-11-27:
@btashton  I had have saw your comment , and I actually find out the question point.  I have compile the `kconfig-frontends` from `https://bitbucket.org/nuttx/tools/src/master/kconfig-frontends/`. and the syntax error really   disappeared.

btashton, 2020-11-27:
It is strange because I was using this `kconfig-frontends/stable 4.11.0.1+dfsg-2 amd64` from Debian 10.  But I am glad to hear you have it working now using the source repo.

avatar1912, 2021-05-24:
> Yeah, I am sure about the question point is that must use the `https://bitbucket.org/nuttx/tools/src/master/kconfig-frontends/` . The Debian's release of `kconfig-frontends 4.11.0.1+dfsg-2` is can't working.

Good Jobs!!! I resolved my problem by this version of kconfig-frontends.